mcguirepr89/BirdNET-Pi #742: Authentication not working after adding in RTSP stream

AtomicLaser, 2023-02-17:
Bit of a weird one, but I seem to be loosing the ability to login to use the 'tools' section after adding in a RTSP stream for some reason.

As far as I am aware, the default login is birdnet with no password, however after adding in a RTSP stream for the microphone, when I click on the tools menu, I get the password prompt of which I tap in birdnet as the username and leave the password field blank.  This refreshes the page and loads up the administration menu, however when pressing any of the admin menu (like settings) the authentication prompt appears again and then entering the same details as above, the login prompt just re-appears.

Any ideas what could be causing this issue?

ehpersonal38, 2023-02-18:
Hi @AtomicLaser! Thank you for the issue report.

I was able to replicate this, and I suspect it's due to your RTSP stream link having special characters in it, which is a no-no for the way the BirdNET-Pi configuration file handles things. So, I've gone ahead and changed the way that's handled in the code, and can confirm that it fixes the issue as expected. https://github.com/mcguirepr89/BirdNET-Pi/commit/75bc9918ef5b9e82ac8dba82e22df6b30075e8b2

To restore your installation then, here's what you should do:
1. Open `~/BirdNET-Pi/birdnet.conf` in any text editor.
2. Find the string `RTSP_STREAM`, and wrap double quotes around your RTSP stream link.  For example:
Before: `RTSP_STREAM=http://username@example.com/`
After: `RTSP_STREAM="http://username@example.com/"`
3. Refresh your BirdNET-Pi webpage. Your installation should now be un-broken, and you should be able to access Tools as normal now. 
5. Navigate to Tools -> System Controls -> Update, and the fix I just pushed for this issue will be installed on your system, and you should now be able to update your RTSP stream link from Tools -> Settings -> Advanced Settings as normal without this issue occuring in the future.

Take care!

closed 2023-02-18 by ehpersonal38

AtomicLaser, 2023-02-19:
@ehpersonal38 Thanks for looking into this - I have done the update and it seems to have fixed the issue for me, although I will follow up in a few days once things has bedded in a bit more.

For extra clarity, the URL which crashed the authentication was 
`rtsp://192.168.1.71:554/user=xxxxx_password=xxxxx_channel=1_stream=0.sdp?real_stream`

I had managed to manually edit the config whilst the ticket was pending, and used a different stream link
`rtsp://xxxxx:xxxxx@192.168.1.175:554/h264Preview_01_main` which is a different make of camera which didn't throw up any issues with the auntentication.

The question here is what is defined as special characters which breaks the code?  Both link has special characters in, so assuming that the question mark or the equals sign would have broken the first link?