mcguirepr89/BirdNET-Pi #743: Disabling birdweather audio uploads

robtinkers, 2023-02-17:
I would like to contribute to birdweather, but I live in a city centre apartment, sometimes with substantial human activity outside.

As a result, I would be uncomfortable uploading audio, regardless of confidence levels and tuning parameters.

Would it be possible to have an option to notify birdweather of detections, but never upload any audio?

ehpersonal38, 2023-02-18:
I combed through the BirdWeather API and I didn't see any endpoints allowing this. If you want to talk to the developer directly about this you can post here: https://github.com/mcguirepr89/BirdNET-Pi/discussions/82.

BirdNET-Pi does however offer quite a robust "Privacy Mode," where you can effectively prevent any audio clips that contain human noises from being recorded/uploaded to BirdWeather. With the new model we implemented last month, the accuracy is extremely good in my experience.

ehpersonal38, 2023-02-18:
I just remembered that we're using a different API than the official one, I'll check it out soon and will update this.

robtinkers, 2023-02-18:
I did have a look at `server.py` and uploading the soundscape is a different call to adding a detection, but they are linked by a soundscapeId. Unfortunately I couldn't find documentation for any "add a detection" endpoint, so unclear if a valid soundscapeId is required.

ehpersonal38, 2023-02-18:
@robtinkers Can you try changing the 3 soundscape variables to blank strings and seeing if it still works?

robtinkers, 2023-02-18:
I've only just requested a BirdWeather ID so it might be a minute until I can try anything. Will update here when I can.

And I guess there is the separate question of if BirdWeather even wants detections without the accompanying soundscape (which is separate from the technical question of how to actually do that.) I'll ask Tim when he gets back to me.

robtinkers, 2023-02-18:
"At present, we're requiring audio"

So I guess that closes the issue.

closed 2023-02-18 by ehpersonal38