mcguirepr89/BirdNET-Pi #744: Spectrogram not recording, but interface detected

axelsoundingwild, 2023-02-18:
Everything was working fine until I've updated the Pi today, both system and BirdnetPi.

The spectrogram now is empty, showing no incoming audio but the interface works. I've tested the USB mic with another Pi and it works fine. How to sort this out? What kind of log would you need? Thanks

  [image]

ehpersonal38, 2023-02-18:
Hi there,

Looking at the spectrogram there does seem to be a faint signal, which is indicative that the microphone is still connected. I'm thinking the gain might've gotten turned down. So, can you try following these steps to increase the micrphone gain and see if that helps? https://github.com/mcguirepr89/BirdNET-Pi/wiki/Adjusting-your-sound-card

lloydbayley, 2023-02-26:
Did you have any luck with @ehpersonal38 's instructions? Do you have any further info?

axelsoundingwild, 2023-02-26:
Apologies for the delayed response

I have restarted the pi with the interface detached, then restarted it with
the interface plugged and it worked.

Not sure what happened…

-- 
*Axel Drioli*
*SoundingWild.com <http://soundingwild.com/> *

'My Garden Of A Thousand Bees' won the Wildscreen's 2022 Golden Panda Award
<https://wildscreen.org/my-garden-of-a-thousand-bees-leads-the-way-at-2022-wildscreen-panda-awards/>,
wildlife sound recordings provided by Axel.

Follow our nomadic 4x4 Overlanding journey capturing wildlife stories on
the African-Eurasian flyways, Sounding Wild Exploration
<https://www.soundingwild.com/>

Check out what audiences think about our work!

https://youtu.be/OwFGPWEpQK4

axelsoundingwild, 2023-02-26:
The gain was fine by the way.

-- 
*Axel Drioli*
*SoundingWild.com <http://soundingwild.com/> *

'My Garden Of A Thousand Bees' won the Wildscreen's 2022 Golden Panda Award
<https://wildscreen.org/my-garden-of-a-thousand-bees-leads-the-way-at-2022-wildscreen-panda-awards/>,
wildlife sound recordings provided by Axel.

Follow our nomadic 4x4 Overlanding journey capturing wildlife stories on
the African-Eurasian flyways, Sounding Wild Exploration
<https://www.soundingwild.com/>

Check out what audiences think about our work!

https://youtu.be/OwFGPWEpQK4

lloydbayley, 2023-02-26:
That's one hell of a signature!
Was a bit lost there for a sec.

Nothing to apologise for. Was just following up! :)

It's a bit weird how a restart cures things from time to time. It doesn't really give you a definitive answer does it? Still, that's the maddening world of computers/electronics.

Glad it's sorted out!

axelsoundingwild, 2023-02-26:
Ha, at least you noticed it :))

I know, so annoying at times..

Hope you fixed yours also

Axel

-- 
*Axel Drioli*
*SoundingWild.com <http://soundingwild.com/> *

'My Garden Of A Thousand Bees' won the Wildscreen's 2022 Golden Panda Award
<https://wildscreen.org/my-garden-of-a-thousand-bees-leads-the-way-at-2022-wildscreen-panda-awards/>,
wildlife sound recordings provided by Axel.

Follow our nomadic 4x4 Overlanding journey capturing wildlife stories on
the African-Eurasian flyways, Sounding Wild Exploration
<https://www.soundingwild.com/>

Check out what audiences think about our work!

https://youtu.be/OwFGPWEpQK4

lloydbayley, 2023-02-26:
Mine's been fine. Was just trying to help out in discussions by triaging things a bit and following up. Glad it works for you and all is well. Don't tell the washing-machine and fridge that there was a problem or they will round up the dishwasher and will all go out on strike!

closed 2023-02-27 by ehpersonal38