mcguirepr89/BirdNET-Pi #745: Mic only recording background noise

Metropolis1927, 2023-02-18:
I'm trying to get birdnetpi set up where I live but all I'm getting from the mic is background noise. Also, the spectrogram doesn't match what others show from their screenshots. It is missing the axis information and only displays the spectrogram, or possibly, the spectrogram is overlapping the other data. I'll try to attach a snip:
[image].

If I tap on the sound card, the "noise" will display on the spectrogram but other noise does not.

I am using a Pi 3 B+, a movo sound card (https://www.amazon.com/Movo-Microphone-External-Computer-Headphone/dp/B089NJMMGB), and a Rugged Mount Mono Primo EM272Z1 from micbooster.com. 

After running into this issue I searched this github for troubleshooting and tried a couple things from other posts (especially 551): 1) adjusted aslamixer -m gain and other settings for the sound card but this had no effect. 2) changed the microphone under advanced settings from default to the dsnoop name found on arecord -L output with no change to outcome. 3) restarted core services 4) checked journalctl -f and confirmed I didnt have a channel error, channel is set to 1 in advanced settings. 

I've tried plugging the sound card and mic into my laptop and changed my microphone settings to the new devices and they seem to be working just fine together. They recorded just fine on an online spectrogram (https://listeningtowaves.com/spectrograminfo).

What am I missing or not understanding? Any help would be greatly appreciated.

ehpersonal38, 2023-02-20:
Hi,

Thanks for detailing all the steps you've taken to try and troubleshoot this. Unfortunately I'm not really sure what the issue is. Could you tell me what version of Raspberry Pi OS you have on your 3B+? This command will let you know if you don't already:

`cat /etc/os-release`

Metropolis1927, 2023-02-20:
Thanks for the command; It's running Debian GNU/Linux 11 (bullseye)

ehpersonal38, 2023-02-20:
Hmm, that all looks good then. This is looking like a hardware issue. Could you try a new USB port on the Pi?

Metropolis1927, 2023-02-20:
I tried a different USB port and no change. Could it be the sound card? If so, do you know one that has worked for others? I didn't find any specific make/model that works or doesn't work when I searched the discussions other than I need a TRS socket for the microphone I'm using.

ehpersonal38, 2023-02-21:
The UGREEN sound cards are highly recommended. Here's a TRS one: https://www.amazon.com/UGREEN-External-Headphone-Microphone-Desktops/dp/B01N905VOY/?pldnSite=1&th=1

Metropolis1927, 2023-02-21:
Thank you. I've got one inbound... 1 March expected date.

Metropolis1927, 2023-03-01:
That did it! Replacing the MOVO sound card with the UGREEN sound card got it working. It's recording birds just fine and shows up on the spectrogram. The axis data on the live spectrogram is still missing but I'm not worried about that for my purposes. Thanks for the help.

closed 2023-03-01 by Metropolis1927