mcguirepr89/BirdNET-Pi #748: Incorrect command line argument on pi Zero setup

parcon, 2023-02-20:
On the page here (https://github.com/mcguirepr89/BirdNET-Pi/wiki/RPi0W2%20Installation%20Guide), the sed command is incorrect.

sudo sed -i 's/CONF_SWAPSIZE=100/CONF_SWAPSIZE=2048/g' /etc/dphys-swapfile &&
sudo sed -i 's/#CONF_MAXSWAP=2048/CONF_MAXSWAP=4096/g' /etc/dphys-swapfile &&
sudo -i sed '/^exit 0/i sudo iw wlan0 set power_save off' /etc/rc.local &&
sudo reboot

should be

sudo sed -i 's/CONF_SWAPSIZE=100/CONF_SWAPSIZE=2048/g' /etc/dphys-swapfile &&
sudo sed -i 's/#CONF_MAXSWAP=2048/CONF_MAXSWAP=4096/g' /etc/dphys-swapfile &&
sudo sed -i '/^exit 0/i sudo iw wlan0 set power_save off' /etc/rc.local &&
sudo reboot

lloydbayley, 2023-02-20:
Well-spotted! I'll see about fixing that.

ehpersonal38, 2023-02-20:
Good spot, thanks @parcon!