mcguirepr89/BirdNET-Pi #749: can not get into the Settings menu

duboisvb, 2023-02-20:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
I have installed Bridget- Pi and when I click on the tools icon it asked for a user password combination. I enter "birdnet" and leave the password blank.   It then brings up some options.  Some work fine but when I click on the Settings , it again asks for the User/passwod combination.  I tried the "birdnet" and no password but it just brings up the log in screen again.  

I cannot get into the Settings Menu. 

Can you help?  thanks

ehpersonal38, 2023-02-20:
Hi, I can think of a few reasons this may be happening:
1. Your CADDY_PWD (the web UI password) has special characters in it. Symbols like !@#$%^&* are not allowed in the password.
2. Your LATITUDE and LONGITUDE are not set properly. They need to be formatted like so: `29.2491`.
3. Some other error with your birdnet configuration file.

To fix this I recommend opening your birdnet configuration file located at `~/BirdNET-Pi/birdnet.conf` (~/ means your home directory, so on my system the full path will be `/home/pi/BirdNET-Pi/birdnet.conf`) and checking to see if any of the above issues ^ are occuring. If they are, you can edit the file to correct them.

duboisvb, 2023-02-20:
Thanks, not sure what actual problem was but after opening the BirdNET-Pi.conf file. i found the Latitude was written as LATITUDE == whereas Longitude only had one =. 
I entered the correct numbers, removed an equal sign and now it works as advertised.  Thanks

closed 2023-02-20 by ehpersonal38