mcguirepr89/BirdNET-Pi #756: Vanilla install failed at getting lat/lon, ifconfig.co/json is presenting a cloudflare bot challenge

w2xg, 2023-02-22:
```
Beginning ./install_config.sh
+ birdnet_conf=/home/icarus/BirdNET-Pi/birdnet.conf
++ curl -s4 ifconfig.co/json
++ jq .latitude
parse error: Invalid numeric literal at line 1, column 10
+ LATITUDE=
+ exit 1
The installation exited unsuccessfully.
```
curl -s4 ifconfig.co/json  yields html... looks to be a cloudflare bot challenge:

```
<!DOCTYPE html>
<html lang="en-US">
<head>
    <title>Just a moment...</title>
    <meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
    <meta http-equiv="X-UA-Compatible" content="IE=Edge">
    <meta name="robots" content="noindex,nofollow">
    <meta name="viewport" content="width=device-width,initial-scale=1">
    <link href="/cdn-cgi/styles/challenges.css" rel="stylesheet">

</head>
<body class="no-js">
    <div class="main-wrapper" role="main">
    <div class="main-content">
        <h1 class="zone-name-title h1">
            <img class="heading-favicon" src="/favicon.ico" alt="Icon for ifconfig.co"
                 onerror="this.onerror=null;this.parentNode.removeChild(this)">
            ifconfig.co
        </h1>
        <h2 class="h2" id="challenge-running">
            Checking if the site connection is secure
        </h2>
        <noscript>
            <div id="challenge-error-title">
                <div class="h2">
                    <span class="icon-wrapper">
                        <div class="heading-icon warning-icon"></div>
                    </span>
                    <span id="challenge-error-text">
                        Enable JavaScript and cookies to continue
                    </span>
                </div>
            </div>
        </noscript>
        <div id="trk_jschal_js" style="display:none;background-image:url('/cdn-cgi/images/trace/mana        ged/nojs/transparent.gif?ray=79d93a966aff8c59')"></div>
        <div id="challenge-body-text" class="core-msg spacer">
            ifconfig.co needs to review the security of your connection before proceeding.
        </div>
```

Had to do nano ./install_config.sh and manually input lat/lon on lines 10 and 11, removing the curls.

lloydbayley, 2023-02-23:
That's weird, just tried the ```curl -s4 ifconfig.co/json``` here and it spat out the desired JSON.
Is yours still giving the challenge if you run it manually now? Maybe there was a problem?

ehpersonal38, 2023-02-23:
ifconfig.co seems to have Cloudflare protection in front of it - which will block users with specific IP ranges, or from specific countries. I've gone ahead and switched ifconfig.co to a different API which should hopefully solve the issue for the future. Thanks for the report! And thanks @lloydbayley for testing things out

closed 2023-02-23 by ehpersonal38