mcguirepr89/BirdNET-Pi #758: can not install BirdNet

duboisvb, 2023-02-23:
Hello, I get an error when executing the curl command to install the program  

Can you advise? Thanks 

birder@raspberrypi:~ $ curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh | bash
Cloning into '/home/birder/BirdNET-Pi'...
remote: Enumerating objects: 130, done.
remote: Counting objects: 100% (130/130), done.
remote: Compressing objects: 100% (120/120), done.
remote: Total 130 (delta 7), reused 64 (delta 6), pack-reused 0
Receiving objects: 100% (130/130), 128.09 MiB | 5.97 MiB/s, done.
Resolving deltas: 100% (7/7), done.
Updating files: 100% (119/119), done.
+ exec
+++ date +%F
++ tee -i installation-2023-02-23.txt
+ set -e
+ my_dir=/home/birder/BirdNET-Pi
+ export my_dir=/home/birder/BirdNET-Pi
+ my_dir=/home/birder/BirdNET-Pi
+ cd /home/birder/BirdNET-Pi/scripts
++ uname -m
+ '[' aarch64 '!=' aarch64 ']'
+ ./install_config.sh
+ set -e
+ trap 'exit 1' SIGINT SIGHUP
+ echo 'Beginning ./install_config.sh'
Beginning ./install_config.sh
+ birdnet_conf=/home/birder/BirdNET-Pi/birdnet.conf
++ curl -s4 ifconfig.co/json
++ jq .latitude
parse error: Invalid numeric literal at line 1, column 10
+ LATITUDE=
+ exit 1
The installation exited unsuccessfully.

ehpersonal38, 2023-02-23:
Hi, I just changed the API that gets used to retrieve your latitude and longitude with a hopefully more robust one. I recommend deleting `newinstaller` and then running the install command again. Hopefully that will work for you.

duboisvb, 2023-02-24:
Thanks.  that seemed to work.  so far everything looks good.  Just having a little trouble with setting up my google email for notifications.  are there any examples i could see?

ehpersonal38, 2023-02-24:
For Gmail I believe you need to have 2FA enabled and then set up an "app" password instead of using your Gmail password.

See here for more information: https://github.com/caronc/apprise/issues/377#issuecomment-825187448

duboisvb, 2023-02-25:
Hello,  apologize for being so very new to all this but have 2 issues yet.
1. live audio seems to be working but volume is barely audible.  is there something to adjust.  I have the speaker volume turned all the way up .
2. Gave up on trying to get my gmail to work but created a yahoo email which always "times out" . I believe I have set it up as shown in the examples   

Thanks for any help.

Notifications

[Apprise Notifications](https://github.com/caronc/apprise/wiki) can be setup and enabled for 70+ notification services. Each service should be on its own line.

Apprise Notifications Configuration: 

mailto:// lotsofbirds:birdsgofly@myyahoo.com

Send Test Notification
2023-02-25 17:51:16,928 - INFO - Applying Custom Defaults
2023-02-25 17:51:16,930 - INFO - Notifying 1 service(s) asynchronously.
2023-02-25 17:52:32,043 - WARNING - Connection error while submitting email to myyahoo.com. Reason: timed out
Test sent!

ehpersonal38, 2023-02-25:
1. You can adjust your microphone volume with this guide - https://github.com/mcguirepr89/BirdNET-Pi/wiki/Adjusting-your-sound-card
2. Where are you getting `myyahoo.com` from? I think it should just be `yahoo.com`

duboisvb, 2023-02-26:
thanks again.  got the sound card to work fine  created new yahoo mail with yahoo.com but it still times out. i will keep trying things maybe go back to my gmail account

ehpersonal38, 2023-03-02:
Hey, did you get your Apprise notifications to finally work?

duboisvb, 2023-03-02:
Hello,

not yet.  i ended up doing a complete reinstall for other reasons.  I will play with it a bit more. When I send the test notification i always get a “timed out” error. But i can send emails to the account from my gmail account.

duboisvb, 2023-03-04:
hello, yes i got the gmail account to work.  Thanks    

Is there a facebook group for the Bridget-pi or other place where these issues might be discussed ?

ehpersonal38, 2023-03-04:
That's great to hear!

The best place to discuss BirdNET-Pi related things is probably going to be the Discussions here on GitHub. Link here: https://github.com/mcguirepr89/BirdNET-Pi/discussions. You can also access it by clicking "Discussions" at the very top of the page once you're on this repo.

closed 2023-03-04 by ehpersonal38