mcguirepr89/BirdNET-Pi #759: No recording, but no errors

tmceld, 2023-02-23:
Environment:
Raspberry Pi 4 Model B Rev 1.4
Linux birdnet 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"

Soundcard: https://www.amazon.co.uk/gp/product/B01N905VOY/
Mic: various

```
toast@birdnet:~/BirdNET-Pi$ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
    Playback/recording through the PulseAudio sound server
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
usbstream:CARD=Headphones
    bcm2835 Headphones
    USB Stream Output
usbstream:CARD=vc4hdmi0
    vc4-hdmi-0
    USB Stream Output
hw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Hardware device with all software conversions
sysdefault:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Front output / input
dsnoop:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample snooping device
usbstream:CARD=Device
    USB Audio Device
    USB Stream Output
usbstream:CARD=vc4hdmi1
    vc4-hdmi-1
    USB Stream Output
toast@birdnet:~/BirdNET-Pi$
```

So, starting off with 'default' in tools --> settings --> advanced settings -- audio card: In 'overview' and 'spectrogram' tabs the image updates about every 15 secs or so and 'looks' like it is working - however when i play birdsong through my phone near it, or even whistle or clap, nothing seems to register, and when i hit play in the top right i cannot hear anything.

So I try setting my audio card to `dsnoop:CARD=Device,DEV=0` as per the result of `arecord -L` but then (after a reboot) the spectrogram does not seem to update at all.  

I have tried adjusting the mic sensitivity with `alsamixer -m` but it doesn't make a difference.

My logs look fairly normal (this is from it sitting in my room with no common quails present ;-)

```
6:14:41---Files loaded
16:14:41---RECORDING_LENGTH set to 15
16:14:50---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts/analyze.py --i BirdSongs/February-2023/23-Thursday/2023-02-23-birdnet-16:14:34.wav --o BirdSongs/February-2023/23-Thursday/2023-02-23-birdnet-16:14:34.wav.csv --lat 53.XX --lon -1.XX --week 8 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
16:14:50---ANALYZING AUDIO... DONE! Time 1.5 SECONDS
16:14:50---WRITING RESULTS TO BirdSongs/February-2023/23-Thursday/2023-02-23-birdnet-16:14:19.wav.csv ... DONE! WROTE 0 RESULTS.
16:14:50---READING AUDIO DATA... DONE! READ 5 CHUNKS.
16:14:50---Date-time: 2023-02-23 16:14:34
16:14:51---0.0;3.0-('Pica pica_Eurasian Magpie', 0.04980834)
16:14:51---3.0;6.0-('Strix aluco_Tawny Owl', 0.04015299)
16:14:51---6.0;9.0-('Coturnix coturnix_Common Quail', 0.038167857)
16:14:51---9.0;12.0-('Coturnix coturnix_Common Quail', 0.0391793)
16:14:51---12.0;15.0-('Coturnix coturnix_Common Quail', 0.038983215)
16:14:55---Files loaded
16:14:55---RECORDING_LENGTH set to 15
16:15:06---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts/analyze.py --i BirdSongs/February-2023/23-Thursday/2023-02-23-birdnet-16:14:49.wav --o BirdSongs/February-2023/23-Thursday/2023-02-23-birdnet-16:14:49.wav.csv --lat 53.XX --lon -1.XX --week 8 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
16:15:06---ANALYZING AUDIO... DONE! Time 1.5 SECONDS
16:15:06---WRITING RESULTS TO BirdSongs/February-2023/23-Thursday/2023-02-23-birdnet-16:14:34.wav.csv ... DONE! WROTE 0 RESULTS.
16:15:06---READING AUDIO DATA... DONE! READ 5 CHUNKS.
16:15:06---Date-time: 2023-02-23 16:14:49
16:15:08---0.0;3.0-('Acrocephalus arundinaceus_Great Reed Warbler', 0.10102661)
16:15:08---3.0;6.0-('Coturnix coturnix_Common Quail', 0.040948756)
16:15:08---6.0;9.0-('Coturnix coturnix_Common Quail', 0.038507238)
16:15:08---9.0;12.0-('Coturnix coturnix_Common Quail', 0.041875646)
16:15:08---12.0;15.0-('Coturnix coturnix_Common Quail', 0.04024599)
16:15:11---Files loaded
16:15:11---RECORDING_LENGTH set to 15
16:15:20---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts/analyze.py --i BirdSongs/February-2023/23-Thursday/2023-02-23-birdnet-16:15:04.wav --o BirdSongs/February-2023/23-Thursday/2023-02-23-birdnet-16:15:04.wav.csv --lat 53.XX --lon -1.XX --week 8 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
16:15:20---ANALYZING AUDIO... DONE! Time 1.4 SECONDS
16:15:20---WRITING RESULTS TO BirdSongs/February-2023/23-Thursday/2023-02-23-birdnet-16:14:49.wav.csv ... DONE! WROTE 0 RESULTS.
16:15:20---READING AUDIO DATA... DONE! READ 5 CHUNKS.
16:15:20---Date-time: 2023-02-23 16:15:04
16:15:22---0.0;3.0-('Acrocephalus scirpaceus_Eurasian Reed Warbler', 0.04649886)
16:15:22---3.0;6.0-('Acrocephalus arundinaceus_Great Reed Warbler', 0.070552215)
16:15:22---6.0;9.0-('Coturnix coturnix_Common Quail', 0.03936199)
16:15:22---9.0;12.0-('Coturnix coturnix_Common Quail', 0.043981582)
16:15:22---12.0;15.0-('Coturnix coturnix_Common Quail', 0.044231325)
```
[image]

this *may* be related to: https://github.com/mcguirepr89/BirdNET-Pi/issues/745 ? I'm not sure

ehpersonal38, 2023-02-23:
Hi there,

Can you set the audio card back to `default`, then try accessing the Spectrogram on a Chrome browser by clicking "Spectrogram" next to "Today's Detections" at the top of the page? (it doesn't work on Safari, not sure about Firefox). This will allow you to see the live spectrogram which may help to facilitate your troubleshooting.

But from what I see on the static spectrogram image you provided everything looks normal, so I'm not really sure where the issue lies yet.

So, try out that live spectrogram and get let me know if that affords you any more information.

lloydbayley, 2023-02-24:
Slightly off-topic but: 
Woah! I never thought of looking at BirdNetPi in another browser. There are differences from Firefox (my preferred) and Chrome! I'm going to have to have a good look into this.

ehpersonal38, 2023-02-27:
Did you ever figure this one out?

tmceld, 2023-02-28:
No idea, at this point I think my only option is a reinstall.

 I'll report back

lloydbayley, 2023-03-03:
Any joy yet with your re-install?

lainsan, 2023-03-06:
@tmceld after fideling with soundcards and microphone options in birdnet-pi a lot, my go-to option to check for the working soundcard/micro is the live audio (top right of the webpage). Put something playing a song in front of the microphone and check if you can hear it at your computer.

I also used a ugreen soundcard - but unsuccessfully unfortunately. Gains were too low to be properly recognized.

Just from looking at your logs, it looks like your mic is recording, there's an input to the model and it's trying to find something   (e.g. magpie). That would not happen if the microphone is not recording anything.

I'm also performing a similar second test as you did: play bird songs from youtube in front of the microphone. What i do then is watch the live log first - it will show, as it did above for you - what it heard and made out of it.

ehpersonal38, 2023-03-07:
Closing as stale, please open another issue if your reinstall doesn't fix this, thanks

closed 2023-03-07 by ehpersonal38