mcguirepr89/BirdNET-Pi #769: Failure to install since requirements updated

DrBazUK, 2023-03-01:
For a variety of reasons I have been installing and reinstalling Birdnet.pi on Raspbian and other OS's today. However I am now getting a failure to install with the following error:

+ pip3 install -U -r /home/birder/BirdNET-Pi/requirements.txt
ERROR: Invalid requirement: 'streamlit=1.11.1' (from line 11 of /home/birder/Bir
Hint: = is not a valid operator. Did you mean == ?
The installation exited unsuccessfully.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Raspbian 64bit lite OS and write image using Raspberry Pi imager including user details specified in https://github.com/mcguirepr89/BirdNET-Pi/wiki/Installation-Guide
2. SSH into Raspberry pi using Putty
3. Copy and paste BASH script "curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh | bash"
4. Run through updates until script fails with error above.

**Expected behavior**
It should run without error and has done multiple times today (Tuesday 28 Feb) - I don't know enough code to know if the error is simply a missing = when declaring the variable or not.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem or your installation here.

**Your build**
Currently installing on Raspberry Pi 4 4GB model.

**Code or log snippets**
```
Paste your 
log output 
or 
code here
```

DrBazUK, 2023-03-01:
Have forked and confirmed that with the suggested edit to add a =, the script completes successfully. Maybe you'll make a coder of me yet :D

ehpersonal38, 2023-03-01:
*facepalms. I swear I even doubled checked it before committing!! Thank you for finding this issue!

> Have forked and confirmed that with the suggested edit to add a =, the script completes successfully. Maybe you'll make a coder of me yet :D

Ah yes, I meant to say -- this was an intentional bug to give new contributors a chance to contribute to the repo. Haha

closed 2023-03-01 by ehpersonal38

DrBazUK, 2023-03-01:
> *facepalms. I swear I even doubled checked it before committing!! Thank you for finding this issue!
>
>
> Ah yes, I meant to say -- this was an intentional bug to give new contributors a chance to [...]

Love the intent - I've been hovering around doing bits with code for too long - am finally getting some time to do something about it. Now sleep beckons - 0230hrs -