mcguirepr89/BirdNET-Pi #787: frequency downshift has stopped working

hoover67, 2023-03-17:
Hi folks, the "frequency downshift" in the player worked nicely for a few weeks. I updated my birdnet pi today and now, when I click on the downshift icon, the "play" icon in the player gets greyed out until I unshift the frequency again. I don't see anything bad in daemon.log while this happens, so at the moment it's difficult to provide further debug info.

thanks, Uwe

hoover67, 2023-03-18:
Sure, here you go. Thanks for taking a look! 

Uwe 

```
APPRISE_MINIMUM_SECONDS_BETWEEN_NOTIFICATIONS_PER_SPECIES=0
APPRISE_NOTIFICATION_BODY='A $sciname $comname was just detected with a confidence of $confidence $listenurl'
APPRISE_NOTIFICATION_TITLE="New BirdNET-Pi Detection"
APPRISE_NOTIFY_EACH_DETECTION=0
APPRISE_NOTIFY_NEW_SPECIES=1
APPRISE_NOTIFY_NEW_SPECIES_EACH_DAY=0
APPRISE_WEEKLY_REPORT=1
AUDIOFMT=mp3
BIRDNETPI_URL=

BIRDWEATHER_ID=REDACTED

CADDY_PWD="XXXXXXXXXXXXXX"
CHANNELS=2
CONFIDENCE=0.7
CUSTOM_IMAGE=
CUSTOM_IMAGE_TITLE=""
DATABASE_LANG=en
EXTRACTED=/home/pi/BirdSongs/Extracted
EXTRACTION_LENGTH=
FLICKR_API_KEY=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
FLICKR_FILTER_EMAIL=
FREQSHIFT_HI=6000
FREQSHIFT_LO=3000
FREQSHIFT_PITCH=-1500
FREQSHIFT_TOOL=sox
FULL_DISK=purge
HEARTBEAT_URL=
ICE_PWD=birdnetpi
IDFILE=/home/pi/BirdNET-Pi/IdentifiedSoFar.txt
LAST_RUN=
LATITUDE=52.1342
LONGITUDE=8.6993
MODEL=BirdNET_GLOBAL_3K_V2.2_Model_FP16
OVERLAP=0.0
PRIVACY_THRESHOLD=0
PROCESSED=/home/pi/BirdSongs/Processed
RAW_SPECTROGRAM=0
RECORDING_LENGTH=15
RECS_DIR=/home/pi/BirdSongs
REC_CARD=default
RTSP_STREAM=
SENSITIVITY=1.25
SF_THRESH=0.03
SILENCE_UPDATE_INDICATOR=0
SITE_NAME=""
SOX_SPECTROGRAM_EXTRA_ARGS=" -m -l -w dolph "
THIS_RUN=

```

hoover67, 2023-03-18:
BTW I removed all comments and blanks, hope that's not a problem.

ehpersonal38, 2023-03-18:
Thanks - that all looks good. There are a couple things you can do to debug this further:

1. Run the following command to enable more debugging info on the Recordings page. When you click the shift button, you should now see an alert popup at the top of the screen with either an error message or an "OK": 
`sed -i 's/if(this\.responseText == "OK"){/alert(this.responseText);&/' ~/BirdNET-Pi/scripts/play.php`

If the above command and subsequent testing doesn't yield any new information, try this next step:

2. Run the following command before and after you shift a recording. This will list the files in directory where shifted files are supposed to be created in, so if you don't see any files being created or deleted in this directory, that means something is definitely awry.
`find ~/BirdSongs/Extracted/By_Date/shifted/ -type f \( -name "*.wav" -o -name "*.mp3" \)`

hoover67, 2023-03-18:
>    1. Run the following command before and after you shift a recording.

Thanks for your suggestions!

once I run the sed command above I receive an "ok" message as a popup when
clicking on the downshift button.

As for find, the last successful downshift apparently was on 2022-10-23:

home/pi/BirdSongs/Extracted/By_Date/shifted/2022-10-23/Long-tailed_Tit/Long-tailed_Tit-75-2022-10-23-birdnet-08:39:36.mp3

I've checked the permissions of the "shifted" dir and they look ok.

All the best,

Uwe

ehpersonal38, 2023-03-18:
Hi Uwe,

Thanks for following my steps. Can you try downloading this custom "Recordings" PHP file to your BirdNET-Pi, then repeating the steps where you shift and unshift your detection? You'll be checking your browser's console for any error messages this time. (To open console on Chrome/Edge, press the following keys on your keyboard: CTRL + SHIFT + J, on Firefox: CTRL + SHIFT + K).

`wget https://gist.githubusercontent.com/ehpersonal38/196bd6c64aa4b250cace41a4339b4e53/raw/bbae9fd1e53e96e504a56c2d94761f130cc0ada8/gistfile1.txt -O $HOME/BirdNET-Pi/scripts/play.php`

hoover67, 2023-03-19:
Thanks for the custom play.php. I just installed it and tried shifting a
recording.

Now the popup displays /tmp/shift.sh not found and "OK", while the JS
console displays this:

views.php?view=Recordings&species=Hawfinch:156 shifting freqs of
2023-03-19/Hawfinch/Hawfinch-73-2023-03-19-birdnet-08:54:51.mp3

All the best,

Uwe

> Message ID: ***@***.***>
>

ehpersonal38, 2023-03-19:
That /tmp/shift.sh not found is definitely something I want to investigate. Please install this play.php which has a little more debugging in that regard:

`wget https://gist.githubusercontent.com/ehpersonal38/a90e8c6a9286e5ad41272e06ca2d410f/raw/1e24d9966b130fc6514f455d3a24f157cbdd2fc4/gistfile1.txt -O $HOME/BirdNET-Pi/scripts/play.php`

This time after shifting your detection, you should see a message in your browser's console. Please paste that here so I can take a look and make sure everything looks okay there. For reference mine looks like:
`mkdir -p /home/pi/BirdSongs/Extracted/By_Date/shifted/2023-03-19/American_Crow && echo "/usr/bin/nohup /usr/bin/sox "/home/pi/BirdSongs/Extracted/By_Date/2023-03-19/American_Crow/American_Crow-54-2023-03-19-birdnet-09:43:44.mp3" "/home/pi/BirdSongs/Extracted/By_Date/shifted/2023-03-19/American_Crow/American_Crow-54-2023-03-19-birdnet-09:43:44.mp3" pitch -q -1500" > /tmp/shift.sh && chmod +x /tmp/shift.sh;2>&1OK`

hoover67, 2023-03-20:
Here you go:

shifting freqs of
2023-03-20/Eurasian_Kestrel/Eurasian_Kestrel-95-2023-03-20-birdnet-08:26:24.mp3
views.php?view=Recordings&species=Eurasian+Kestrel:131 mkdir -p
/home/pi/BirdSongs/Extracted/By_Date/shifted/2023-03-20/Eurasian_Kestrel &&
echo "/usr/bin/nohup /usr/bin/sox
"/home/pi/BirdSongs/Extracted/By_Date/2023-03-20/Eurasian_Kestrel/Eurasian_Kestrel-95-2023-03-20-birdnet-08:26:24.mp3"
"/home/pi/BirdSongs/Extracted/By_Date/shifted/2023-03-20/Eurasian_Kestrel/Eurasian_Kestrel-95-2023-03-20-birdnet-08:26:24.mp3"
pitch -q -1500" > /tmp/shift.sh && chmod +x /tmp/shift.sh;2>&1sh: 1:
/tmp/shift.sh: not found
OK

This is weird behaviour because I can run this "echo" in a shell without
any issues, shift.sh gets created with the correct contents.

Also, there's plenty of space available under /tmp and it's probably not a
permission issue, either. Does caddy write any meaningful logs that I could
check? /var/log/caddy is empty on my system.

All the best,

Uwe

croisez, 2023-03-20:
I tell you to check about any error in the debug console, because I encounter some error myself:
there was an error related to the use of "setAttribute" on an undefined object.
This "elem" object is given as parameter by the http.onload() function, so it should always exist.

Now, I do not have such error.
I have done a series of shift/unshift on a series of mp3 files, and I got no errors anymore.
What I have done in fact is the following:
- go to the main page of your site
- at the Most Recent Detection, click on the shortcut to open in new tab
- on the URL of you browser, you should have something with index.php?filename=.....mp3
- append this string at the end of the URL, and press ENTER key: "&doshift=true"

On my setup, now I do not get any error in the Javascript console after done this manipulation.
I suspect perhaps a problem of browser cache data...

ehpersonal38, 2023-03-20:
@croisez Thanks for the input, though I suspect that your issue is different from Uwe's. I had him do some modifications to his `play.php` which enabled a little more debugging info, and you can see the PHP-side of his system fail when he runs the following commands to write the `sox` command to `/tmp/shift.sh` and then run it:

```
mkdir -p /home/pi/BirdSongs/Extracted/By_Date/shifted/2023-03-20/Eurasian_Kestrel &&
echo "/usr/bin/nohup /usr/bin/sox "/home/pi/BirdSongs/Extracted/By_Date/2023-03-20/Eurasian_Kestrel/Eurasian_Kestrel-95-2023-03-20-birdnet-08:26:24.mp3" "/home/pi/BirdSongs/Extracted/By_Date/shifted/2023-03 20/Eurasian_Kestrel/Eurasian_Kestrel-95-2023-03-20-birdnet-08:26:24.mp3" pitch -q -1500" > /tmp/shift.sh && 
chmod +x /tmp/shift.sh;2>&1
/tmp/shift.sh 2>&1
```

That command for him returns the following error:
`/tmp/shift.sh: not found`

@hoover67 Unfortunately I don't think caddy would log this sort of thing. Can you show me the permissions of your BirdNET-Pi/scripts directory with this command please?
`ls -lh ~/BirdNET-Pi/scripts/`

ehpersonal38, 2023-03-20:
I don't think there will be anything inside though, `/tmp/shift.sh: not found` indicates that the files were never created at all. So I'm thinking a permissions issue of some kind.

Also why are these commands echo'd into a .sh file and then run? Why not just run the commands directly from the PHP?

croisez, 2023-03-20:
i have done that on purpose to make debugging mire easier.
But of course now, we could do a popen or a system call.

ehpersonal38, 2023-03-20:
Hi @hoover67, I've just identified an issue with the BirdNET-Pi code that indeed caused a permissions issue. I've sorted this out and if you'd please update your system via Tools -> System Controls -> Update, and then try again, we can see if your issue is fixed.

hoover67, 2023-03-21:
Hi folks,

it's looking better now, I've removed all old directories in the "shifted"
folder (curiously owned by the "pi" user, not "caddy"), checked the perms
on "extracted" (777, which looked ok if a bit unsafe :-)) and now the
shifting seems to work again for the time being.

Thanks for your support!

All the best,

Uwe

closed 2023-03-21 by ehpersonal38

croisez, 2023-03-21:
On my setup, the /tmp/shift.sh and /tmp/unshift.sh are no longer created since last update of today

croisez, 2023-03-21:
I have made the following:
`# sudo touch /home/birdnet/BirdNET-Pi/scripts/blacklisted_images.txt
`
and
`scripts/ # sudo ./update_birdnet_snippets.sh
`
But it does not solve my issue

croisez, 2023-03-21:
After a little analysis, it appears that, on my system, all files below  /home/birdnet/BirdSongs are belonging to user birdnet:birdnet.
As file /tmp/shift.sh is created only if first command is successful, then it is never created.
```
$ mkdir -p /home/birdnet/BirdSongs/Extracted/By_Date/shifted/2023-03-21/Rougegorge_familier
mkdir: cannot create directory ‘/home/birdnet/BirdSongs/Extracted/By_Date/shifted/2023-03-21’: Permission denied
```
I have done the following command, and now shift.sh is well created, and freq downshifting is back
`sudo chown -R caddy:caddy BirdSongs`
But, I don't have any clue why all files under BirdSongs are belonging to user birdnet.

ehpersonal38, 2023-03-21:
@croisez I had this issue too, I've solved it by just tacking "sudo" onto the front of all the shifting commands in the PHP. Maybe not best practice but it works. Update your system and all should be fixed.

croisez, 2023-03-21:
I wonder how it can work, as Caddy is not in the sudoers list ... ??
You agree that the script is ran by Caddy, with rights associated to caddy user.
I suppose that caddy user is not in sudoers list

ehpersonal38, 2023-03-21:
Black magic? I'm not sure, but on my Pi both `pi` and `caddy` have sudo perms. 

```bash
caddy@birdnetpi:/home/pi $ sudo -l
Matching Defaults entries for caddy on birdnetpi:
    env_reset, mail_badpass, secure_path=/usr/local/sbin\:/usr/local/bin\:/usr/sbin\:/usr/bin\:/sbin\:/bin,
    env_keep+=NO_AT_BRIDGE, env_keep+="http_proxy HTTP_PROXY", env_keep+="https_proxy HTTPS_PROXY", env_keep+="ftp_proxy
    FTP_PROXY", env_keep+=RSYNC_PROXY, env_keep+="no_proxy NO_PROXY"

User caddy may run the following commands on birdnetpi:
    (ALL) NOPASSWD: ALL

```

croisez, 2023-03-21:
Yes you are right, on my PI, caddy has sudoers perm.
There is a file /etc/sudoers.d/010_caddy-nopasswd
which contains "caddy ALL=(ALL) NOPASSWD: ALL"
Then your solution is working.

croisez, 2023-03-21:
BTW, if you want, we could remove the use of /tmp/shift.sh and /tmp/unshift.sh, by doing a shell_exec() directly.
Are you going to code it, or you want me to send you a merge request ?
I would prefer the first solution ;-)

croisez, 2023-03-21:
```
git diff play.php
diff --git a/scripts/play.php b/scripts/play.php
index 943a93f..a87c1c2 100644
--- a/scripts/play.php
+++ b/scripts/play.php
@@ -112,24 +112,19 @@ if(isset($_GET['shiftfile'])) {
   $freqshift_tool = $config['FREQSHIFT_TOOL'];
 
   if ($freqshift_tool == "ffmpeg") {
-          $cmd = "/usr/bin/nohup /usr/bin/ffmpeg -y -i \"".$pi.$filename."\" -af \"rubberband=pitch=".$config['FREQSHIFT_LO']."/".$config['FREQSHIFT_HI']."\" \"".$shifted_path.$filename."\"";
-    shell_exec("mkdir -p ".$shifted_path.$dir." && echo \"".$cmd."\" > /tmp/shift.sh && chmod +x /tmp/shift.sh");
+    $cmd = "sudo /usr/bin/nohup /usr/bin/ffmpeg -y -i \"".$pi.$filename."\" -af \"rubberband=pitch=".$config['FREQSHIFT_LO']."/".$config['FREQSHIFT_HI']."\" \"".$shifted_path.$filename."\"";^M
+    shell_exec("mkdir -p ".$shifted_path.$dir." && ".$cmd);^M
 
   } else if ($freqshift_tool == "sox") {
     //linux.die.net/man/1/sox
     $soxopt = "-q";
     $soxpitch = $config['FREQSHIFT_PITCH'];
-          $cmd = "/usr/bin/nohup /usr/bin/sox \"".$pi.$filename."\" \"".$shifted_path.$filename."\" pitch ".$soxopt." ".$soxpitch;
-    shell_exec("sudo mkdir -p ".$shifted_path.$dir." && sudo echo \"".$cmd."\" > /tmp/shift.sh && sudo chmod +x /tmp/shift.sh");
+    $cmd = "sudo /usr/bin/nohup /usr/bin/sox \"".$pi.$filename."\" \"".$shifted_path.$filename."\" pitch ".$soxopt." ".$soxpitch;^M
+    shell_exec("sudo mkdir -p ".$shifted_path.$dir." && ".$cmd);^M
   }
-
-  shell_exec("sudo /tmp/shift.sh");
-  shell_exec("sudo rm -f /tmp/shift.sh");
     } else {
-  $cmd = "rm -f " . $shifted_path.$filename;
-  shell_exec("sudo echo \"".$cmd."\" > /tmp/unshift.sh && sudo chmod +x /tmp/unshift.sh");
-  shell_exec("sudo /tmp/unshift.sh");
-  shell_exec("sudo rm -f /tmp/unshift.sh");
+  $cmd = "sudo rm -f " . $shifted_path.$filename;^M
+  shell_exec($cmd);^M
     }
 
     echo "OK";

```

ehpersonal38, 2023-03-22:
Very good, it works! I'll push that code to the repo right now. Thanks @croisez .