mcguirepr89/BirdNET-Pi #788: Birdnet showing non english names for birds

Punkley, 2023-03-17:
Names for birds are using non english names, when, latin names exist.

See attached
[image]

Kingsley

ehpersonal38, 2023-03-17:
Hi, can you run the following command in your Pi's terminal and copy/paste the output?

`~/BirdNET-Pi/scripts/birds.db "SELECT com_name FROM detections GROUP BY com_name;"`

lloydbayley, 2023-03-18:
Might want to bung an sqlite3 in there to make:
```
sqlite3 ~/BirdNET-Pi/scripts/birds.db "SELECT com_name FROM detections GROUP BY com_name;"
```

Punkley, 2023-03-18:
sqlite3 ~/BirdNET-Pi/scripts/birds.db "SELECT com_nam                                                    e FROM detections GROUP BY com_name;"
Australian Magpie
Black-faced Cuckooshrike
Crimson Rosella
Eastern Yellow Robin
Galah
Little Corella
Little Raven
Little Wattlebird
Magpie-lark
Mistletoebird
Musk Lorikeet
New Holland Honeyeater
Pied Butcherbird
Rainbow Lorikeet
Red Wattlebird
Southern Boobook
Spotted Dove
Sulphur-crested Cockatoo
Weebill
Willie-wagtail
Yellow-tailed Black-Cockatoo
Горобець хатній
Дрізд чорний
Коловодник болотяний
Сапсан
Сипуха

ehpersonal38, 2023-03-18:
I suppose you changed your database language to Russian for a little while? I recommend removing those entries from the database so they don't appear in the web interface anymore.

`sqlite3 ~/BirdNET-Pi/scripts/birds.db "DELETE FROM detections WHERE com_name IN ('Горобець хатній', 'Дрізд чорний', 'Коловодник болотяний', 'Сапсан', 'Сипуха')"
`

Punkley, 2023-03-18:
Never was russian and it is still happening today, I am using the v2.2 DB if that helps
[image]

Kingsley

ehpersonal38, 2023-03-18:
Can I see the output of this command?

`ls -lh ~/BirdNET-Pi/model`

Punkley, 2023-03-19:
oh looked, I was using label_uk.txt I have changed it to label_en.txt

ehpersonal38, 2023-03-19:
Yeah that's Ukranian - closing this issue now.

closed 2023-03-19 by ehpersonal38