mcguirepr89/BirdNET-Pi #793: Apprise Notifications Not Saving On Update

paulhoff, 2023-03-20:
I put info into the apprise notification section.  The send test message works successfully.  However, when I push update all the info I have entered disappears.

ehpersonal38, 2023-03-20:
Interesting. Are you on the latest update? Also please run the following command on your Raspberry Pi and copy and paste the output of your birdnet.conf. You will need to censor any sensitive data such as passwords, API keys, etc:

`cat ~/BirdNET-Pi/birdnet.conf`

paulhoff, 2023-03-20:
Thanks.  I am not sure how to check what version I am running.  I am
running the most updated version from
https://github.com/mcguirepr89/BirdNET-Pi
<https://mailtrack.io/trace/link/89781f98e1e53948dd52b4cc579ca6f16446546a?url=https%3A%2F%2Fgithub.com%2Fmcguirepr89%2FBirdNET-Pi&userId=867221&signature=b433c6e79f2d6d8b>

Also the Apprise.txt contains nothing.

conf is below.

birdnet:~/BirdNET-Pi$ cat birdnet.conf
################################################################################
#                    Configuration settings for BirdNET-Pi
   #
################################################################################

# Optional: Site Title for banner

SITE_NAME="birdnet"

#--------------------- Required: Latitude, and Longitude
----------------------#

## The shell substitution below guesses these based on your network. THESE
NEED
## TO BE CHANGED TO STATIC VALUES
## Please only go to 4 decimal places. Example:43.3984

LATITUDE=redadcted
LONGITUDE=-104.9431

#--------------------------------- Model
--------------------------------------#
#_____________The variable below configures which BirdNET model
is_____________#
#______________________used for detecting bird
audio.__________________________#
#_It's recommended that you only change these values through the web
interface.#

MODEL=BirdNET_GLOBAL_3K_V2.2_Model_FP16
SF_THRESH=0.03

#---------------------  BirdWeather Station Information
-----------------------#
#_____________The variable below can be set to have your
BirdNET-Pi____________#
#__________________also act as a BirdWeather listening
station_________________#

BIRDWEATHER_ID=Egwne6wjeX7fv4zHmroYMR8A

#-----------------------  Web Interface User Password
------------------------#
#____________________The variable below sets the
'birdnet'_____________________#
#___________________user password for the Live Audio
Stream,___________________#
#_________________Tools, System Links, and the Processed files
________________#

## CADDY_PWD is the plaintext password (that will be hashed) and used to
access
## certain parts of the web interface

CADDY_PWD="redacted"

#-------------------------  Live Audio Stream
--------------------------------#
#_____________The variable below configures/enables the
live___________________#
[... 170 lines cut ...]

AUDIOFMT=mp3

## DATABASE_LANG is the language used for the bird species database
DATABASE_LANG=en

## HEARTBEAT_URL is a location to ping every time some analysis is done
## no information is sent to the the URL, its a heart beat to show that the
## analysis is continuing

HEARTBEAT_URL=

## SILENCE_UPDATE_INDICATOR is for quieting the display of how many commits
## your installation is behind by, relative to the Github repo. This number
## appears next to "Tools" when you're 50 or more commits behind.

SILENCE_UPDATE_INDICATOR=0

## These are just for debugging
LAST_RUN=
THIS_RUN=
IDFILE=/home/admin/BirdNET-Pi/IdentifiedSoFar.txt
RAW_SPECTROGRAM=0
CUSTOM_IMAGE=
CUSTOM_IMAGE_TITLE=""

ehpersonal38, 2023-03-20:
Thanks, your config all looks good there. I'm thinking this is a permissions issue. Can you please run the following command and copy and paste the output here so I can take a look?

`ls -lh ~/BirdNET-Pi/`

soumilgurjar, 2023-03-20:
I faced the same issue when setting up my apprise notifications despite being on the latest update. As mentioned by @ehpersonal38 , it was indeed a permissions issues and the solution was changing permissions as described in https://github.com/mcguirepr89/BirdNET-Pi/discussions/381#discussioncomment-3005877 .

ehpersonal38, 2023-03-20:
@soumilgurjar Thanks for bringing this to my attention! I did fairly recently change some code which set the permissions for BirdNET-Pi related files. I thought it was a harmless fix but apparently not. I've just done a commit now which should hopefully sort out all of this.

@paulhoff Go ahead and update your system via Tools -> System Controls -> Update, and please see if your issue is fixed.

paulhoff, 2023-03-23:
I updated to the most recent, rebooted.  That did NOT fix the problem.  However following the permissions command from #381 did fix it.  Thanks for all the help.

closed 2023-03-23 by ehpersonal38