mcguirepr89/BirdNET-Pi #799: I don't get notifications

Kouwe-sudo, 2023-03-22:
I want to receive notification from Birdnet when a new specie is discovered. I configured the section that should take care of that with aprise and everything looks okay. When I push the button for a test e-mail a get it together with a picture of some penguins. So everything seems to work. I checked one of the boxes: "Notify each species first detection of the day" but even after a week I received no email tough there are plenty of detections every day (> 100).  Of course a saved my settings.
Do I forget something or is it really a bug? I can hardly believe that because a lot more people would have complaint and it would been fix long ago.
I even tried 'all' in the box which species are to be reported and did not sum all the species that should be reported (comma separated)
I use a Pi 4+

ehpersonal38, 2023-03-22:
Hi, could you run the following command on your Pi and then copy and paste the output here so I can diagnose further?

`cat ~/BirdNET-Pi/birdnet.conf | grep APPRISE`

Kouwe-sudo, 2023-03-23:
This are the settings:

#-----------------------  Apprise Miscellanous Configuration
-------------------#

APPRISE_NOTIFICATION_TITLE="New BirdNet discovery"
APPRISE_NOTIFICATION_BODY='A $comname is just discovered $flickrimage
$listenurl'
APPRISE_NOTIFY_EACH_DETECTION=0
APPRISE_NOTIFY_NEW_SPECIES=0
APPRISE_WEEKLY_REPORT=0
APPRISE_NOTIFY_NEW_SPECIES_EACH_DAY=1
APPRISE_MINIMUM_SECONDS_BETWEEN_NOTIFICATIONS_PER_SPECIES=600

#----------------------  Flickr Images API Configuration
-----------------------#
## If FLICKR_API_KEY is set, the web interface will try and display bird
images
## for each detection. If FLICKR_FILTER_EMAIL is set, the images will
only be
## displayed from a particular Flickr user (e.g. yourself).

FLICKR_API_KEY=03453#############################
FLICKR_FILTER_EMAIL=

ehpersonal38, 2023-03-24:
Thanks for that. I've just tested notifications on my system with the same settings and can confirm everything works as expected. Maybe you could try taking `$flickrimage` out of your notification body temporarily to see if that has any effect? If that doesn't work then you can also try enabling "Notify each detection," then checking "View Log" after you get a detection to see if there are any errors thrown by Apprise.

Kouwe-sudo, 2023-03-28:
> Did you have any luck with @ehpersonal38s' idea to see if you could fix this?

Not yet. I didn’t find the time. I think I reinstall Birdnet first. Perhaps it has something to do with my install of apprise being ‘not on PATH’. Now I realize it was already part of the Birdnet installation.

lloydbayley, 2023-03-29:
Well, that certainly could be. I'd be glad to hear how you go from a new install.
Remember, if you don't want to lose what you've already captured, make a quick backup. :)

raybellis, 2023-04-01:
I've having problems with notifications too.

I'm using mailtos:// and the test works, but when I save the settings they disappear.

lloydbayley, 2023-04-02:
We may as well close this one and can re-open if needed. What do you think @ehpersonal38 ?

Kouwe-sudo, 2023-04-03:
You could do that but the problem isn’t solved. In the last few days I had a total of 1054 discoveries and 21 unique species but not a single notification (with the box: ‘notify every new discovery’ checked). In the log not a single word about apprise.This way Birdnet is not what I hoped it would be.

lloydbayley, 2023-04-03:
Sorry....Hadn't heard for 5 days and you said you were going to re-install. Assumed the re-install worked.

Kouwe-sudo, 2023-04-03:
> No worries, I'll try to help you get to the bottom of this once and for all.
> Could you please edit the file ~/BirdNET-Pi/scripts/utils/notifications.py and add notify("test","test") right underneath print("notfications"). Here's a snippet of the modified code starting from line 206:
> ...
>                         .replace("$week", week)\
>                         .replace("$latitude", latitude)\
>                         .replace("$longitude", longitude)\
>                         .replace("$cutoff", cutoff)\
>                         .replace("$sens", sens)\
>                         .replace("$flickrimage", image_url if "{" in body else >
>                         .replace("$overlap", overlap)\
>                         + " (only seen " + str(int(numberDetections)) + " times>
>                     notify(notify_body, notify_title, image_url)
>                     species_last_notified[comName] = int(timeim.time())
>                 con.close()
>             except sqlite3.Error:
>                 print("Database busy")
>                 time.sleep(2)
>
> if __name__ == "__main__":
>     print("notfications")
>     notify("test","test")
> Once this modification is made, please run the file:
> /home/pi/BirdNET-Pi/birdnet/bin/python3 notifications.py
> Are there any errors from Apprise?

I will try this next week. I’m on holiday now.

Kouwe-sudo, 2023-04-10:
This is what I get after adding the line requested:

ehpersonal38, 2023-04-10:
It looks like your message cut off there and didn't include everything. Can you try again?

Kouwe-sudo, 2023-04-11:
What did you expect? After the rerun I get pretty much the same output. Isn’t that part of the problem or am I doing something completely wrong?

ehpersonal38, 2023-04-11:
Try running this Python3 script, which will try sending a test notification via both the Apprise CLI and the Apprise Python package:
```python
import subprocess
import os.path
import apprise

# Run the command with subprocess
subprocess_cmd = "~/BirdNET-Pi/birdnet/bin/apprise -c ~/BirdNET-Pi/apprise.txt -vv -t 'test' -b 'sent via apprise CLI'"
subprocess.run(subprocess_cmd, shell=True, capture_output=True, text=True)

# Send the notification with apprise
userDir = os.path.expanduser('~')
APPRISE_CONFIG = userDir + '/BirdNET-Pi/apprise.txt'
apobj = apprise.Apprise()
config = apprise.AppriseConfig()
config.add(APPRISE_CONFIG)
apobj.add(config)
apobj.notify(body="sent via apprise python package",title="test")
```

Note you'll need to use 
`~/BirdNET-Pi/birdnet/bin/python3`

Instead of `python3` to run it

Kouwe-sudo, 2023-04-13:
This is what I get (unless I do something wrong. Although I used the standard procedure to load Birdnet on my Pi it seems that the module APPRISE isn't present. But I do get the test-mail.

Traceback (most recent call last):
  File "/home/pi/BirdNET-Pi/birdnet/bin/test.py", line 3, in <module>
    import apprise
ModuleNotFoundError: No module named 'apprise'

ehpersonal38, 2023-04-13:
Try installing Apprise:

`~/BirdNET-Pi/birdnet/bin/pip3 install apprise`

Kouwe-sudo, 2023-04-14:
After adding apprise with pip3 install apprise I get this:

  WARNING: The script markdown_py is installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script apprise is installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.

ehpersonal38, 2023-04-14:
Interesting. To be clear you should use the command

`~/BirdNET-Pi/birdnet/bin/pip3 install apprise`

NOT

`pip3 install apprise`.

As long as you did the first command, then I think you should be good to go. Can you test to see if notifications work? You can try running that `test.py` script we created earlier with this command:

`~/BirdNET-Pi/birdnet/bin/python3 ~/BirdNET-Pi/birdnet/bin/test.py` 
(I read above that `~/BirdNET-Pi/birdnet/bin/test.py` is where you saved the script, change the location if necessary)

Kouwe-sudo, 2023-04-21:
Due to a lack of time, I stop trying to figure out why I am not getting notifications from Birdnet. Maybe I'll try again next spring. May I thank you very much for the time and effort to uncover the cause of the problem. For now you can close the issue.

closed 2023-04-21 by ehpersonal38