mcguirepr89/BirdNET-Pi #805: Incorrect Time

StrontiumSteel, 2023-03-26:
When setting the time from within BirdNetPi it is exaclt an hour out. I suspect an issue with the Australia/Darwin timezone within the app. I say that, as when I set it from the command-line with "sudo raspi-config" it is fine. Unfortunately everytime I have to change a setting I have to re-do the time via the command prompt

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Tools"
2. Click on 'Settings'
3. Scroll down to 'Time and Date'
4. Select Timezone Austalia/Darwin
5. Time is one hour out

**Expected behavior**
I would expect the time to be correct. 

**Screenshots**
Not required

**Additional context**
Brand new build, Latest Pi software. 

**Your build**
Pi 4

**Code or log snippets**
```
Paste your 
log output 
or 
code here
```

lloydbayley, 2023-03-26:
Question. Do you have the box ticked to set the time and date from the internet?

StrontiumSteel, 2023-03-26:
Yes I have that box ticked. 
timedatectl
               Local time: Sun 2023-03-26 21:42:27 ACST
           Universal time: Sun 2023-03-26 12:12:27 UTC
                 RTC time: n/a
                Time zone: Australia/Darwin (ACST, +0930)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no

[image]

For some reason the time change isn't taking from settings right now, but the above image shows that it thinks it should be an hour in the future. (if it pasted correctly)

StrontiumSteel, 2023-03-26:
Hmm.. not consistent. Sorry, for some reason it's not repeatable every time. I'll play some more and report back if I can work something out.

lloydbayley, 2023-03-26:
Well that's interesting...Hate the inconsistent problems! Makes debugging bloody harder! :)
Good luck in the hunt!

StrontiumSteel, 2023-03-26:
It's consistently showing exactly an hour in the future, and _sometimes_ pushing the system clock there too. For eg, right now: 
timedatectl
               Local time: Sun 2023-03-26 21:50:58 ACST
           Universal time: Sun 2023-03-26 12:20:58 UTC
                 RTC time: n/a
                Time zone: Australia/Darwin (ACST, +0930)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
[image]

lloydbayley, 2023-03-26:
Hmm....when you save settings in BirdNet-Pi, it will update the system data/time as that's what it's running on. There isn't a separate BirdNet-Pi date/time stored so that makes it all the more interesting that it only does it sometimes. You have a play, I'll have a think.

StrontiumSteel, 2023-03-26:
All good, and thanks for the help. The fact that the GUI is consistently showing an hour in the future makes me think it's looking somewhere wrong for it's timezone lookup table. But as I'm the only Australia/Darwin active user I guess it's just bothering me :-)

ehpersonal38, 2023-03-27:
Okay so I think I've fixed this in the latest commit. The issue was that the displayed time wasn't taking into account timezones properly. So I switched the date() object with a DateTime one, which should help. Though this won't impact the backend setting ot the time at all, so I don't understand why sometimes your system time would be set to the incorrect time too. All of that is handled by the built-in linux time management utilities.

Also if anyone wants a good laugh: https://www.youtube.com/watch?v=-5wpm-gesOY 🙂

StrontiumSteel, 2023-03-27:
With a little more testing it seems the time was only set wrong on initial install. That could have been down to user error. The time displaying wrong in the GUI was just adding to my confusion. Thanks for your help

lloydbayley, 2023-03-28:
Well I'm glad it's resolved. I must say I have been guilty of 'the problem is between the chair and the keyboard' before. :)