mcguirepr89/BirdNET-Pi #819: Rebooted to change my pi case. When plugged back in I cannot access birdnetpi.local

Billium28, 2023-04-03:
I got my Libre Renegade installed with birdnet using the portability tool. I got it running and wanted to put my pi in a new better case. I shutdown in the settings tab and did the case change over. I went to plug the pi back in and I do not see the UI at the url. I tried the IP address as well but no page. I do vaguely remember having to do something to auto boot this app when I played around w an older slower pi I had a couple months ago. But I'm not sure.

Is there a command I can type to get it to boot up by ssh?

ehpersonal38, 2023-04-03:
Interesting - can you check if the caddy webserver is running?
`sudo service caddy status`

lloydbayley, 2023-04-06:
.

lloydbayley, 2023-04-13:
?

closed 2023-04-16 by ehpersonal38