mcguirepr89/BirdNET-Pi #833: Permission denied when trying to update

StrontiumSteel, 2023-04-08:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
Permission denied error when pressing the update button

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'tools'
2. Click on 'System Controls'
3. Select 'Update'  (currently shows 2203 things need updated)
4. Error: 'sh: 1: update_birdnet.sh: Permission denied'

**Expected behavior**
Up until a few days ago it simply updated fine. No permissions error

**Your build**
Pi 4

StrontiumSteel, 2023-04-08:
I've noticed that there were no execute permissions on the update_birdnet.sh file in scripts directory. I've done chmod a+x on this file and re-tried. It is now working, however as this only runs birdnet something else had removed these permissions. Posted here in case someone else has the same issue and it can help them in the future.

ehpersonal38, 2023-04-08:
Unfortunately the permissions of some of the .sh files got changed in a recent update. I've gone and corrected the issue but anyone facing it should manually run the commands above ^

lloydbayley, 2023-04-08:
Yep!
```
chmod a+x ~/BirdNET-Pi/scripts/update_birdnet.sh 
```
fixed it for me too.

Hope it wasn't anything I did in my PR or testing! I was very careful...

ehpersonal38, 2023-04-08:
@lloydbayley No worries, it wasn't you! It was this commit:https://github.com/mcguirepr89/BirdNET-Pi/commit/99c308827bff28761f5a4cf4717e5308733397c7

Which is really annoying because that was created automatically by my Git desktop client when I went to push some changes from a PR. There was never any indication it was randomly changing file perms... and as for the 2200+ updates, I don't know the cause of that right now. I figure changing the perms on an sh file should be completely unrelated to the PHP that fetches the git "# of commits behind by" status. 😖

lloydbayley, 2023-04-08:
Blimey! I'll keep an eye on my Github Desktop client now in case something similar happens in a PR!
Why don't we just say that the 2200+ never happened? :) Much easier altogether.

oliverk71, 2023-04-10:
Got the same error, but additionally cannot login anymore.

StrontiumSteel, 2023-04-10:
> Got the same error, but additionally cannot login anymore.

Weird. Can you still get to the web page?

oliverk71, 2023-04-10:
To be more exactly: I cannot login via SSH using my credentials, anymore. Server says: login incorrect. 
The web page is reachable and birdnetpi is running, I can also access tools and settings with birdnet user.

lloydbayley, 2023-04-10:
Hmm...Can you access the Web Terminal with the credentials?

oliverk71, 2023-04-10:
I tried my SSH agent app (Juice) where the credentials were stored. I also tried to login via web terminal and console. 
But wait ... I remember I changed user password for another pi (at least I was believing) last week. I should first check that.

lloydbayley, 2023-04-10:
I've made that mistake before...but don't tell anyone...especially don't put it on the internet! :)

oliverk71, 2023-04-10:
Okay, I asked the friend to whom I gave the pi to login into my birdnetpi via web terminal using his password. Did not work, so most probably, I did not change the pw of the wrong pi. But he will check the pi I gave him, just to make sure he can login. If he can, the reason must be something else. 

Until I found a solution I cannot change permissions and update.

lloydbayley, 2023-04-10:
That's really weird. The update 'unpleasantness' wouldn't have had any effect on login credentials. I'm struggling to work out what has gone on in your case. I don't mean to seem dismissive at all but I have a feeling that something has happened on your end coincidentally. That always makes things difficult to debug but I'll try to help where I can!

oliverk71, 2023-04-10:
Is it maybe possible that the keyboard locale was changed by commit [99c3088](https://github.com/mcguirepr89/BirdNET-Pi/commit/99c308827bff28761f5a4cf4717e5308733397c7)?

lloydbayley, 2023-04-10:
I highly doubt it unless it triggered something unexpected and unforseen but then surely we'd be seeing more reports?

birdsval, 2023-04-10:
If you have another linux-machine available, you can mount your sd-card there and change the permission that way. That worked for me (I had forgotten my password …). 

After that the update was possible again.

lloydbayley, 2023-04-12:
@oliverk71 How did you go? Did you manage to get into the Pi or put the card in another machine and edit the persimmons there?

oliverk71, 2023-04-14:
@lloydbayley Sorry for late answer, I was not at home for a few days. Well, it turned out that everything was absolutely okay, but I mixed the passwords of my pis. And instead of finding the nearest solution quickly, I focussed on update issue and the fact I configured another pi and maybe mixed them (what was not the case). Hope, I learnt the lesson. 

After I managed to login with correct credentials, I just changed permissions like recommended above and was able to update the pi, again.

lloydbayley, 2023-04-15:
Ahh well that's marvellous. I'm glad you worked it out and it didn't turn out to be something ominous because, frankly, I was a bit stumped as to where to go from there, so was secretly hoping it was something like that. Excellent news. 
As for the lesson, we are ALWAYS learning. It never stops. It's a good thing. (As long as you don't press a big red button under a desk and wonder what it might do when in a secure facility.....) 😄

lloydbayley, 2023-04-16:
C'est Fini!

closed 2023-04-25 by ehpersonal38