mcguirepr89/BirdNET-Pi #844: ImportError: cannot import name 'markdown' from 'markdown'

LeromX, 2023-04-11:
Now i can see the following error messages in the log:
08:29:33---ImportError: cannot import name 'markdown' from 'markdown' (BirdNET-Pi/birdnet/lib/python3.9/site-packages/markdown/__init__.py)
08:29:53---Traceback (most recent call last):
08:29:53---  File "/usr/local/bin/server.py", line 18, in <module>
08:29:53---    from utils.notifications import sendAppriseNotifications
08:29:53---  File "BirdNET-Pi/scripts/utils/notifications.py", line 1, in <module>
08:29:53---    import apprise
08:29:53---  File "BirdNET-Pi/birdnet/lib/python3.9/site-packages/apprise/__init__.py", line 55, in <module>
08:29:53---    from .Apprise import Apprise
08:29:53---  File "BirdNET-Pi/birdnet/lib/python3.9/site-packages/apprise/Apprise.py", line 29, in <module>
08:29:53---    from .conversion import convert_between
08:29:53---  File "BirdNET-Pi/birdnet/lib/python3.9/site-packages/apprise/conversion.py", line 27, in <module>
08:29:53---    from markdown import markdown

BR
Henrik

_Originally posted by @LeromX in https://github.com/mcguirepr89/BirdNET-Pi/discussions/839#discussioncomment-5576931_

ehpersonal38, 2023-04-11:
For every "cannot import" Python error you see, you can manually install each package by running the following command, replacing <package name> with the actual name of the package.

`~/BirdNET-Pi/birdnet/bin/pip3 install <package name>`

So in this particular case you will run the command:
`~/BirdNET-Pi/birdnet/bin/pip3 install markdown`

Though really I feel like at this point we're playing wack-a-mole with pip packages. What really is the root cause of all of this? Markdown should have been installed at the same time you installed your `requirements.txt`, since it's a prerequisite of Apprise. Something is definitely awry with your system, but I guess it won't hurt to play wack-a-mole for a while and see if your BirdNET-Pi is somehow functional afterwards.

LeromX, 2023-04-12:
Im getting this:
birdnetpi:~/BirdNET-Pi$ ~/BirdNET-Pi/birdnet/bin/pip3 install markdown
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: markdown in ./birdnet/lib/python3.9/site-packages (3.4.3)

As You mentioned, ill probably just try to re-image my microsd-card

lloydbayley, 2023-04-12:
Raspberry Pi 3 or 4, the error you are getting has no bearing on that. It would seem that there is something fundamentally wrong with your install, as the packages MUST be installed by the install as it's part of the requirements.txt, so I think it would be best to re-image. Don't forget to backup if you have an established database and exclusion list!

LeromX, 2023-04-12:
Alright,,, now its all working.
I identified two issues with my previous setup.
1. I had insufficient power so i got some low voltage warnings.
2. I did not create a big Swapfile (since I run on a Pi 3 i need a Swapfile due to lower memory).

Now it all works after i re-installed from scratch.
1. Install Pi OS
2. Create a Swapfile (i made one 2048 instead of 100).
3. Install BirdNET-Pi

lloydbayley, 2023-04-12:
Weird. Again, I wouldn't have thought low voltage would cause package problems but who knows, perhaps there was some corruption...Glad you got it sorted.

closed 2023-04-13 by ehpersonal38