mcguirepr89/BirdNET-Pi #855: RTSP stream does not work

Ulflindberg, 2023-04-15:
**Describe the bug**
As soon as I add a RTSP stream to be the source birdnet pi stops working. 
**To Reproduce**
Steps to reproduce the behavior:
I remove the RTSP stream and start the log and I can see that the Birdnet works. Adding the RTSP stream again and opens the log and I can see in the log that RTPS stream is addad but then nothing happens.

**Expected behavior**
My earlier installation worked very well with RTSP stream but after update it does not work
**Screenshots**
[image]

**Additional context**
Add any other context about the problem or your installation here.

**Your build**
Raspberrypi 3

**Code or log snippets**
```
Paste your 
log output 
or 
code here
```

jaredb7, 2023-04-16:
Hi @Ulflindberg 

This wouldn't be a RTSP stream from a Ubiquiti camera would it? (just noticed the port is all), at camera identifier hash thing at the end). 
If so they seem to have 2 audio streams which confuses the recording system after it updated to enable simultaneous RTSP recording... it doesn't know what audio to stream to use. It worked before because we weren't playing with output mapping.

Can you try running this ffprobe command via SSH or via Tools -> Web Terminal, just double check the URL as I just typed it from your screenshot.

`ffprobe -hide_banner -show_streams -of json  rtsp://192.168.4.1:7447/YBSisVM3cgwjElyq`

From the output, check if there are 2 audio streams, example screenshot below
[image]

If so, you'll need a workaround to get recording going again, can you edit the birdnet_recording.sh file used by the bird using Tools -> File Manager, then browse to **/ pi / BirdNET-Pi / scripts**
[image]

Find birdnet_recording.sh, edit it and add `:a:0` after `${MAP_ID}` ... so it becomes `${MAP_ID}:a:0`.
Save and restart the Recording service under Tools -> Services
Recording
[image]

Let me know how you go.

Ulflindberg, 2023-04-16:
Hi @jaredb7 ,

Yes it is a Ubiquiti camera and I have edited the file as you suggested and now it seems to be running. I have just restarded so I will let it run and see that everything works before celebrating. Thanks a lot for your help.

Ulf

Ulflindberg, 2023-04-16:
It has now been running for a few hours and works really good.

closed 2023-04-16 by Ulflindberg

jaredb7, 2023-04-16:
Awesome great to hear it's working :) , we hope to have that fix implemented properly soon so it'll just work.