mcguirepr89/BirdNET-Pi #856: Suggestions for monitoring recording status?

apawloski, 2023-04-15:
First of all - kudos for a great project. This is my second season using it and am having a lot of fun with it.

I don't use check in on this app everyday, but occasionally I'll visit my instance and notice that it has not been recording or analyzing for several days. I think this is an issue with the actual recording since the timestamp at the spectrogram on the main page is stale. 

I'm not sure _why_ this happens (possibly related to https://github.com/mcguirepr89/BirdNET-Pi/issues/393 ?), but frankly since a simple restart fixes it (though sometimes it takes a few tries), I'm not very motivated to debug that thoroughly.

Instead, since this is fixed by restarts, I'd like to set up a monitoring script that will notify me when recording has stopped (or maybe just restarts automatically) when recording/analyzing breaks. 

Does anybody have any advice on what I can monitor to verify that recording is happening within BirdNET-Pi?

lloydbayley, 2023-04-15:
First of all, a quick-question. Are you completely up-to date with the latest software? This was a problem before that a fix was deployed for and that may be the case if you aren't up to date.

apawloski, 2023-04-15:
That's a good question. Where can I find that? The git log shows it's at a commit from yesterday https://github.com/mcguirepr89/BirdNET-Pi/commit/2be0b394ee64078e94abfb37a5feb2a1f1a8c9a7, but I don't remember updating yesterday.

ehpersonal38, 2023-04-15:
If git log shows that then you I think you must've updated within the last day. But the fix Lloyd is talking about was done about a month ago, so as long as you updated since then we can rule that bug out as being the culprit.

I think to keep the solution simple here, as you said you're not interested in debugging anything, you can just add the following line to your crontab which will restart the BirdNET server and client every hour:
`0 * * * * sudo service birdnet_server restart && sudo service birdnet_analysis restart`

apawloski, 2023-04-16:
Thanks. The only issue with a cronjob is that the reboot solution doesn't always work for me after one restart (sometimes it does), but eventually it will work after a few tries. 

I'm comfortable writing a script or emitting metrics to something that can alert me, but I guess really I'm asking your thoughts on what I can use as an indicator that recording is happening successfully. For example, are they saved to the db? Maybe I can run a query for the latest and compare it to current time.

ehpersonal38, 2023-04-16:
Sure, the database lies at `~/BirdNET-Pi/scripts/birds.db`, and it has two columns `Date` and `Time`. The table is called `detections`. An example entry is something like:
```
Date | Time
2022-04-17 | 11:14:20
```

lloydbayley, 2023-04-20:
Perhaps this could become a discussion rather than an issue. Thoughts?

apawloski, 2023-04-20:
Yes, happy to move there. Thanks.

closed 2023-04-20 by apawloski