mcguirepr89/BirdNET-Pi #858: Processing Backlog of Audio on Fresh install

StevenBrehmer, 2023-04-16:
Hi There - I had previously been running birdnet for a few months with no issue. 
As of early April - I was not showing any birds being identified... I finally caved and reformatted the sd card for a fresh install. 
I was able to install it successfully - but  I have a message at the to pof the overview page saying...
"Your system is currently processing a backlog of audio. This can take several hours before normal functionality of your BirdNET-Pi resumes."
I have added my rstp stream back in (streaming from ubiquiti camera) - and I can hear the live audio when i am on the spectrogram page, but It does not seem to be making it to the analyzer. 

Any advice would be greatly appreciated. Big fan of the birdnet-pi :)

ehpersonal38, 2023-04-16:
Hi, could you please provide the output of running the following diagnostic command in a terminal window of your BirdNET-Pi?

```
bash ~/BirdNET-Pi/scripts/print_diagnostic_info.sh
```

StevenBrehmer, 2023-04-16:
sorry its pretty long
`========== caddy status ==========
● caddy.service - Caddy
     Loaded: loaded (/lib/systemd/system/caddy.service; enabled; vendor preset: enabled)
    Drop-In: /etc/systemd/system/caddy.service.d
             └─override.conf
     Active: active (running) since Sun 2023-04-16 13:51:06 EDT; 2h 18min ago
       Docs: https://caddyserver.com/docs/
   Main PID: 658 (caddy)
      Tasks: 10 (limit: 3933)
        CPU: 6.519s
     CGroup: /system.slice/caddy.service
             └─658 /usr/bin/caddy run --environ --config /etc/caddy/Caddyfile

Apr 16 13:51:06 birdnetpi caddy[658]: {"level":"info","ts":1681667466.1974077,"logger":"admin","msg":"admin endpoint started","address":"localhost:2019","enforce_origin":false,"origins":["//localhost:2019","//[::1]:2019","//127.0.0.1:2019"]}
Apr 16 13:51:06 birdnetpi caddy[658]: {"level":"info","ts":1681667466.213905,"logger":"tls.cache.maintenance","msg":"started background certificate maintenance","cache":"0x40003eae00"}
Apr 16 13:51:06 birdnetpi caddy[658]: {"level":"warn","ts":1681667466.214478,"logger":"http","msg":"server is listening only on the HTTP port, so no automatic HTTPS will be applied to this server","server_name":"srv0","http_port":80}
Apr 16 13:51:06 birdnetpi caddy[658]: {"level":"info","ts":1681667466.2556129,"logger":"tls","msg":"cleaning storage unit","description":"FileStorage:/var/lib/caddy/.local/share/caddy"}
Apr 16 13:51:06 birdnetpi caddy[658]: {"level":"info","ts":1681667466.2559638,"logger":"tls","msg":"finished cleaning storage units"}
Apr 16 13:51:06 birdnetpi caddy[658]: {"level":"info","ts":1681667466.2562,"logger":"http.log","msg":"server running","name":"srv0","protocols":["h1","h2","h3"]}
Apr 16 13:51:06 birdnetpi caddy[658]: {"level":"info","ts":1681667466.2687037,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}
Apr 16 13:51:06 birdnetpi caddy[658]: {"level":"info","ts":1681667466.2693558,"msg":"serving initial configuration"}
Apr 16 13:51:06 birdnetpi systemd[1]: Started Caddy.
Apr 16 14:06:01 birdnetpi caddy[658]: {"level":"error","ts":1681668361.5735664,"logger":"http.handlers.reverse_proxy","msg":"aborting with incomplete response","error":"write tcp xxx.xxx.xxx.176:80->xxx.xxx.xxx.234:59424: write: broken pipe"}

========== birdnet_analysis status ==========
● birdnet_analysis.service - BirdNET Analysis
     Loaded: loaded (/home/USERNAME/BirdNET-Pi/templates/birdnet_analysis.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) since Sun 2023-04-16 16:09:42 EDT; 305ms ago
    Process: 166304 ExecStart=/usr/local/bin/birdnet_analysis.sh (code=exited, status=0/SUCCESS)
   Main PID: 166304 (code=exited, status=0/SUCCESS)
        CPU: 287ms

========== birdnet_log status ==========
● birdnet_log.service - BirdNET Analysis Log
     Loaded: loaded (/home/USERNAME/BirdNET-Pi/templates/birdnet_log.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2023-04-16 14:32:32 EDT; 1h 37min ago
   Main PID: 49508 (gotty)
      Tasks: 6 (limit: 3933)
        CPU: 25ms
     CGroup: /system.slice/birdnet_log.service
             └─49508 /usr/local/bin/gotty --address localhost -p 8080 -P log --title-format BirdNET-Pi Log birdnet_log.sh

Apr 16 14:32:32 birdnetpi systemd[1]: Started BirdNET Analysis Log.
Apr 16 14:32:32 birdnetpi gotty[49508]: 2023/04/16 14:32:32 Loading config file at: /home/USERNAME/.gotty
Apr 16 14:32:32 birdnetpi gotty[49508]: 2023/04/16 14:32:32 GoTTY is starting with command: birdnet_log.sh
Apr 16 14:32:32 birdnetpi gotty[49508]: 2023/04/16 14:32:32 HTTP server is listening at: http://127.0.0.1:8080/log/

========== birdnet_recording status ==========
● birdnet_recording.service - BirdNET Recording
     Loaded: loaded (/home/USERNAME/BirdNET-Pi/templates/birdnet_recording.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2023-04-16 14:32:32 EDT; 1h 37min ago
   Main PID: 49495 (bash)
      Tasks: 2 (limit: 3933)
        CPU: 15min 59.422s
     CGroup: /system.slice/birdnet_recording.service
             ├─ 49495 bash /usr/local/bin/birdnet_recording.sh
             └─166303 ffmpeg -hide_banner -loglevel error -nostdin -vn -thread_queue_size 512 -i rtsps://xxx.xxx.xxx.xxx.1:7441/05drAiNfB5JIPFr5?enableSrtp -map 0 -t 30 -acodec pcm_s16le -ac 2 -ar 48000 file:/home/USERNAME/BirdSongs/StreamData/2023-04-16-birdnet-RTSP_1-16:09:40.wav

Apr 16 16:09:31 birdnetpi birdnet_recording.sh[166054]: Error initializing output stream 0:0 --
Apr 16 16:09:34 birdnetpi birdnet_recording.sh[166122]: [wav @ 0x55ca5f4430] WAVE files have exactly one stream
Apr 16 16:09:34 birdnetpi birdnet_recording.sh[166122]: Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
Apr 16 16:09:34 birdnetpi birdnet_recording.sh[166122]: Error initializing output stream 0:1 --
Apr 16 16:09:37 birdnetpi birdnet_recording.sh[166176]: [wav @ 0x55638a2890] WAVE files have exactly one stream
Apr 16 16:09:37 birdnetpi birdnet_recording.sh[166176]: Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
Apr 16 16:09:37 birdnetpi birdnet_recording.sh[166176]: Error initializing output stream 0:0 --
Apr 16 16:09:40 birdnetpi birdnet_recording.sh[166237]: [wav @ 0x55a6141430] WAVE files have exactly one stream
[... cut ...]
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
usbstream:CARD=Headphones
    bcm2835 Headphones
    USB Stream Output
usbstream:CARD=vc4hdmi0
    vc4-hdmi-0
    USB Stream Output
usbstream:CARD=vc4hdmi1
    vc4-hdmi-1
    USB Stream Output

========= Date and Time ==========
Sun 16 Apr 2023 04:09:45 PM EDT
===========================================
`

ehpersonal38, 2023-04-16:
I see a few issues with your system - `throttled=0x50005` indicates that your Pi isn't getting enough voltage due to an incompatible/cheap power supply. The official Raspberry Pi 4B power supply will solve this issue, or you can look at the specs to see if you have one that's similar: https://www.cytron.io/p-official-rpi-15w-5v-3a-psu-usb-c-us-plug-white

Now as for these errors:
```bash
========== birdnet_recording status ==========
● birdnet_recording.service - BirdNET Recording
Loaded: loaded (/home/USERNAME/BirdNET-Pi/templates/birdnet_recording.service; enabled; vendor preset: enabled)
Active: active (running) since Sun 2023-04-16 14:32:32 EDT; 1h 37min ago
Main PID: 49495 (bash)
Tasks: 2 (limit: 3933)
CPU: 15min 59.422s
CGroup: /system.slice/birdnet_recording.service
├─ 49495 bash /usr/local/bin/birdnet_recording.sh
└─166303 ffmpeg -hide_banner -loglevel error -nostdin -vn -thread_queue_size 512 -i rtsps://xxx.xxx.xxx.xxx.1:7441/05drAiNfB5JIPFr5?enableSrtp -map 0 -t 30 -acodec pcm_s16le -ac 2 -ar 48000 file:/home/USERNAME/BirdSongs/StreamData/2023-04-16-birdnet-RTSP_1-16:09:40.wav

Apr 16 16:09:31 birdnetpi birdnet_recording.sh[166054]: Error initializing output stream 0:0 --
Apr 16 16:09:34 birdnetpi birdnet_recording.sh[166122]: [wav @ 0x55ca5f4430] WAVE files have exactly one stream
Apr 16 16:09:34 birdnetpi birdnet_recording.sh[166122]: Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
Apr 16 16:09:34 birdnetpi birdnet_recording.sh[166122]: Error initializing output stream 0:1 --
Apr 16 16:09:37 birdnetpi birdnet_recording.sh[166176]: [wav @ 0x55638a2890] WAVE files have exactly one stream
Apr 16 16:09:37 birdnetpi birdnet_recording.sh[166176]: Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
Apr 16 16:09:37 birdnetpi birdnet_recording.sh[166176]: Error initializing output stream 0:0 --
Apr 16 16:09:40 birdnetpi birdnet_recording.sh[166237]: [wav @ 0x55a6141430] WAVE files have exactly one stream
Apr 16 16:09:40 birdnetpi birdnet_recording.sh[166237]: Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
Apr 16 16:09:40 birdnetpi birdnet_recording.sh[166237]: Error initializing output stream 0:0 --
```

@jaredb7 are these fixed by the PR you made (and I just merged in)?

jaredb7, 2023-04-16:
@ehpersonal38 yep that's the same error reported previously and fixed in PR you merged in with the `${MAP_ID}:a:0` change.

Also long and or cheap USB C leads/extension leads can also contribute to the voltage drop issue too they're using an appropriate power supply. Just food for thought anyway, but  the recording issue will get Steven back and operating to some level.

StevenBrehmer, 2023-04-17:
Thanks Guys - it is the one that came with the pi - but ive had this for a few years. maybe its starting to go ill swap out and update my system. 

Just updated the system and Its back working- thank you so much for the support!!!