mcguirepr89/BirdNET-Pi #863: Analysis stops and needs many reboots

gerritzen, 2023-04-19:
**Describe the bug**
Today and yesterday, the analyzer stopped working around 5:30 when more and more detections start coming in. Restarting `birdnet_analysis.service` triggers the analysis of one recording and one recording only. 

**Expected behavior**
Automatic processing of all recordings with little to no backlog building up. 

**Your build**
Raspberry Pi 4, 8GB, clean install from 3 weeks ago, up to date. 

**Code or log snippets**
```
Apr 19 08:08:29 birdnetpi systemd[1]: Started BirdNET Analysis.
Apr 19 08:09:15 birdnetpi birdnet_analysis.sh[403]: Files loaded
Apr 19 08:09:15 birdnetpi birdnet_analysis.sh[403]: RECORDING_LENGTH set to 15
Apr 19 08:09:15 birdnetpi birdnet_analysis.sh[403]: /home/pi/BirdNET-Pi/birdnet/bin/python3 /home/pi/BirdNET-Pi/scripts/analyze.py --i /home/pi/BirdSongs/April-2023/19-Wednesday/2023-04-19-birdnet-07:07:28.wav --o /home/pi/BirdSongs/April-2023/19-Wednesday/2023-04-19-birdnet-07:07:28.wav.csv --lat 0.XX --lon 0.XX --week 15 --overlap 0.0 --sensitivity 1.25 --min_conf 0.6 --birdweather_id "IN_USE"
Apr 19 08:09:19 birdnetpi birdnet_analysis.sh[1481]: 0.0;3.0-('Fringilla coelebs_Buchfink', 0.47220695)
Apr 19 08:09:19 birdnetpi birdnet_analysis.sh[1481]: 3.0;6.0-('Fringilla coelebs_Buchfink', 0.1853611)
Apr 19 08:09:19 birdnetpi birdnet_analysis.sh[1481]: 6.0;9.0-('Parus major_Kohlmeise', 0.34171224)
Apr 19 08:09:19 birdnetpi birdnet_analysis.sh[1481]: 9.0;12.0-('Sylvia atricapilla_Mönchsgrasmücke', 0.14290392)
Apr 19 08:09:19 birdnetpi birdnet_analysis.sh[1481]: 12.0;15.0-('Ficedula hypoleuca_Trauerschnäpper', 0.21519433)
Apr 19 08:09:19 birdnetpi birdnet_analysis.sh[403]: RECORDING_LENGTH set to 15
Apr 19 08:09:19 birdnetpi birdnet_analysis.sh[403]: /home/pi/BirdNET-Pi/birdnet/bin/python3 /home/pi/BirdNET-Pi/scripts/analyze.py --i /home/pi/BirdSongs/April-2023/19-Wednesday/2023-04-19-birdnet-07:07:43.wav --o /home/pi/BirdSongs/April-2023/19-Wednesday/2023-04-19-birdnet-07:07:43.wav.csv --lat 0.XX --lon 0.XX --week 15 --overlap 0.0 --sensitivity 1.25 --min_conf 0.6 --birdweather_id "IN_USE"
```

After this it just stops (location replaced with 0). This was taken around 08:20am, so there had been no output for 10 minutes already.

oldgristmill, 2023-05-27:
I was having the same problem on an absolutely new install this morning, and clearing my BirdWeatherID seems to have solved the problem.

dorsetwurzel, 2023-06-10:
I had the same problem. Cleanminstall only 2 days old. Same time of day (as dawn chorus gets underway) analysis seems to stop. As above, removed the BirdweatherID and analysis of recordings resumes

avaryx, 2023-08-08:
I, too, was having this issue, and, too, found it was improved when I removed my BirdWeather ID. Using [glances](https://nicolargo.github.io/glances/), I found that my iowait was spiking, often times over 90%. This told me it was caused by my internal SD card. More on that in a bit.

First off, I'm on a Raspberry Pi 4B, as seen here:
```
OS: Debian GNU/Linux 11 (bullseye) aarch64
Host: Raspberry Pi 4 Model B Rev 1.4
Kernel: 6.1.21-v8+
Uptime: 2 hours, 31 mins
Packages: 835 (dpkg)
Shell: bash 5.1.4
Terminal: /dev/pts/0
CPU: BCM2835 (4) @ 1.800GHz
Memory: 584MiB / 7812MiB
```

The SD card I was using was a [SanDisk Ultra 64GB](https://www.amazon.com/SanDisk-Ultra-microSDXC-Memory-Adapter/dp/B073JYVKNX?th=1). Knowing this wasn't the fastest card offered, I cloned it onto a [SanDisk Extreme 64GB](https://www.amazon.com/SanDisk-Extreme-microSD-UHS-I-Adapter/dp/B07FCMBLV6) and replaced it in the device. I then ran for 24 hours without BirdWeather ID with no issues. I then put my my BirdWeather ID back in place and it has been operational and reporting for over 24 hours with no issues. Additionally, my iowait rarely goes above 0%.

The differences between the two cards (at a very high level) are:

- SanDisk Ultra line: 10 - 80 MB/s
- SanDisk Extreme line: 20 - 300MB/s

As you can see, the Extreme line does provides much higher overall speeds. Those details were found [on WD's SanDisk site](https://support-en.wd.com/app/answers/detailweb/a_id/38990/~/sandisk-products-transfer-speed).

At this point, considering the health of the device, I want to say my issues stemmed from an SD card that was inadequate for the load being generated by BirdPi due to both audio extraction and recognition, as well as submissions to BirdWeather.