mcguirepr89/BirdNET-Pi #866: Package missing for newest Raspberry Pi OS Lite (64-bit) (version February 21st 2023)

puzzlewolf, 2023-04-19:
First of all, thanks for creating this :bird: :+1:, I plan to set it up as a gift for my dad :)

**Describe the bug**
I tried to run the installation script for BirdNET-Pi on a freshly downloaded and flashed Rasperry Pi OS Lite (64-bit). The installation was unsucessful, with a complaint about missing the `python3-venv` package.

**To Reproduce**
I didn't use Raspberry Pi Imager, so I'm listing my steps for completeness
1. Download [Raspberry Pi OS Lite (64-bit)](https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2023-02-22/2023-02-21-raspios-bullseye-arm64-lite.img.xz). At this time, the current version is "February 21st 2023", sha256 `883eb0006c8841b7950ef69a7bf55f73c2250ecc15e6bf507f39f0d82fa2ea0a`.
2. Flash it to sd-card with `dd`.
3. Boot the pi, setup username and keyboard locale.
4. Use `rpi-config` to configure WIFI and enable SSH.
5. Apply [RPi0W2 Installation Guide](https://github.com/mcguirepr89/BirdNET-Pi/wiki/RPi0W2-Installation-Guide#how-to-install-on-the-raspberry-pi-0-w-2) 
6. Run `apt update` and `apt install vim`
7. Run `curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh | bash`

**Expected behavior**
A succesful installation of BirdNET-Pi.

**Additional context**
After uninstalling (with `/usr/local/bin/uninstall.sh && cd ~ && rm -drf BirdNET-Pi`) and installing the package with  `apt install python3-venv`, the installation and BirdNET-Pi work :tada: 

**Your build**
Raspberry PI 3B (no plus). I know it's probably not powerful enough, it's what I have at the moment.

**Code or log snippets**
Tail of the terminal output of the failed install, the full log was overwritten by the new install before I realized.
```
...
+ '[' -d /home/username/BirdSongs ']'
+ install_birdnet
+ cd /home/username/BirdNET-Pi
+ echo 'Establishing a python virtual environment'
Establishing a python virtual environment
+ python3 -m venv birdnet
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt-get install python3-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/home/username/BirdNET-Pi/birdnet/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']

The installation exited unsuccessfully.
```

lloydbayley, 2023-04-20:
Very weird. I just installed it here and it went fine. Was hoping it wasn't something weird in the latest updates.
I don't have a Pi3B but I was using a 3B+ to test.

The only other thing I didn't do was the apt update (installer will do that) and the install vim.
Was there anything else you did that you didn't list?

Also, is there only one user on the pi and if not, is the first user the one you're running it as?
There are sometimes 'unpleasantness-es' if not running as user 1000:1000 but I doubt whether that's the case here.

puzzlewolf, 2023-04-21:
Thanks for checking, @lloydbayley and @ehpersonal38!

The installation log would be helpful, but it's not there because it was overwritten by the second attempt, so I looked at apt's logs in `/var/log/apt/term.log` and `/var/log/apt/history.log`. Here's the entire `history.log`:
```
Start-Date: 2023-04-19  15:46:18
Commandline: apt install vim
Requested-By: vancitronen (1000)
Install: vim:arm64 (2:8.2.2434-3+deb11u1), libgpm2:arm64 (1.20.7-8, automatic), vim-runtime:arm64 (2:8.2.2434-3+deb11u1, automatic)
End-Date: 2023-04-19  15:47:01

Start-Date: 2023-04-19  15:55:39
Commandline: apt -y install git jq
Requested-By: vancitronen (1000)
Install: libonig5:arm64 (6.9.6-1.1, automatic), git:arm64 (1:2.30.2-1+deb11u2), liberror-perl:arm64 (0.17029-1, automatic), jq:arm64 (1.6-2.1), libjq1:arm64 (1.6-2.1, automatic), git-man:arm64 (1:2.30.2-1+deb11u2, automatic)
End-Date: 2023-04-19  15:56:19

Start-Date: 2023-04-19  15:58:24
Commandline: apt install -y debian-keyring debian-archive-keyring apt-transport-https
Requested-By: vancitronen (1000)
Install: apt-transport-https:arm64 (2.2.4), debian-keyring:arm64 (2021.07.26)
End-Date: 2023-04-19  15:58:38

Start-Date: 2023-04-19  15:59:37
Commandline: apt -qqy upgrade
Requested-By: vancitronen (1000)
Upgrade: libcurl4:arm64 (7.74.0-1.3+deb11u5, 7.74.0-1.3+deb11u7), raspberrypi-bootloader:arm64 (1:1.20230106-1, 1:1.20230405-1), libcurl3-gnutls:arm64 (7.74.0-1.3+deb11u5, 7.74.0-1.3+deb11u7), tzdata:arm64 (2021a-1+deb11u8, 2021a-1+deb11u9), libtiff5:arm64 (4.2.0-1+deb11u3, 4.2.0-1+deb11u4), libraspberrypi0:arm64 (1:2+git20220616~133208+6e8f786-1, 1:2+git20230322~143557+9d5250f-1), python3-picamera2:arm64 (0.3.8-1, 0.3.9-1), libraspberrypi-bin:arm64 (1:2+git20220616~133208+6e8f786-1, 1:2+git20230322~143557+9d5250f-1), libraspberrypi-dev:arm64 (1:2+git20220616~133208+6e8f786-1, 1:2+git20230322~143557+9d5250f-1), libraspberrypi-doc:arm64 (1:2+git20220616~133208+6e8f786-1, 1:2+git20230322~143557+9d5250f-1), libcamera-apps-lite:arm64 (0~git20230125+9f08463-1, 0~git20230301+54a781d-1), libcamera0:arm64 (0~git20230124+9b860a66-1, 0~git20230302+923f5d70-1), curl:arm64 (7.74.0-1.3+deb11u5, 7.74.0-1.3+deb11u7), raspberrypi-kernel:arm64 (1:1.20230106-1, 1:1.20230405-1), python3-libcamera:arm64 (0~git20230124+9b860a66-1, 0~git20230302+923f5d70-1), python3-v4l2:arm64 (0.3.1-1, 0.3.2-1), raspberrypi-sys-mods:arm64 (20221019, 20230329), linux-libc-dev:arm64 (1:1.20230106-1, 1:1.20230405-1)
End-Date: 2023-04-19  16:02:12

Start-Date: 2023-04-19  16:17:18
Commandline: apt install python3-venv
Requested-By: vancitronen (1000)
Install: python3-lib2to3:arm64 (3.9.2-1, automatic), python3-venv:arm64 (3.9.2-3), python-pip-whl:arm64 (20.3.4-4+rpt1+deb11u1, automatic), python3.9-venv:arm64 (3.9.2-1, automatic), python3-distutils:arm64 (3.9.2-1, automatic)
End-Date: 2023-04-19  16:17:24

Start-Date: 2023-04-19  16:25:39
Commandline: apt install -qqy caddy ftpd sqlite3 php-sqlite3 alsa-utils pulseaudio avahi-utils sox libsox-fmt-mp3 php php-fpm php-curl php-xml php-zip icecast2 swig ffmpeg wget unzip curl cmake make bc libjpeg-dev zlib1g-dev python3-dev python3-pip python3-venv lsof
Requested-By: vancitronen (1000)
Install: fontconfig:arm64 (2.13.1-4.2, automatic), libvulkan1:arm64 (1.2.162.0-1, automatic), libsphinxbase3:arm64 (0.8+5prealpha+1-12, automatic), libxcb-present0:arm64 (1.14-3, automatic), libpulse0:arm64 (14.2-2+rpt1, automatic), libopenal-data:arm64 (1:1.19.1-2, automatic), libserd-0-0:arm64 (0.30.10-2, automatic), libvpx6:arm64 (1.9.0-1, automatic), libcodec2-0.9:arm64 (0.9.2-4, automatic), libxxf86vm1:arm64 (1:1.1.4-1+b2, automatic), libltdl7:arm64 (2.4.6-15, automatic), librubberband2:arm64 (1.9.0-1, automatic), libgdk-pixbuf2.0-bin:arm64 (2.42.2+dfsg-1+deb11u1, automatic), libglx-mesa0:arm64 (20.3.5-1+rpt5+rpi1, automatic), libtwolame0:arm64 (0.4.0-2, automatic), libzvbi-common:arm64 (0.2.35-18, automatic), va-driver-all:arm64 (2.10.0-1, automatic), libjack-jackd2-0:arm64 (1.9.17~dfsg-1, automatic), libdatrie1:arm64 (0.2.13-1, automatic), libthai0:arm64 (0.1.28-3, automatic), libxrender1:arm64 (1:0.9.10-1, automatic), dbus-user-session:arm64 (1.12.24-0+deb11u1, automatic), libpython3-dev:arm64 (3.9.2-3, automatic), libglvnd0:arm64 (1.3.2-1, automatic), php7.4-readline:arm64 (7.4.33-1+deb11u3, automatic), libsensors-config:arm64 (1:3.6.0-7, automatic), php7.4-curl:arm64 (7.4.33-1+deb11u3, automatic), libx11-xcb1:arm64 (2:1.7.2-1, automatic), libxshmfence1:arm64 (1.3-1, automatic), libpango-1.0-0:arm64 (1.46.2-3, automatic), libsox-fmt-mp3:arm64 (14.4.2+git20190427-2+deb11u2), libwavpack1:arm64 (5.4.0-1, automatic), libmpg123-0:arm64 (1.26.4-1, automatic), libpulsedsp:arm64 (14.2-2+rpt1, automatic), libwayland-cursor0:arm64 (1.18.0-2~exp1.1, automatic), php:arm64 (2:7.4+76), libx264-160:arm64 (2:0.160.3011+gitcde9a93-2.1, automatic), libshine3:arm64 (3.1.1-2, automatic), libpgm-5.3-0:arm64 (5.3.128~dfsg-2, automatic), libjpeg-dev:arm64 (1:2.0.6-4), libarchive13:arm64 (3.4.3-2+deb11u1, automatic), avahi-utils:arm64 (0.8-5+deb11u1), libaom0:arm64 (1.0.0.errata1-3, automatic), sox:arm64 (14.4.2+git20190427-2+deb11u2), pulseaudio:arm64 (14.2-2+rpt1), libjpeg62-turbo-dev:arm64 (1:2.0.6-4, automatic), libxcb-dri2-0:arm64 (1.14-3, automatic), libx265-192:arm64 (3.4-2, automatic), libass9:arm64 (1:0.15.0-2, automatic), libxcb-dri3-0:arm64 (1.14-3, automatic), libcairo2:arm64 (1.16.0-5+rpt1, automatic), libgbm1:arm64 (20.3.5-1+rpt5+rpi1, automatic), libaacs0:arm64 (0.9.0-2, autom [... cut ...] fsg-1+deb11u1, automatic)
End-Date: 2023-04-19  16:34:39
```
So, I installed `vim` at 15:46:18 and then started the BirdNET-Pi installation. It installed `git` and `jq` (`newinstaller.sh`, L32) and went on to lines 15 and 18 from `scrips/install_services.sh`. But the installation of the all the packages (line 20) isn't there. `term.log` also doesn't list a response for that command, so it probably wasn't executed. So I guess that leaves line 19? 

Can you think of any other logs that might help?

LeromX, 2023-04-28:
I have a Raspberry 3B running and it has worked for at least a week 24/7 so far.
This is what You have to do:
1. Start from scratch - format the MicroSD card
2. Instal Raspberry OS
3. Create a larger Swap-file, I made mine 2GB - this is due to Pi3 having less memory than Pi4.
4. Proceed with installation as usual

This works for me and here is my Pi3B reporting to  Birdweather: https://app.birdweather.com/stations/925

Good luck!

lloydbayley, 2023-05-01:
Righty ho! What's happening with this issue Wolf Of Puzzles? Did you get any further with your testing etc? 😄

lloydbayley, 2023-05-15:
Stale...

closed 2023-05-15 by ehpersonal38

thomas8240, 2023-05-21:
What is the status of this issue? 
I am running into that sounds like the issue orginally reported by puzzlewolf.
I am running Raspberry Pi 4 Model B Rev 1.5

The tail of the log...
---------------------- Start of tail -----------------------
Establishing a python virtual environment
+ python3 -m venv birdnet
+ source ./birdnet/bin/activate
++ deactivate nondestructive
++ '[' -n '' ']'
++ '[' -n '' ']'
++ '[' -n /usr/bin/bash -o -n '' ']'
++ hash -r
++ '[' -n '' ']'
++ unset VIRTUAL_ENV
++ '[' '!' nondestructive = nondestructive ']'
++ VIRTUAL_ENV=/home/birdnet/BirdNET-Pi/birdnet
++ export VIRTUAL_ENV
++ _OLD_VIRTUAL_PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
++ PATH=/home/birdnet/BirdNET-Pi/birdnet/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
++ export PATH
++ '[' -n '' ']'
++ '[' -z '' ']'
++ _OLD_VIRTUAL_PS1=
++ PS1='(birdnet) '
++ export PS1
++ '[' -n /usr/bin/bash -o -n '' ']'
++ hash -r
+ pip3 install -U -r /home/birdnet/BirdNET-Pi/requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
The installation exited unsuccessfully.
------------------------------------  end of log ------------------

Please advise.,

ehpersonal38, 2023-05-21:
@thomas8240

Please see this related issue, I commented a fix that I think should work if you scroll down

https://github.com/mcguirepr89/BirdNET-Pi/discussions/775

thomas8240, 2023-05-22:
@ehpersonal38,
Thanks for the response.
You said in #775 ....... 
          So I think the issue is that your Python is the wrong version. You're using 3.7.3 when you should be on at least 3.9.

          tflite_runtime-2.6.0-cp39-none-linux_aarch64.whl
‎‎‎‎‎‎‎‎‎‎‎            ‎‎‎                              ^^^ the 39 means it only works with Python 3.9.

I ran the commands
birdnet@birdnetpi1:~/BirdNET-Pi$ ~/BirdNET-Pi/birdnet/bin/python3
**Python 3.9.2 (default, Mar 12 2021, 04:06:34)**
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import platform
>>> platform.architecture()
**('32bit', 'ELF')**
>>>

I looks like that I am using Python 3.9 32 bit version. 

Does the "aarch64" mean that it Python 3.9 should be 64 bit version??

Thanks again,
Thomas8240

ehpersonal38, 2023-05-22:
@thomas8240 Yes, please see this active discussion for people facing this issue.

https://github.com/mcguirepr89/BirdNET-Pi/discussions/847#discussioncomment-5966935