mcguirepr89/BirdNET-Pi #874: Update seems to have issues with git

wokkeltje13, 2023-04-26:
`++ sudo -u pi git -C /home/pi/BirdNET-Pi rev-parse HEAD
++ set +x
+ sudo -u pi git -C /home/pi/BirdNET-Pi reset --hard
HEAD is now at 2be0b394e fixy
+ set +x
+ sudo -u pi git -C /home/pi/BirdNET-Pi fetch origin main
From https://github.com/mcguirepr89/BirdNET-Pi
 * branch                main       -> FETCH_HEAD
error: cannot lock ref 'refs/remotes/origin/main': unable to resolve reference 'refs/remotes/origin/main': reference broken
 ! [new branch]          main       -> origin/main  (unable to update local ref)
+ set +x
+ sudo -u pi git -C /home/pi/BirdNET-Pi switch -C main --track origin/main
warning: ignoring broken ref refs/remotes/origin/main
fatal: invalid reference: origin/main
+ set +x
+ sudo -u pi git -C /home/pi/BirdNET-Pi diff --stat 2be0b394ee64078e94abfb37a5feb2a1f1a8c9a7 HEAD
+ set +x
+ my_dir=/home/pi/BirdNET-Pi/scripts
+ sudo systemctl stop birdnet_server.service
+ sudo systemctl stop birdnet_recording.service
+ services=(chart_viewer.service spectrogram_viewer.service icecast2.service extraction.service birdnet_recording.service birdnet_log.service birdnet_stats.service)
+ for i in "${services[@]}"
+ sudo systemctl restart chart_viewer.service
+ for i in "${services[@]}"
+ sudo systemctl restart spectrogram_viewer.service
+ for i in "${services[@]}"
+ sudo systemctl restart icecast2.service
+ for i in "${services[@]}"
+ sudo systemctl restart extraction.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_recording.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_log.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_stats.service
+ sudo systemctl start birdnet_server.service
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      2118695/python3     
+ logger '[/usr/local/bin/restart_services.sh] birdnet_server.service is running'
+ break
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      2118695/python3     
+ status=0
+ ((  status != 0  ))
+ sudo systemctl restart birdnet_analysis.service
`

Only running for a couple of days on a rPI4 with latest OS version fully updated.
I noticed some updated files in the github repo so tried an update but there seem to be some errors in the git commands. Is this normal or do I have an issue on my environment?

ehpersonal38, 2023-04-27:
Just tested on my system and updates are working as expected. Could you try the following commands to see if they resolve the issue?

```
git gc --prune=now
git remote prune origin
```

wokkeltje13, 2023-04-27:
I receive the following info when running the 2 commands
[image]

This is a fresh install from a couple of days ago.

wokkeltje13, 2023-04-27:
The error in the update seems to be gone, but don't know if all is OK now

`++ sudo -u pi git -C /home/pi/BirdNET-Pi rev-parse HEAD
++ set +x
+ sudo -u pi git -C /home/pi/BirdNET-Pi reset --hard
HEAD is now at 2be0b394e fixy
+ set +x
+ sudo -u pi git -C /home/pi/BirdNET-Pi fetch origin main
From https://github.com/mcguirepr89/BirdNET-Pi
 * branch                main       -> FETCH_HEAD
error: cannot lock ref 'refs/remotes/origin/main': unable to resolve reference 'refs/remotes/origin/main': reference broken
 ! [new branch]          main       -> origin/main  (unable to update local ref)
+ set +x
+ sudo -u pi git -C /home/pi/BirdNET-Pi switch -C main --track origin/main
warning: ignoring broken ref refs/remotes/origin/main
fatal: invalid reference: origin/main
+ set +x
+ sudo -u pi git -C /home/pi/BirdNET-Pi diff --stat 2be0b394ee64078e94abfb37a5feb2a1f1a8c9a7 HEAD
+ set +x
+ my_dir=/home/pi/BirdNET-Pi/scripts
+ sudo systemctl stop birdnet_server.service
+ sudo systemctl stop birdnet_recording.service
+ services=(chart_viewer.service spectrogram_viewer.service icecast2.service extraction.service birdnet_recording.service birdnet_log.service birdnet_stats.service)
+ for i in "${services[@]}"
+ sudo systemctl restart chart_viewer.service
+ for i in "${services[@]}"
+ sudo systemctl restart spectrogram_viewer.service
+ for i in "${services[@]}"
+ sudo systemctl restart icecast2.service
+ for i in "${services[@]}"
+ sudo systemctl restart extraction.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_recording.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_log.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_stats.service
+ sudo systemctl start birdnet_server.service
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      3455007/python3     
+ logger '[/usr/local/bin/restart_services.sh] birdnet_server.service is running'
+ break
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      3455007/python3     
+ status=0
+ ((  status != 0  ))
+ sudo systemctl restart birdnet_analysis.service
`

ehpersonal38, 2023-04-27:
Try these:

```
git remote set-head origin --auto
git gc```

closed 2023-04-27 by ehpersonal38