mcguirepr89/BirdNET-Pi #876: "Close" Flickr image not visible on mobile

ehpersonal38, 2023-04-27:
**Describe the bug**
On mobile if you open a Flickr image, the close button at the bottom is obscured and unable to be clicked.

lloydbayley, 2023-04-27:
Oh YUCK. That's going to be a bugger...investigating...

lloydbayley, 2023-04-27:
That's weird, I can't replicate this. It works ok on mine with the only requirement being scrolling by swiping up on the text, not the picture.

ehpersonal38, 2023-04-27:
@lloydbayley where'd your comments go? I tried what you said (scrolling on the text and not the picture) and it led me to discover the root of this problem. Will push a fix shortly!

closed 2023-04-27 by ehpersonal38

lloydbayley, 2023-04-28:
Ahh...I second-guessed myself and thought I didn't know what I was talking about so I zapped my ramblings. Turns out, from looking at your commit, it was exactly where I was poking about. Nevermind.

ehpersonal38, 2023-04-28:
Don't second guess! Your contributions are valuable here!

commit 2c77964b (2023-04-27): fix #876