mcguirepr89/BirdNET-Pi #878: Analysis stops (or does not start), when user is NOT logged in

HolgerTs, 2023-04-28:
**Describe the bug**
There is a problem with services, when birdnet-user is not logged in (e.g. by ssh). In that case no recording and no processing is done. So when one want to have 24/7 analyzing, he has to keep open ssh connection to birdnet-user.

**To Reproduce**
Steps to reproduce the behavior:
1. do a fresh install of os-image
2. create a (second) admin account
3. login with birdnet-user-account and install birdnet
4. after successful install, an restart occures
5. after the restart, log in with admin account and check system - you will see, birdnet services are not running.
6. start a second terminal using birdnet-user - you will see, birdnet services start running.
7. close terminal connection with logout birdnet-user - you will see, birdnet services stop running.

**Expected behavior**
birdnet services are running, independent a ssh connection is open, or a user is logged in or not.

**Additional information**
After executing step 7 (above), sometimes the services are not stopped, but evering seems to be o.k. But latest after a reboot, the failure can be seen again.

**Your build**
Raspberry Pi 4, 8GB, clean install about 4 weeks ago. Then it stops working. Failure reproduced yesterday with fresh birdnet-install.

**Code or log snippets**
```
journald:
Apr 28 12:01:28 birdnetpi livestream.sh[1056]: Failed to create secure directory (/run/user/1000/pulse): No such file or directory
Apr 28 12:01:28 birdnetpi livestream.sh[1056]: [alsa @ 0x556537d060] cannot open audio device default (No such file or directory)
Apr 28 12:01:28 birdnetpi livestream.sh[1056]: default: Input/output error
Apr 28 12:01:28 birdnetpi systemd[1]: livestream.service: Main process exited, code=exited, status=1/FAILURE
Apr 28 12:01:28 birdnetpi systemd[1]: livestream.service: Failed with result 'exit-code'.

admin@birdnetpi:~ $ systemctl status livestream.service
● livestream.service - BirdNET-Pi Live Stream
     Loaded: loaded (/home/birder/BirdNET-Pi/templates/livestream.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Fri 2023-04-28 12:09:33 CEST; 1s ago
    Process: 2654 ExecStart=/usr/local/bin/livestream.sh (code=exited, status=1/FAILURE)
   Main PID: 2654 (code=exited, status=1/FAILURE)
        CPU: 297ms
        
admin@birdnetpi:~ $ ls -l /run/user
insgesamt 0
drwx------ 6 admin admin 140 28. Apr 12:08 1001

admin@birdnetpi:~ $ cat /etc/systemd/system/livestream.service
[Unit]
Description=BirdNET-Pi Live Stream
After=network-online.target
Requires=network-online.target
[Service]
Environment=XDG_RUNTIME_DIR=/run/user/1000
Restart=always
Type=simple
RestartSec=3
User=birder
ExecStart=/usr/local/bin/livestream.sh
[Install]
WantedBy=multi-user.target
admin@birdnetpi:~ $ 

users:
admin:x:1001:1001:,,,:/home/admin:/bin/bash
birder:x:1000:1000:,,,:/home/birder:/bin/bash

birdnetuser is called birder:
birdnetpi:~/BirdNET-Pi$ id
uid=1000(birder) gid=1000(birder) Gruppen=1000(birder),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),104(input),106(render),108(netdev),997(gpio),998(i2c),999(spi)
birdnetpi:~/BirdNET-Pi$ 
```
Logs were taken using administrator account, having user-id 1001.

[birdnet-user_journald-after-logout.log](https://github.com/mcguirepr89/BirdNET-Pi/files/11353024/birdnet-user_journald-after-logout.log)
[birdnet-user_logged-in.log](https://github.com/mcguirepr89/BirdNET-Pi/files/11353025/birdnet-user_logged-in.log)
[birdnet-user_not-logged-in.log](https://github.com/mcguirepr89/BirdNET-Pi/files/11353026/birdnet-user_not-logged-in.log)

HolgerTs, 2023-04-28:
I did install image without GUI. If I remember correctly, it was this image:
https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2023-02-22/2023-02-21-raspios-bullseye-arm64-lite.img.xz This image I copied to SD-card using dd command. So I also didn't use the Raspberry Pi Imager. br Holger

lloydbayley, 2023-04-29:
Interesting... Tell me, did you have this problem and then do a fresh install to test the theory out or is this your only and active install?

HolgerTs, 2023-04-29:
I did a fist install about 4 weeks ago. Then sometimes later, I recocnized this behavior. I'm not sure, if the problem was from the beginning. After some time of unsuccessful throubleshooting with this installation, I decided to do a fresh install. Problem is seen also with this second install and provided logs are from that.

lloydbayley, 2023-04-29:
Okay. Well, this is quite a tricky one. This is a long-shot. Make sure the system is using the multi-user.target.
I'm wondering if it might be related to a problem I had a long time ago in #319.
There are some commands that may help.
Specifically [here](https://github.com/mcguirepr89/BirdNET-Pi/issues/319#issuecomment-1140266771)

ehpersonal38, 2023-04-29:
Don't mean to butt in here - just want to comment that I think I know the bells that were ringing for @lloydbayley, in case it ends up being useful - https://github.com/mcguirepr89/BirdNET-Pi/issues/302#issuecomment-1172912042.

lloydbayley, 2023-04-30:
No that's fine...I was one off wasn't I?!! It was a year ago so I was having trouble retrieving a record from the old database! :)

HolgerTs, 2023-04-30:
Giving the command:
```
sudo loginctl enable-linger $USER
```
immediately lets the system work correctly. Also after a reboot everything seems to be fine. Will observe it for the next hours, days...

lloydbayley, 2023-04-30:
Good one. Hope it stays working!

lloydbayley, 2023-05-05:
How's the system running @HolgerTs ? Still alive and kicking?

HolgerTs, 2023-05-05:
uptime: more than 5 days. No loss of data could be seen. So, maybe ```sudo loginctl enable-linger $USER``` is not a smartly solution, but it's a good work around.

HolgerTs, 2023-05-06:
Just did a reboot for testing. System works normally after that, too.

ehpersonal38, 2023-05-06:
Closing as resolved!

closed 2023-05-06 by ehpersonal38