mcguirepr89/BirdNET-Pi #879: Software running but no birds detected.

hugsworth, 2023-04-28:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
A clear and concise description of what the bug is.
I am installing birdnetpi on a rasberry Pi 3B+. I have installed it using your curl command and it would great. Once installed it detects nothing. I can hear the birds on the playback. The log seems to indicate that it is recording and analyzing the data. It appears to be identifying birds in the log but then it writes 0 results. I have reloaded from the OS up several times both on the lite and regular versions of the OS.
**To Reproduce**
Steps to reproduce the behavior:
1. Install the most recent version of the 64 bit lite PI OS.
2. Run the curl command to install the tool.
3. Log into the web interface and set long and latitude.
4. Check log which shows 0 new records over and over again.
5. Listen to the feed to make sure my Mic is working.

**Expected behavior**
Birds to be identified and saved to the local database.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem or your installation here.

**Your build**
Rasberry Pi 3B+

**Code or log snippets**
```
21:03:48---RECORDING_LENGTH set to 15
21:03:57---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts/analyze.py --i BirdSongs/April-2023/28-Friday/2023-04-28-birdnet-21:03:40.wav --o BirdSongs/April-2023/28-Friday/2023-04-28-birdnet-21:03:40.wav.csv --lat 48.XX --lon 117.XX --week 16 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
21:03:57---ANALYZING AUDIO... DONE! Time 3.5 SECONDS
21:03:57---WRITING RESULTS TO BirdSongs/April-2023/28-Friday/2023-04-28-birdnet-21:03:25.wav.csv ... DONE! WROTE 0 RESULTS.
21:03:57---READING AUDIO DATA... DONE! READ 5 CHUNKS.
21:03:57---Date-time: 2023-04-28 21:03:40
21:04:01---0.0;3.0-('Carpodacus sibiricus_Long-tailed Rosefinch', 0.14818247)
21:04:01---3.0;6.0-('Passer montanus_Eurasian Tree Sparrow', 0.0635462)
21:04:01---6.0;9.0-('Calliope calliope_Siberian Rubythroat', 0.11774561)
21:04:01---9.0;12.0-('Emberiza spodocephala_Black-faced Bunting', 0.06586223)
21:04:01---12.0;15.0-('Emberiza spodocephala_Black-faced Bunting', 0.046970073)
21:04:04---Files loaded
21:04:05---RECORDING_LENGTH set to 15
21:04:11---BirdNET-Pi/birdnet/bin/python3 BirdNET-Pi/scripts/analyze.py --i BirdSongs/April-2023/28-Friday/2023-04-28-birdnet-21:03:55.wav --o BirdSongs/April-2023/28-Friday/2023-04-28-birdnet-21:03:55.wav.csv --lat 48.XX --lon 117.XX --week 16 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7
21:04:11---ANALYZING AUDIO... DONE! Time 3.5 SECONDS
21:04:11---WRITING RESULTS TO BirdSongs/April-2023/28-Friday/2023-04-28-birdnet-21:03:40.wav.csv ... DONE! WROTE 0 RESULTS.
21:04:12---READING AUDIO DATA... DONE! READ 5 CHUNKS.
21:04:12---Date-time: 2023-04-28 21:03:55
21:04:15---0.0;3.0-('Emberiza spodocephala_Black-faced Bunting', 0.18273588)
21:04:15---3.0;6.0-('Passer montanus_Eurasian Tree Sparrow', 0.12137944)
```

ehpersonal38, 2023-04-29:
Can you try opening the View Logs page on your computer, then play a recording of a local bird (check YouTube) on your phone and hold it near the microphone? I am interested in the confidence numbers it generates for that bird.

hugsworth, 2023-04-29:
I will give it a try. I am working on another project today but should have an update in a day or so.

hugsworth, 2023-04-30:
I was able to test this and it worked. I am still getting very low bird hits but at this point I don't think it is an issue with the software but rather an issue with my microphone and  how I have it configured. Thanks for your help but I am going to close this issue as chocked up to inexperience.

closed 2023-04-30 by hugsworth