mcguirepr89/BirdNET-Pi #884: Error message in weekly report

Svardsten53, 2023-05-02:
**Describe the bug**
Sometimes, not every week, I get an error message in the weekly report. If I change the value 10) to 7) in weekly_report.php, the error message disappears this week.

**Expected behavior**
Weekly report every week without error message.

**Screenshots**

[image]

**Additional context**
The error seems to be due to the number of species compared between the different weeks. At first I thought it was due to very few species a week, but in recent weeks the number of species has been around 40.

**Your build**
Raspberry Pi 4B

ehpersonal38, 2023-05-02:
Interesting, I think this happens when you have a never before seen species in your top 10 list, so when it tries to generate the % change value, there's no previous week data to compare to for that species. Hence the value is just 0, since 0 detections of that species in the prior week.

@lloydbayley want to have a go at this? It's line 213 of weekly_report, it needs to check if the previous week count for the species is > 0, and if not just set the percentage chance to "0%"

lloydbayley, 2023-05-03:
Eww! Divide by zeros! I'm on it.

lloydbayley, 2023-05-03:
Should be right to go now. Please check it because there seems to be another commit attached to my fix and I don't understand why. Should it not only be one commit?

lloydbayley, 2023-05-03:
Eeeexxxxxcellentttt! (Picture Mr. Burns)

ehpersonal38, 2023-05-03:
Thank you!! And I think that extra commit occurs after you update your fork from the main repo, it has to create a "merge" commit which ends up in the commit logs like that. It's just a nuisance, not harmful though. Thanks again!

lloydbayley, 2023-05-03:
Roger that! Will remember that for next time! As it was, I accidentally created a PR in the Kahst repo. I checked it to make sure it was this repo and it must have flicked back to that one. I saw over 2000 changes and thought WHAT???????. Couldn't delete it, so had to close it and comment that I made a whoops and hope he deletes it so my fox paws is not discovered. :)

ehpersonal38, 2023-05-03:
Hehehe, welcome to the graveyard https://github.com/kahst/BirdNET-Lite/pulls?q=is%3Apr+is%3Aclosed

lloydbayley, 2023-05-03:
Shhh....don't tell everyone! Don't put it on the internet, whatever you do! :)

closed 2023-05-05 by ehpersonal38