mcguirepr89/BirdNET-Pi #889: Unable to set Tools password

stevenidavis, 2023-05-04:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
Using the web interface I am unable to set a password for tools. Firefox will not allow passwords to be sent over HTTP (must be HTTPS) and SSL is not available on the server.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to birdnetpi.local (or your locally configured address)
2. Click on Tools
3. Click on settings
4. Scroll down to bottom of page and click Advanced Settings
5. Scroll down to BirdNET-PI Password and enter a password.
6. Scroll to bottom and click Update Settings
7. Firefox gives and error saying you should not pass a password over an insecure connection. There is no way to disable this in recent versions.

**Expected behavior**
The tools password should be updated.

**Additional context**
Add any other context about the problem or your installation here.

**Your build**
Raspberry Pi 4

**Code or log snippets**
```
Paste your 
log output 
or 
code here
```

ehpersonal38, 2023-05-04:
Hi, thanks for reporting this issue. I tried the steps you listed to reproduce it and I am unable to. My Firefox is the latest version (112.0.2). I do see a warning message when I hover over the password but that doesn't prevent me from clicking "Update Settings." Any tips for how I can reproduce this?

[image]

croisez, 2023-05-11:
@stevenidavis , could you try to set the password in the configuration file itself, and see if this workaround is working ?
This could at least allows you to protect your installation.
The parameter to change in birdnet.conf is : CADDY_PWD

lloydbayley, 2023-05-15:
How did you go @stevenidavis ? Any further information?

stevenidavis, 2023-05-15:
It is throwing some sort of error because there is a special character in the password. Will debug more later to figure out which one.

jaredb7, 2023-05-16:
Ahhh special characters, that'll definitely be it. Should be easy to fix/handle, I'll see if I can get something together a later.

croisez, 2023-05-16:
i am curious to know which special character lead to this error... but what error, btw ? You did'nt mention it ?

jaredb7, 2023-05-16:
Normally it's `& + = ? %` as they special meaning URL or as data components in requests. You'll often get missing text or no text depending on where they fall and how they're interpreted in the text.

lloydbayley, 2023-05-16:
It's a tricky thing. It's just noted in the docs about not using special characters so far.

stevenidavis, 2023-05-16:
20:42:45---< CADDY_PWD=
20:42:45---The birdnet.conf file has changed
20:42:47---/etc/birdnet/birdnet.conf: line 41: unexpected EOF while looking for matching ``'
20:42:48---/etc/birdnet/birdnet.conf: line 41: unexpected EOF while looking for matching ``'
20:42:48---/etc/birdnet/birdnet.conf: line 228: syntax error: unexpected end of file

jaredb7, 2023-05-16:
> It's a tricky thing. It's just noted in the docs about not using special characters so far.

Hmm, also appears you can't use special chars in password as the web ui warns you when it doesn't alphanumerical (there is a pattern match on the password field). 
I removed the pattern attribute so I can do whatever to try test it and and it all actually seems to work....
[image]

--Settings INI also ok
May 16 22:24:51 BirdNET birdnet_analysis.sh[79980]: < CADDY_PWD=
May 16 22:24:51 BirdNET birdnet_analysis.sh[79980]: ---
May 16 22:24:51 BirdNET birdnet_analysis.sh[79980]: > CADDY_PWD=b1rdn3t!&+=?%'"
May 16 22:24:51 BirdNET birdnet_analysis.sh[79836]: The birdnet.conf file has changed

Until you do something like this  

- `b1rdn3t!&+'=@?\*(_)~% 

I tested back and forth and it's fine until the backquote character (I placed it at the start but doesn't matter where) appears in the password

Then I get similar messages as the op

> May 16 22:48:06 BirdNET birdnet_analysis.sh[143160]: /etc/birdnet/birdnet.conf: line 38: unexpected EOF while looking for matching ``'
> May 16 22:48:06 BirdNET birdnet_analysis.sh[143160]: /etc/birdnet/birdnet.conf: line 225: syntax error: unexpected end of file
> May 16 22:48:06 BirdNET birdnet_analysis.sh[143160]: /etc/birdnet/birdnet.conf: line 38: unexpected EOF while looking for matching ``'
> May 16 22:48:06 BirdNET birdnet_analysis.sh[143160]: /etc/birdnet/birdnet.conf: line 225: syntax error: unexpected end of file
> May 16 22:48:07 BirdNET env[143260]: /etc/birdnet/birdnet.conf: line 38: unexpected EOF while looking for matching ``'
> May 16 22:48:07 BirdNET env[143260]: /etc/birdnet/birdnet.conf: line 225: syntax error: unexpected end of file
> 

So keeping in mind that we're limiting users to a alphanumerical password, I returned the code to normal and tried

- 123ab`c
(backquote is between b & c (again position doesn't matter as mentioned above)
[image]

This doesn't cause a prompt that I'm using characters which are not allowed and also
The password doesn't submit or save and I get behaviour that sounds like what the OP described, the password field gets focus and I get the insecure password warning box.
Which makes it sound like I'm not allowed to send a password over a insecure site, but it's some weird behaviour caused by the backtick. Ideally it should just warn the user about a password that doesn't match the format we set (re: first screenshot)
[image]

@stevenidavis are you able to confirm that you were using a backtick ? Just to confirm my findings.

fairly sure we can resolve this via tweaking the patter but not sure.

lloydbayley, 2023-05-16:
Yep. Agreed. This is why we sort of left it in the GUI as "don't do it!" 😄 
Backticks and any kind of quotes can be diabolical if combined in odd ways not thought of. I think there's also an element of not wanting to accidentally introduce a vulnerability whereby it could be used to url-encode something nasty and force it through...

stevenidavis, 2023-05-16:
I can confirm there is a backtick in my password. Will try removing it this afternoon. Really appreciate all the time you guys are investing in this. This system really has been something interesting to setup and use.

ehpersonal38, 2023-05-16:
@jaredb7 Good detective work! I think what may be happening is that the "insecure connection" popup is preventing the form validation error popup from showing. I guess this should be a bug report opened with Mozilla. In the meantime I'll just add a message above the password input to not use special characters.

croisez, 2023-05-16:
I don't understand. On my setup, I can only set a password with AZ az 09 chars.
If I use a backtick in it and press enter, the form does not send my password to advanced.php, due to the pattern="[A-Za-z0-9]+" property of the <input> tag.
So, what is causing the error reported by @stevenidavis ?

ehpersonal38, 2023-05-16:
No I couldn't actually submit the password, on all browsers if the password doesn't match the pattern then the form will not be submitted.

The main issue is that what I said in my comment above yours. It's a cosmetic issue where the user isn't informed that their password can't be submitted due to being invalid, because the "insecure connection" popup box is preventing the "invalid password" box from appearing.

jaredb7, 2023-05-16:
> I don't understand. On my setup, I can only set a password with AZ az 09 chars. If I use a backtick in it and press enter, the form does not send my password to advanced.php, due to the pattern="[A-Za-z0-9]+" property of the input tag. So, what is causing the error reported by @stevenidavis ?
> 

Correct, that's the expected and observed behaviour. In Firefox though it behaves the same until you use the backtick then there's a cosmetic issue which makes it appear as if there's different issue.

2nd, the error posted a bit later about ` unexpected end of file ` is most likely because it was suggested that the OP edit the birdnet.conf file and set the password there as a workaround. They've taken the password (which has a backtick) that they couldn't submit via the web ui and added it into the config file, causing the 2nd causing that error since backticks are an illegal character.

stevenidavis, 2023-05-19:
Just to close this out, I was able to set a password when it was all alphanumeric and did not have any special characters