mcguirepr89/BirdNET-Pi #895: Browsing recordings by species on web interface does not show audio player as expected

croisez, 2023-05-08:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
For some recordings, if I go to the menu Recordings->BySpecies, I am expecting to see a spectrogram + the audio player at the bottom.
But, for some reasons, for some species, I do not see the audio player.
Instead, the spectrogram is clickable and a click on it redirect the browser to the audio file, but not using the ?filename= scheme in the URL. 

**To Reproduce**
Example of bad behavior:
Example of expected behavior:
go to Recordings=>BySpecies=>Troglodyte Mignon
By clicking on spectrogram, browser goes to http://birdnetpi.croisez.be/By_Date/2023-05-07/Troglodyte_mignon/Troglodyte_mignon-90-2023-05-07-birdnet-21:01:25.mp3 (the actual MP3 content)
`<a href="/By_Date/2023-05-07/Troglodyte_mignon/Troglodyte_mignon-90-2023-05-07-birdnet-21:01:25.mp3">
[image]`

**Expected behavior**
Example of expected behavior:
go to Recordings=>BySpecies=>Sitelle Torchepot
`<video onplay="setLiveStreamVolume(0)" onended="setLiveStreamVolume(1)" onpause="setLiveStreamVolume(1)" controls="" poster="/By_Date/2023-05-02/Sittelle_torchepot/Sittelle_torchepot-76-2023-05-02-birdnet-06:35:57.mp3.png" preload="none" title="/By_Date/2023-05-02/Sittelle_torchepot/Sittelle_torchepot-76-2023-05-02-birdnet-06:35:57.mp3">
<source src="/By_Date/2023-05-02/Sittelle_torchepot/Sittelle_torchepot-76-2023-05-02-birdnet-06:35:57.mp3">
</video>`

**Additional context**
I don't know why there is a different behavior, between a bird and another.
You can see on the HTML code excerpts that there are differences.  

**Your build**
The latest: https://github.com/mcguirepr89/BirdNET-Pi/commit/bddd8b3b9f31629e5b0c2d66a4b93d93457fc92d

Svardsten53, 2023-05-08:
That is also my experience.

blogdemoi, 2023-05-08:
I also thought it was an issue but actually it is the expected behaviour, @ehpersonal38 explains it here:

> Heh, actually this isn't a bug at all. What's happening is I made a change to the code last month or to improve performance on the Recordings page. I noticed that the page would become virtually unusable when there was a large amount of media player elements visible on the screen at one time. So now, for pages on Recordings which have > 100 recordings, it just removes the media player altogether and reverts to a simpler "click-to-listen" type interface. That's why you only notice this for some species and not others. Hope that clears things up for you.

Source: https://github.com/mcguirepr89/BirdNET-Pi/discussions/824

croisez, 2023-05-08:
As @blogdemoi reported to me, there is a discussion related to this issue: https://github.com/mcguirepr89/BirdNET-Pi/discussions/824
Basically, for pages on Recordings which have > 100 recordings, it just removes the media player altogether and reverts to a simpler "click-to-listen" type interface.

But now, there is a problem with the ShiftFreq feature, which tries to alter the media URL after the frequency shifting operation.
You cannot immediately hear the result of the frequency shifting.
I will update the PHP code to try to search for the <a> token and alter the media URL in the HREF section.

ehpersonal38, 2023-05-08:
@croisez If that's too much trouble I can just revert this change and instead add a "Load 40 More detections" button like we have on "Today's Detections." That's probably how it should've been done in the first place.

croisez, 2023-05-08:
i just produced a code modification and a pull request

ehpersonal38, 2023-05-09:
https://github.com/mcguirepr89/BirdNET-Pi/pull/896 this one? I merged it, thanks! I'll close this issue now too.

closed 2023-05-09 by ehpersonal38