mcguirepr89/BirdNET-Pi #897: DetectionsOver30d panel should disappear when another identical panel is opened

croisez, 2023-05-09:
**Description**
As you can see on this [screenshot](https://i.imgur.com/icIANAg.png), there should be a way to make the currently shown panel before opening a new one.

**To Reproduce**
1. On the main page (Overview), click on the icon for displaying frequency of detections, for a chosen bird.
A "Detections over 30d" panel is shown.
2. Click on the same icon, but for another specie, and so on.
3. All the panels are on top of another.

**Expected behavior**
I think that we should better make the current panel disappear before displaying the next one.

**Screenshots**
[here](https://i.imgur.com/icIANAg.png)

**Your build**
https://github.com/mcguirepr89/BirdNET-Pi/commit/53fa65e364e9249826d561007f489b4f23ba6ab0

lloydbayley, 2023-05-09:
Fair point. It will clear itself if you scroll slightly but I see the 'annoyance-factor'. My AJAX is not strong-enough to deal with this, however, I'm sure that @ehpersonal38 will have some good input on this.

lloydbayley, 2023-05-15:
Resulting PR. Can be closed after test and merge.

closed 2023-05-22 by ehpersonal38