mcguirepr89/BirdNET-Pi #903: Web terminal height issue: 4 last lines are hidden on small screens

croisez, 2023-05-11:
**Describe the bug**
When I enter the web terminal, it has systematically 4 lines hidden at the bottom.

**To Reproduce**
1. go to Tools => Web Terminal
2. enter user login and password.
3. display some content on the terminal, for example: list the current directory
```
ls -1R
```
4. On my system, I have to push 4 times on the ENTER key to see the terminal prompt
```
birdnet-pi4:~/BirdNET-Pi$
```
5. Move the browser window to another screen, and enlarge the height of this browser window until you begin to see the green frame below the web terminal

**Expected behavior**
The terminal prompt should be displayed immediately.
The terminal should display every lines for which it has been configured.
We should be able to see the green frame below the web terminal, even on small landscape screens.

**Screenshots**
Issue occuring when height of browser is too small: https://i.imgur.com/qJvDPMh.png
Issue resolved when height of browser is sufficient: https://i.imgur.com/d7b2W6q.png

**Additional context**
I am using Chrome on WIN11.
I have an external LCD screen, positioned in portrait mode, so having a wide vertical span, and with that config, the last lines of the terminal are not hidden.

**Your build**
https://github.com/mcguirepr89/BirdNET-Pi/commit/b9c2a0e5b708a863bc6217db85dc013099389dc4

**Code **
I would like to investiguate, but I don't know where to look in the code for this web terminal configuration and behavior

lloydbayley, 2023-05-12:
Yep, you're right. I have experienced this in other places like the log view too. I haven't looked into this more but I will. As a temp thing, put your mouse cursor up in the top green bit and scroll. This should bring the bottom line into view. That's how I've been doing it but I suppose I could actually have a look at this. I'm just in the middle of a project at the moment, so give me a couple of days.

croisez, 2023-05-12:
Thank you for this workaround. Very useful for me.
Concerning this issue, I think that it could be related to the <iframe> we are using to mask complex URL.
Actually, if you test the Web terminal without using iframe, for example, by using the complex URL directly, you don't get problems of hidden lines in the web terminal.
Try it: http://YOU-BIRDNET-IP/views.php?view=Webterm 
On my setup, with such URL, Web terminal is behaving well.

lloydbayley, 2023-05-12:
You're dead right. I will get onto it asap. I have been putting up with it myself instead of doing something about it.
Time for action! :) Soon.
Thanks for the extra pointers. It's exactly as you say with the banner.

croisez, 2023-05-12:
I found I have found a way to keep the iframe and the banner, while having a good height configured for Gotty.
Could you test it on your side ?
The mod is the following: 
```
$ git diff style.css
diff --git a/homepage/style.css b/homepage/style.css
index 69c350a..346fb2f 100644
--- a/homepage/style.css
+++ b/homepage/style.css
@@ -26,7 +26,7 @@ iframe {
   padding: 0;
   margin: 0;
   border: none;
-  height: 93%;
+  height: 90%;
   width: 100%;
 }

@@ -836,4 +836,4 @@ pre#timer.bash {

 #ddnewline::before {
    content: none;
-}
\ No newline at end of file
+}
```
Here is a screenshot of my RPI4, with this CSS modification
https://i.imgur.com/jzobXNP.png

croisez, 2023-05-12:
yes I will do it.
Thanks for testing

croisez, 2023-05-12:
I had already a pending pull request: https://github.com/mcguirepr89/BirdNET-Pi/pull/898
So when I pushed my css file modification now, I was surprised that this last commit was part of the pending pull request... but this is definitely normal behavior.
So, sorry, I have pushed a commit which is not related to the subject of the initial idea of "Removing DetectionsPanel"

lloydbayley, 2023-05-12:
Don't worry about it. It's just the way of things. I'll have a look at your other one when I get the chance to test it and then Ethan can push it through at some stage. I think he's a bit on the busy side at the moment. Unfortunately, I do not posses the magic powers to do that.

croisez, 2023-05-12:
Yes, I know that sometimes, we could have a real life behind our passions :-D

lloydbayley, 2023-05-15:
Resulting PR. Can be closed after merge and test.

croisez, 2023-05-20:
I verified on my Macbook this morning, and it seems that <iframe height=90%> was not enough to resolve the Gotty terminal masked on some layouts.
I propose 85%.

```
birdnet-pi4:~/BirdNET-Pi/homepage$ git diff
diff --git a/homepage/style.css b/homepage/style.css
index 346fb2f..d3a7c4a 100644
--- a/homepage/style.css
+++ b/homepage/style.css
@@ -26,7 +26,7 @@ iframe {
   padding: 0;
   margin: 0;
   border: none;
-  height: 90%;
+  height: 85%;
   width: 100%;
 }
```

Would you test if this is OK on your computer ?

lloydbayley, 2023-05-20:
Yeh, I must admit that It 99% resolved it on my MacBook and I thought that was sufficient and didn't want to be difficult about the one line difference but you are correct. Will re-test this on my smaller screens and make sure it's ok.

lloydbayley, 2023-05-20:
Fantastic. Well-done! Please PR. It's a winner!

croisez, 2023-05-20:
Correction added to my current PR: https://github.com/mcguirepr89/BirdNET-Pi/pull/912/commits/64603f146f913257eb390d0689c3a72101f23577

lloydbayley, 2023-05-20:
Good onya! I'd make it so (Picard) now but I don't have 'the power'! :) Well-done again!

croisez, 2023-05-20:
Yes, let us make it so ! :-D
https://www.youtube.com/watch?v=xteKObnaA2c

lloydbayley, 2023-05-20:
Oh that's GOLD! Thank you. Bookmarked.

lloydbayley, 2023-05-22:
I think we can Closee-Closee this issue now!

closed 2023-05-22 by ehpersonal38