mcguirepr89/BirdNET-Pi #904: Make BirdNET-Pi easier to containerize

DomAmato, 2023-05-11:
Not really a bug report, more a feature request, but looking over the install instructions my impression is that making this work with something like [Balena](https://www.balena.io/) might not be too difficult but that since everything is wrapped up in scripts I'm not sure how well that would work with the docker build steps.

Theres some issues with going that route since once a container restarts in balena its a totally new image, meaning long term data storage would likely require an external source. However I think that being able to create and manage a fleet of these to cover large natural areas would be a god send for a lot of the small organizations that want to cover a larger area.

I'm working with one right now and am about to dive into making it work but just a thought on making this a more easily managed IoT solution for small NGOs :)

lloydbayley, 2023-05-12:
There's a great deal of discussion about this and VM's etc in discussion #251. It may be of interest to you.

DomAmato, 2023-05-12:
thanks for leading me to that, ill close this issue then

closed 2023-05-12 by DomAmato