mcguirepr89/BirdNET-Pi #905: Error message on Species Stats page:  "RangeError: min (1) is equal/bigger than max (1)"

dgoadby, 2023-05-12:
**Describe the bug**

On the Species Stats page at 01:00 today I selected "Single Day View". The "Resample Resolution" was 15 mins (standard option). There was only one specie detected for the day so far.  For today's date the message below appears in the left-hand side of the screen.

After the message the system does not crash and, if the previous days data is selected, then that works perfectly.

This is the screenshot:

[image]

This is the full text from the error panel:
 
`RangeError: min (1) is equal/bigger than max (1)

46893/t.prototype.componentDidMount/<@http://birdnetpi1.local/stats/static/js/292.ef38e7f6.chunk.js:1:18248
46893/t.prototype.componentDidMount@http://birdnetpi1.local/stats/static/js/292.ef38e7f6.chunk.js:1:18218
Mu@http://birdnetpi1.local/stats/static/js/main.1d359564.js:2:2445038
Ts@http://birdnetpi1.local/stats/static/js/main.1d359564.js:2:2461918
13095/t.unstable_runWithPriority@http://birdnetpi1.local/stats/static/js/main.1d359564.js:2:3355189
Zo@http://birdnetpi1.local/stats/static/js/main.1d359564.js:2:2403785
Rs@http://birdnetpi1.local/stats/static/js/main.1d359564.js:2:2458655
gs@http://birdnetpi1.local/stats/static/js/main.1d359564.js:2:2454739
30534/Ko/<@http://birdnetpi1.local/stats/static/js/main.1d359564.js:2:2404008
13095/t.unstable_runWithPriority@http://birdnetpi1.local/stats/static/js/main.1d359564.js:2:3355189
Zo@http://birdnetpi1.local/stats/static/js/main.1d359564.js:2:2403785
Ko@http://birdnetpi1.local/stats/static/js/main.1d359564.js:2:2403955
$o@http://birdnetpi1.local/stats/static/js/main.1d359564.js:2:2403888
fs@http://birdnetpi1.local/stats/static/js/main.1d359564.js:2:2451891
enqueueSetState@http://birdnetpi1.local/stats/static/js/main.1d359564.js:2:2407906
80306/m.prototype.setState@http://birdnetpi1.local/stats/static/js/main.1d359564.js:2:2734010
n/o.handleDeltaMsg/<@http://birdnetpi1.local/stats/static/js/main.1d359564.js:2:4693251

Made with Streamlit
`
**Your build**
Latest download on 10th May 2023 with updates. Is there a place to look this up from the web interface?

The hardware on which BirdNET-Pi is running goes here.
RPI3 b+.

lloydbayley, 2023-05-18:
Hi! Is this still happening, or did it settle down?

dgoadby, 2023-05-19:
It happened again this morning. The problem only occurs at the very start of the day when there are few, in any, data records. I will run a test at midnight today to see exactly when it happens and when it stops happening. 

I have some free time so I will have a look at the code. I'm not very strong in PHP but I have been using Python for over 15 years.

lloydbayley, 2023-05-19:
Ok, I'd be interested to see if it's the same count every time that triggers it. Was wondering if it was if none existed.

ehpersonal38, 2023-05-20:
I think this is a bug with the "select number of species to show" slider. It tries to do some math but since there's only one species the math fails and produces this error.

lloydbayley, 2023-05-22:
It's a lot of faffing about for something that seldom occurs...isn't that always the way.
Will look into it a bit more....unless you have @ehpersonal38?

ehpersonal38, 2023-05-22:
Given that it's a non-fatal error, I don't think I'll mess around with the code too much as I have no idea how the streamlit code works. If you find out that it does impair functionality please open another issue and I'll give a better look into this. Thanks!

closed 2023-05-22 by ehpersonal38

lloydbayley, 2023-05-23:
Agreed. I had a look but it was a bit scary!