mcguirepr89/BirdNET-Pi #911: Can't load stats 502 response

mebezac, 2023-05-15:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Click on "Species stats"
2. see a 502 response

**Expected behavior**
going to "Species stats" should load correctly

**Screenshots**

[image]

**Your build**
Orange PI zero 2

**Code or log snippets**

From the syslog when trying to access stats

```
May 15 14:17:40 orangepizero2 caddy[1110]: {"level":"error","ts":1684174660.1175294,"logger":"http.log.error","msg":"dial tcp [::1]:8501: connect: connection refused","request":{"remote_ip":"10.25.30.8","remote_port":"42884","proto":"HTTP/1.1","method":"GET","host":"birdnet.laboratory.casa","uri":"/stats","headers":{"Cdn-Loop":["cloudflare"],"Cf-Visitor":["{\"scheme\":\"https\"}"],"Sec-Ch-Ua-Mobile":["?0"],"User-Agent":["Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/113.0.0.0 Safari/537.36"],"Priority":["u=0, i"],"X-Forwarded-Proto":["https"],"X-Forwarded-Server":["7d5d66ba9163"],"Cf-Ipcountry":["US"],"Dnt":["1"],"X-Forwarded-For":["172.18.0.1"],"Accept-Language":["en-US,en;q=0.9"],"Referer":["https://birdnet.laboratory.casa/views.php?view=Streamlit"],"X-Forwarded-Host":["birdnet.laboratory.casa"],"Sec-Fetch-Site":["same-origin"],"Sec-Fetch-User":["?1"],"Accept-Encoding":["gzip"],"Cookie":[],"Sec-Ch-Ua-Platform":["\"macOS\""],"Sec-Fetch-Mode":["navigate"],"Sec-Ch-Ua":["\"Chromium\";v=\"113\", \"Not-A.Brand\";v=\"24\""],"Sec-Fetch-Dest":["iframe"],"Accept":["text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.7"],"Cf-Connecting-Ip":["23.105.171.72"],"Cf-Ray":["7c7d61096dae826c-IAD"],"Cf-Warp-Tag-Id":["13202e58-5ee0-43c5-8ef1-e33bff5c475c"],"Upgrade-Insecure-Requests":["1"],"X-Forwarded-Port":["443"],"X-Real-Ip":["172.18.0.1"]}},"duration":0.004141997,"status":502,"err_id":"t6etby4a8","err_trace":"reverseproxy.statusError (reverseproxy.go:1299)"}
May 15 14:17:43 orangepizero2 streamlit[3710678]: Traceback (most recent call last):
May 15 14:17:43 orangepizero2 streamlit[3710678]:   File "/home/pi/BirdNET-Pi/birdnet/bin/streamlit", line 5, in <module>
May 15 14:17:43 orangepizero2 streamlit[3710678]:     from streamlit.web.cli import main
May 15 14:17:43 orangepizero2 streamlit[3710678]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/streamlit/__init__.py", line 55, in <module>
May 15 14:17:43 orangepizero2 streamlit[3710678]:     from streamlit.delta_generator import DeltaGenerator as _DeltaGenerator
May 15 14:17:43 orangepizero2 streamlit[3710678]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/streamlit/delta_generator.py", line 43, in <module>
May 15 14:17:43 orangepizero2 streamlit[3710678]:     from streamlit.elements.arrow_altair import ArrowAltairMixin
May 15 14:17:43 orangepizero2 streamlit[3710678]:   File "/home/pi/BirdNET-Pi/birdnet/lib/python3.9/site-packages/streamlit/elements/arrow_altair.py", line 36, in <module>
May 15 14:17:43 orangepizero2 streamlit[3710678]:     from altair.vegalite.v4.api import Chart
May 15 14:17:43 orangepizero2 streamlit[3710678]: ModuleNotFoundError: No module named 'altair.vegalite.v4'
May 15 14:17:45 orangepizero2 systemd[1]: birdnet_stats.service: Main process exited, code=exited, status=1/FAILURE
May 15 14:17:45 orangepizero2 systemd[1]: birdnet_stats.service: Failed with result 'exit-code'.
May 15 14:17:45 orangepizero2 systemd[1]: birdnet_stats.service: Consumed 14.945s CPU time.
```

from my installation log, `altair` was installed correctly:

```
Successfully installed MarkupSafe-2.1.2 PyYAML-6.0 altair-5.0.0 apprise-1.2.1 attrs-23.1.0 audioread-3.0.0 blinker-1.6.2 cachetools-5.3.0 certifi-2023.5.7 cffi-1.15.1 charset-normalizer-3.1.0 click-8.1.3 colorama-0.4.4 contourpy-1.0.7 cycler-0.11.0 decorator-5.1.1 fonttools-4.39.4 gitdb-4.0.10 gitpython-3.1.31 idna-3.4 importlib-metadata-6.6.0 iniconfig-2.0.0 jinja2-3.1.2 joblib-1.2.0 jsonschema-4.17.3 kiwisolver-1.4.4 librosa-0.9.2 llvmlite-0.39.1 markdown-3.4.3 markdown-it-py-2.2.0 matplotlib-3.6.3 mdurl-0.1.2 numba-0.56.4 numpy-1.19.5 oauthlib-3.2.2 packaging-23.1 paho-mqtt-1.6.1 pandas-1.4.4 pillow-9.5.0 pip-23.1.2 platformdirs-3.5.0 plotly-5.14.1 pluggy-1.0.0 pooch-1.7.0 protobuf-3.20.3 py-1.11.0 pyarrow-12.0.0 pycparser-2.21 pydeck-0.8.1b0 pygments-2.15.1 pympler-1.0.1 pyparsing-3.0.9 pyrsistent-0.19.3 pytest-7.1.2 pytest-mock-3.7.0 python-dateutil-2.8.2 pytz-2023.3 pytz-deprecation-shim-0.1.0.post0 requests-2.30.0 requests-oauthlib-1.3.1 resampy-0.4.2 rich-13.3.5 scikit-learn-1.2.2 scipy-1.10.1 seaborn-0.12.2 semver-3.0.0 setuptools-67.7.2 six-1.16.0 smmap-5.0.0 soundfile-0.12.1 streamlit-1.19.0 suntime-1.2.5 tenacity-8.2.2 tflite-runtime-2.6.0 threadpoolctl-3.1.0 toml-0.10.2 tomli-2.0.1 toolz-0.12.0 tornado-6.3.1 typing-extensions-4.5.0 tzdata-2023.3 tzlocal-4.3 urllib3-2.0.2 validators-0.20.0 watchdog-3.0.0 wheel-0.40.0 zipp-3.15.0
```

ehpersonal38, 2023-05-16:
Here is the fix, I will implement this soon.

https://discuss.streamlit.io/t/modulenotfounderror-no-module-named-altair-vegalite-v4/42921/6

In the meantime you can install an earlier version of Altair.

`~/BirdNET-Pi/bin/pip3 install altair<4`
^^ I don't know if the < syntax actually works here and I can't test it right now. You may need to lookup an actual version number on PyPi for altair and then replace `<4` with `=={version code}`. Whatever you do make sure to use pip3 binary located at `~/BirdNET-Pi/bin/pip3`

mebezac, 2023-05-16:
Thanks! Here's the command that fixed it for me:

`~/BirdNET-Pi/birdnet/bin/pip3 install 'altair<5'`

closed 2023-05-16 by ehpersonal38