mcguirepr89/BirdNET-Pi #915: Security findings

gibbypip, 2023-05-18:
Hey there! 

I'd like to responsibly disclose some security findings I've come across but would like to submit them in a non-public way. Is there an email address I can send the information to? 

Thanks!

lloydbayley, 2023-05-19:
Good one. @ehpersonal38 is the one for this. I won't give out the email here but I'm sure he will be in touch when he wakes up. :)

Valodim, 2023-05-20:
...the short version here is that birdnetpi is pretty much one big security hole. It is fine to have in your local network, but I would really not recommend anyone expose their instance to the internet without a serious amount of rework.

I don't mean to be negative though - it's a great piece of software, it gets the results, and that's what matters! Thank you for this project, I just set it up for my father-in-law and it works great!

ehpersonal38, 2023-05-20:
Heh, the double edged sword of open sourced projects. Would rather have it this way though than a closed source project where the only people finding the vulnerabilities are the bad guys.

lloydbayley, 2023-05-21:
Wow!
Anyway...Thanks very much @gibbypip for your message. Much-appreciated.
The matter is now in-hand.
If you would like to close this issue, that's fine!

gibbypip, 2023-05-22:
Right on! Appreciate the communications so far. Let me know via the established channels if you need anything else. Keep up the great work everyone!

closed 2023-05-22 by gibbypip

lloydbayley, 2023-05-22:
Thanks again for your kind words. :)