mcguirepr89/BirdNET-Pi #923: BirdNET-Analyzer model - which birds are included?

dl5dla, 2023-05-23:
Hi,
first of all thank you very much for this great software. 

It looks like that the "Halsbandsittich" (Psittacula krameri) is not detected here. As I am using the "BirdNET-Analyzer model", I wondering, whether this bird is included in the list of covered birds. Can I somewhere check, which birds are handled by the BirdNET-Analyzer?
Thanks.
Peter

nkwsy, 2023-05-23:
Should be, in `BirdNET-Pi/model/labels.txt`.  

Just went through this, @ehpersonal38 suggested to do `$HOME/BirdNET-Pi/scripts/install_language_label_nm.sh -l en && $HOME/BirdNET-Pi/scripts/restart_services.sh` which fixed my problem. 

The command `cat ~/BirdNET-Pi/model/labels.txt | grep "Psittacula` outputs:
```
Psittacula alexandri_Red-breasted Parakeet
Psittacula columboides_Malabar Parakeet
Psittacula cyanocephala_Plum-headed Parakeet
Psittacula eupatria_Alexandrine Parakeet
Psittacula krameri_Rose-ringed Parakeet```

dl5dla, 2023-05-23:
Thanks for your reply. I just played a recorded bird song close to the microphone, and in the log file the bird was recognized:

May 23 18:10:56 birdnetPI python3[2104063]: READING AUDIO DATA... DONE! READ 10 CHUNKS.
May 23 18:10:56 birdnetPI python3[2104063]: Date-time: 2023-05-23 18:10:24
May 23 18:11:08 birdnetPI birdnet_analysis.sh[439413]: 0.0;3.0-('Turdus merula_Amsel', 0.07858646)
May 23 18:11:08 birdnetPI birdnet_analysis.sh[439413]: 3.0;6.0-('Turdus merula_Amsel', 0.081180185)
May 23 18:11:08 birdnetPI birdnet_analysis.sh[439413]: 6.0;9.0-('Turdus viscivorus_Misteldrossel', 0.09707422)
May 23 18:11:08 birdnetPI birdnet_analysis.sh[439413]: 9.0;12.0-('Turdus merula_Amsel', 0.17348407)
May 23 18:11:08 birdnetPI birdnet_analysis.sh[439413]: 12.0;15.0-('Engine_Engine', 0.11502921)
May 23 18:11:08 birdnetPI birdnet_analysis.sh[439413]: 15.0;18.0-('Turdus merula_Amsel', 0.31662804)
May 23 18:11:08 birdnetPI birdnet_analysis.sh[439413]: 18.0;21.0-('Coragyps atratus_Rabengeier', 0.07120131)
**May 23 18:11:08 birdnetPI birdnet_analysis.sh[439413]: 21.0;24.0-('Psittacula krameri_Halsbandsittich', 0.59118605)
May 23 18:11:08 birdnetPI birdnet_analysis.sh[439413]: 24.0;27.0-('Psittacula krameri_Halsbandsittich', 0.7410629)
May 23 18:11:08 birdnetPI birdnet_analysis.sh[439413]: 27.0;30.0-('Psittacula krameri_Halsbandsittich', 0.8258372)**

But it's not shown in the Overview nor the Today's Detections. Not sure what the numbers in the log mean? Are these a hint that the recognition is outside the confidence?
Could it be that this bird is derated because this specie is not to be expected to live in Germany?

ehpersonal38, 2023-05-23:
> Could it be that this bird is derated because this specie is not to be expected to live in Germany?

Exactly, the model is detecting it with a confidence of 0.6-0.8, but then the post processing is saying "nope, the bird is not detected in this region frequently enough so I'm ignoring it." You can edit this behavior by decreasing the Species Occurence Frequency Threshold in Tools -> Settings.

ehpersonal38, 2023-05-23:
[image]

dl5dla, 2023-05-23:
Ok understood, thanks. So I have to play a little bit with the threshold. On the "more info" page it is mentioned, that
"a species needs to, on average, be seen on at least 3% of historically submitted eBird checklists"
and
"So, the lower the threshold, the rarer the species it will include."
That's confusing for me: So, if I reduce the value e.g. to 0,01 (1%), I would expect, that is more likely to get the bird included (?). 
But, as in Germany there are only some thousands of these birds (mainly in the big cities) I expect that detection might be negative anyway, no matter how I change the threshold (?).

ehpersonal38, 2023-05-23:
Yes you're right, 0.01 will include more species. If you update your system with Tools -> System Controls -> Update (I just pushed a bug fix), then try the Species List Tester located at Tools -> Model -> Species Occurance Frequency Threshold -> More Info, you can play around with the threshold until you get a species list that includes all the species you want to be detected and doesn't include the ones you don't want to be.

dl5dla, 2023-05-24:
Great, thank you. The species list tester now gave me a threshold value of about 0.02 to include the bird. When I understand correctly, this value is some kind of a moving target, so it can happen that I have to adapt the threshold in the future in case that the percentage of the bird decreases in the historical data.
As reducing the threshold in my understanding increases the probability of false positives, an alternative methode might be to also manage an "Included species list", where I could enter "my" bird and overrule the global decision of the model (?).
Again, thank you very much.

closed 2023-05-25 by ehpersonal38