mcguirepr89/BirdNET-Pi #927: Potential issue with new feature: "Added possibility in Spectrogram view to activate frequency shifting"

croisez, 2023-05-24:
Recently, I proposed a [pull request](https://github.com/mcguirepr89/BirdNET-Pi/pull/912/commits/04dae073f6f2bfed58389bb7d697a38010ffa42c), which is now merged.

@ehpersonal38 [reported to me](https://github.com/mcguirepr89/BirdNET-Pi/pull/912#issuecomment-1557491160) that he was experiencing problem of reconnection on his RPI between the audio control of the frontend and the livestream service.
Sometimes for him, at the end of the freq_shift toggling process, the audio control is not always reconnecting, he said.

On my setup, unfortunatelly, this is not happening, so I cannot debug this issue myself.

The idea of Ethan is to set more time as argument to the SetTimeout() function. See https://github.com/mcguirepr89/BirdNET-Pi/pull/912/commits/04dae073f6f2bfed58389bb7d697a38010ffa42c#diff-c901dd8212b39fc5b368439e21b44452aa0f39f8adf686069da8a69224f858c3 on line 343.

I would like to ask you all, to report here your own experience concerning this issue, which will be valuable to resolve this potential bug.
Thank you.
Louis.

lloydbayley, 2023-05-25:
Well, this is weird. I can't actually get the options to come up. Clicking on Spectrogram just loads the spectrogram. However, if I use chrome, the options display for a fleeting second and then the spectrogram loads over the top and all options are gone. Basically, I cannot access the options in any way, shape or form. Am trying to test things and this seems to be a different issue to yours but it making it impossible to test. Will keep trying.

croisez, 2023-05-25:
> Well, this is weird. I can't actually get the options to come up. Clicking on Spectrogram just loads the spectrogram. However, if I use chrome, the options display for a fleeting second and then the [...]

What is your configuration?
type of RPI
type of Browser, OS.

lloydbayley, 2023-05-25:
We've had browser-based peculiarities before but do the testing of commits and code on all sorts of browsers and machines, so there's not really a 'type' I use.

ehpersonal38, 2023-05-25:
Hm, I can't replicate this one either now. Okay to close this one if you want.

croisez, 2023-05-25:
okay

jaredb7, 2023-05-26:
I couldn't reproduce this reliably either (on a x86 install & with RTSP streams) but the times it did happen /stream returns a 404 because it isn't up yet. I increased the time-out up to 4 seconds and that made it more consistent. On slower devices that might need to to be tweaked a little higher but around 4 seconds might be a good start if that's what you were thinking.

croisez, 2023-05-29:
Just proposed a pull request, for adding configuration of this timeout value.
Default value for this variable set to 4000 ms.
https://github.com/mcguirepr89/BirdNET-Pi/pull/937