mcguirepr89/BirdNET-Pi #929: Species Stats page not loading (Firefox and Chrome)

dennyChandler, 2023-05-25:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
It seems that the "Species Stats" service is in a restart-cycle on the latest version. Navigating to it on the site produces no data or maybe a white screen, logs in systemctl state the service has started and is running, but cycles within a minute.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'birdnet webpage'
2. Click on 'Species Stats'
4. Page is blank

**Expected behavior**
Expect to see the Species Stats page when navigating to it

**Screenshots**
[image]

**Additional context**
Add any other context about the problem or your installation here.

**Your build**
The hardware on which BirdNET-Pi is running goes here.

**Code or log snippets**
```
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed May 24 09:35:39 2023 from 192.168.1.159
denny@birdnet:~/BirdNET-Pi$ sudo service birdnet_stats status
● birdnet_stats.service - BirdNET Stats
     Loaded: loaded (/home/denny/BirdNET-Pi/templates/birdnet_stats.service; en>
     Active: active (running) since Thu 2023-05-25 12:04:44 CDT; 1s ago
    Process: 1129177 ExecStartPre=/bin/sleep 60 (code=exited, status=0/SUCCESS)
   Main PID: 1129587 (streamlit)
      Tasks: 1 (limit: 1599)
        CPU: 36ms
     CGroup: /system.slice/birdnet_stats.service
             └─1129587 /home/denny/BirdNET-Pi/birdnet/bin/python3 /home/denny/B>

May 25 12:03:44 birdnet systemd[1]: Starting BirdNET Stats...
May 25 12:04:44 birdnet systemd[1]: Started BirdNET Stats.
```
______________________________________________________________________________________________________________________________

went and re-checked status at end of write up and here are the logs now:
```
denny@birdnet:~/BirdNET-Pi$ sudo service birdnet_stats status
● birdnet_stats.service - BirdNET Stats
     Loaded: loaded (/home/denny/BirdNET-Pi/templates/birdnet_stats.service; en>
     Active: activating (start-pre) since Thu 2023-05-25 12:06:07 CDT; 13s ago
Cntrl PID: 1130158 (sleep)
      Tasks: 1 (limit: 1599)
        CPU: 2ms
     CGroup: /system.slice/birdnet_stats.service
             └─1130158 /bin/sleep 60

May 25 12:06:07 birdnet systemd[1]: Starting BirdNET Stats...
```

dennyChandler, 2023-05-25:
To clarify: All other functionality is intact. Just this page/service seem to have an issues.

Absolutely love the project!

ehpersonal38, 2023-05-25:
I did add a new caching feature the other day, maybe its not supported on your version of Streamlit. Can you check the version and report back?
`~/BirdNET-Pi/birdnet/bin/pip3 show streamlit | grep Version`

tom-heisenlin, 2023-05-26:
I agree with dennyChandler: very nice project - thank you very much

my findings:

birdnet@birdnetpi:~/BirdNET-Pi/birdnet/bin$ pip3 show streamlit | grep Version
WARNING: Package(s) not found: streamlit

and

birdnet@birdnetpi:/$ sudo find / -iname "streamlit"
find: ‘/proc/1256779’: No such file or directory
find: ‘/proc/1257358’: No such file or directory
find: ‘/proc/1257393’: No such file or directory
find: ‘/proc/1257398’: No such file or directory
/home/birdnet/BirdNET-Pi/birdnet/lib/python3.9/site-packages/streamlit
/home/birdnet/BirdNET-Pi/birdnet/bin/streamlit
birdnet@birdnetpi:/$

but in birdnet I can see streamlit....

[image]

lloydbayley, 2023-05-26:
I know why you can't find it. We are running in a Python environment, so it is absolutely essential that you run commands withing the environment. The command that @ehpersonal38 gave was:

```
 ~/BirdNET-Pi/birdnet/bin/pip3 show streamlit | grep Version
```
Which ensures that it runs within the environment that Birdnet-Pi establishes.

```
pip3 show streamlit | grep Version
```
is running outside of the environment and therefore will return invalid results.

Give it a try by using the full command.

tom-heisenlin, 2023-05-26:
.... I was struggeling with the not known command to me.... sorry for that
here is what I found:

birdnet@birdnetpi:/$  ~/BirdNET-Pi/birdnet/bin/pip3 show streamlit | grep Version
Version: 1.19.0
birdnet@birdnetpi:/$

lloydbayley, 2023-05-26:
Absolutely no need to apologise. This is how we all learn.
That version is the same as the one I have, so it would seem normal.
I'm sure @ehpersonal38 will be in touch when he sees your results.

ehpersonal38, 2023-05-26:
Thanks for that Lloyd! Very helpful!

@tom-heisenlin I'm going to have you try updating your BirdNET-Pi to a different branch where I've removed the caching function I added last week. We will see if this makes any difference. Here is the command you can copy/paste and run:

`~/BirdNET-Pi/scripts/update_birdnet.sh -b streamlit-without-cache`

If it runs correctly you should see something like this in your console (buried among a lot of other messages) indicating that the updates were applied.

[image]

Then you can try to see if Species Stats returns anything but a blank page.

tom-heisenlin, 2023-05-26:
applied:
~/BirdNET-Pi/scripts/update_birdnet.sh -b streamlit-without-cache

result: no Species Stats :(
I cannot see what you indicated above - here is the output;

Linux birdnetpi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch6                                                                                                   4

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri May 26 21:12:00 2023
birdnet@birdnetpi:~/BirdNET-Pi$ ~/BirdNET-Pi/scripts/update_birdnet.sh -b streamlit-without-cache                                                                                  ++ sudo -u birdnet git -C /home/birdnet/BirdNET-Pi rev-parse HEAD
++ set +x
+ sudo -u birdnet git -C /home/birdnet/BirdNET-Pi reset --hard
HEAD is now at 049b62d add flowchart
+ set +x
+ sudo -u birdnet git -C /home/birdnet/BirdNET-Pi fetch origin streamlit-without-cache
From https://github.com/mcguirepr89/BirdNET-Pi
 * branch            streamlit-without-cache -> FETCH_HEAD
+ set +x
+ sudo -u birdnet git -C /home/birdnet/BirdNET-Pi switch -C streamlit-without-cache --track origin/streamlit-without-cache
fatal: invalid reference: origin/streamlit-without-cache
+ set +x
+ sudo -u birdnet git -C /home/birdnet/BirdNET-Pi diff --stat 049b62d6aed9c3a68ec183969f79828ebbaa1291 HEAD
+ set +x
+ my_dir=/home/birdnet/BirdNET-Pi/scripts
+ sudo systemctl stop birdnet_server.service
+ sudo systemctl stop birdnet_recording.service
+ services=(chart_viewer.service spectrogram_viewer.service icecast2.service extraction.service birdnet_recording.service birdnet_log.service birdnet_stats.service)
+ for i in "${services[@]}"
+ sudo systemctl restart chart_viewer.service
+ for i in "${services[@]}"
+ sudo systemctl restart spectrogram_viewer.service
+ for i in "${services[@]}"
+ sudo systemctl restart icecast2.service
+ for i in "${services[@]}"
+ sudo systemctl restart extraction.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_recording.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_log.service
+ for i in "${services[@]}"
+ sudo systemctl restart birdnet_stats.service
+ sudo systemctl start birdnet_server.service
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ for i in {1..5}
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
+ sleep 5
+ systemctl is-active --quiet birdnet_server.service
+ grep 5050 /dev/fd/63
++ netstat -tulpn
tcp        0      0 127.0.0.1:5050          0.0.0.0:*               LISTEN      2230469/python3
+ status=0
+ ((  status != 0  ))
+ sudo systemctl restart birdnet_analysis.service
birdnet@birdnetpi:~/BirdNET-Pi$

ehpersonal38, 2023-05-26:
Try this then:
```
git remote rm origin
git remote add origin https://github.com/mcguirepr89/BirdNET-Pi
git fetch --all
git pull
git branch --set-upstream-to=origin/main main
git checkout -f main
~/BirdNET-Pi/scripts/update_birdnet.sh -b streamlit-without-cache
```

tom-heisenlin, 2023-05-28:
No success - unfortunately :(
thank you anyhow for the support.

Now I did a full new install, starting with the RaPi Imager.....
before I saved the database.
afterr new install and importing the database I do have Species Stats available !
 
message in the overview:
Your system is currently processing a backlog of audio. This can take several hours before normal functionality of your BirdNET-Pi resumes.
so, I will let the system calm down before further testing...

Tom

lloydbayley, 2023-05-28:
Yep, that's normal. It will disappear on it's own. Fingers crossed for a well-behaved system!

dennyChandler, 2023-05-29:
Sorry for my absence, work and family stuff got busy. Is there anything I can do to help at this point?

tom-heisenlin, 2023-05-31:
Seems to work fine now.
re-installing from 0 solved the problem
tnx for all support
Tom

ehpersonal38, 2023-06-01:
I'm not sure where to go from here, a reimage is my best advice unfortunately.

dennyChandler, 2023-06-08:
Word, I'll give that a shot when I have a chance. Thanks for looking into it.

closed 2023-06-08 by dennyChandler