mcguirepr89/BirdNET-Pi #933: Scroll Bar Disappears on Chrome

oldgristmill, 2023-05-27:
New install on a RP4 and there are no vertical scrollbars on any of the pages served by Birdnet-Pi. Same issue seems to exist with Microsoft Edge. Haven't checked on Safari or other browsers that use a different rendering engine.

Can anyone verify this problem? Is it the browser or the structure of the pages being served?

lloydbayley, 2023-05-28:
When you say scroll bar disappears, was it ever there? Did it suddenly disappear?
Also, does the page scroll if you use the mouse wheel or drag etc etc?
I'm not able to replicate this on any browser/operating system. Very odd!

jaredb7, 2023-05-28:
I don't think I've noticed a scrollbar, but when I check... it's only there for Firefox but not Chrome or Edge. 
Not sure if this is by design?, the no scrollbars I mean.

Chrome and Edge use the same rendering engine so it could be a browser thing?

[image]
[image]
[image]

oldgristmill, 2023-05-28:
@jaredb7, thanks for confirming that I'm not crazy! I have no idea how to figure out whether it's a bug with the Chrome render engine or with the html/css/js being output by birdnet-pi.

ehpersonal38, 2023-05-30:
Yes all scrollbars are disabled, but on Firefox and IE it wasn't working. I've just done a commit which removes them. Thanks!

ehpersonal38, 2023-06-02:
Patrick removed them long ago, probably for aesthetic reasons. But I now agree with you that they can be quite useful so I've added them back in a new commit - 5336736b38143d7f73dccd707de126fd53b8efad.