mcguirepr89/BirdNET-Pi #947: Failed to unzip the labels text file due to a wrongly determined file path

joberreiter, 2023-06-08:
Failed to unzip the model labels "labels_l18n.zip" at initial install try, see code below. Full installation log is attached.

**Expected behavior**
Successfully extract the labels_l18n.zip at (first) installation. 

**Additional context**
The file "labels_l18n.zip" is available and in the correct location, the unzip executable is available, access and file rights are correct, but the unzipper is started with a wrong file path ("/home/radxa/BirdNET-Pi/model/labels_l18n.zip"), although the install location of birdNET-Pi was correctly determined ("/home/jochen/BirdNET-Pi").
Correct archive file path would be "/home/jochen/BirdNET-Pi/model/labels_l18n.zip", and the file can be found at this location.

For clarification: The user "radxa" exists and is the root user, but is not the user BirdNET-Pi was installed under.

**Your build**
* BirdNET-Pi v0.13, commit [92d083a](https://github.com/mcguirepr89/BirdNET-Pi/commit/92d083a14c74a8051616e5d6ef351713d5888d1c)
* Linux rock4se 5.10.110-8-rockchip #f9d1b1529 SMP Fri May 12 14:07:29 UTC 2023 aarch64 GNU/Linux
* Radxa ROCK 4 SE powered by RK3399-T with a dual core Cortex-A72 @1.5GHz and quad core Cortex-A53@1GHz ARM processor, 4GB 64bit LPDDR4.
* microphone MOVO USB-M1

**Code or log snippets**
```
[...]
+ cd /home/jochen/BirdNET-Pi/scripts
+ ./install_language_label.sh -l en
unzip:  cannot find or open /home/radxa/BirdNET-Pi/model/labels_l18n.zip, /home/radxa/BirdNET-Pi/model/labels_l18n.zip.zip or /home/radxa/BirdNET-Pi/model/labels_l18n.zip.ZIP.
unzip:  cannot find or open /home/radxa/BirdNET-Pi/model/labels_l18n.zip, /home/radxa/BirdNET-Pi/model/labels_l18n.zip.zip or /home/radxa/BirdNET-Pi/model/labels_l18n.zip.ZIP.
+ exit 0
```
[installation-2023-06-05.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/11687436/installation-2023-06-05.txt)

joberreiter, 2023-06-08:
I briefly scanned the scripts -- are you assuming that the user that has installed BirdNET-Pi always have uid 1000?
e.g.:
``` clear_all_data.sh
[...]
USER=$(awk -F: '/1000/ {print $1}' /etc/passwd)
[...]
```
This may cause serious troubles. Maybe we need a configuration option to specify the user under that BirdNET-Pi runs?

ehpersonal38, 2023-06-08:
My official answer is that this software is intended to be installed on a fresh system, as referenced by the installation guide and disclaimer that "installing BirdNET-Pi on top of other servers is not supported." It's just best for everybody, there will be so many little edge cases and problems (like this one) that will crop up when installing on a system has been tinkered with.

closed 2023-06-08 by ehpersonal38