mcguirepr89/BirdNET-Pi #949: Flickr images not corresponding to species

joberreiter, 2023-06-08:
With my installation, the flickr images are completely screwed up -- maybe this happens only on a non-english instance?
Flickr API key is set, no problem with the retrieval of a flickr photo URL itself, it seems that they are just the wrong one's.
It seems that the shown images have generic captions like "bird", "birding", etc.
Blacklisting the image works, but does not improve the situation (just another non-corresponding image comes up).

**Expected behavior**
(Almost) correct images corresponding to the species name.

**To Reproduce**
Although on level "info", the log shows no useable information on e.g. the requested flickr search URL etc.
Please advise what log / which file I have to provide for that info.

**Screenshots**
[image]

**Your build**
* BirdNET-Pi v0.13, commit [92d083a](https://github.com/mcguirepr89/BirdNET-Pi/commit/92d083a14c74a8051616e5d6ef351713d5888d1c)
* Linux rock4se 5.10.110-8-rockchip #f9d1b1529 SMP Fri May 12 14:07:29 UTC 2023 aarch64 GNU/Linux
* Radxa ROCK 4 SE powered by RK3399-T with a dual core Cortex-A72 @1.5GHz and quad core Cortex-A53@1GHz ARM processor, 4GB 64bit LPDDR4.
* microphone MOVO USB-M1

[diagnostic_output_2023-06-08.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/11689473/diagnostic_output_2023-06-08.txt)

ehpersonal38, 2023-06-08:
Thanks for this bug report! I made a mistaken commit with the 2.4 update which caused your `model/flickr_labels.txt` file to get messed up. To repair your installation you will need to follow these steps:
1. Go to **Tools** > **Settings**.
2. Change the **Model** to the model that you are not using.
3. Click **Update Settings** at the bottom of the page.
4. Change the **Model** back to the original one (the one you want to use).

closed 2023-06-08 by ehpersonal38

joberreiter, 2023-06-11:
Have done also a "git pull" to be up to date with your changes, but that has not fixed the issue.
The situation two 2 days after:
[image]

Additionally, I now get weird detections with a very high confidence rate, but this may not correlate with this issue.

joberreiter, 2023-06-11:
Issue resolved, was a local problem with the wrong user id.

BirdNET-Pi assumes that the uid is 1000 (as it is on a fresh raspberry installation), and I had created an extra user for birdnet.

For sake of completeness, this was how I solved this on my machine:

 `grep -rl '/1000/' ~/BirdNET-Pi/ | xargs sed -i 's=/1000/=/$(id -u)/=g'`

commit caaa70b6 (2023-06-08): fix #949 for future