mcguirepr89/BirdNET-Pi #951: BirdNET Installation Fails with "Permission Denied" Error for install_language_label_nm.sh on Raspberry Pi 4

Rebero, 2023-06-10:
Subject: BirdNET Installation Fails with "Permission Denied" Error on Raspberry Pi

Product: BirdNET-Pi
Operating System: Raspberry Pi Lite 64 bit 
Hardware: Raspberry Pi 4 

Description:
I am trying to install BirdNET on my Raspberry Pi using the official installation guide. When running the provided installation script (https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh). However, the installation fails with a "Permission denied" error.

/home/reneeber/BirdNET-Pi/scripts/install_birdnet.sh: line 40: ./install_language_label_nm.sh: Permission denied
+ exit 1
The installation exited unsuccessfully

Steps to Reproduce:
Ran the provided installation command: curl -s https://raw.githubusercontent.com/mcguirepr89/BirdNET-Pi/main/newinstaller.sh | bash
Received the "Permission denied" error message.

What I've Tried:
Made sure that the installation script has the necessary execute permissions (chmod +x).
Attempted to run the installation command as the root user using sudo, which resulted in an error message saying the script should be run as a non-root user.
Checked that my user is in the sudo group and the permissions of my Bash executable.
Despite these steps, I am still unable to install BirdNET and continue to receive the "Permission denied" error. Any assistance in resolving this issue would be greatly appreciated.

mikedepalma, 2023-06-10:
Can confirm I'm experiencing the same issue. Fresh install, Pi OS Lite 64 bit, first command run was the install script. Bails during the language label install.

rototom, 2023-06-11:
as a dirty workaround:
''chmod +x BirdNET-Pi/scripts/install_language_label_nm.sh ''
does the trick

masyfil, 2023-06-11:
Same issue on a raspberry pi 3b+. Chmod on sh didn't work for me.. the installer script don't run a second time.

joberreiter, 2023-06-11:
@masyfil: Running `chmod +x ~/BirdNET-Pi/scripts/*.sh` once should do the trick without the need to re-run the installer.

masyfil, 2023-06-11:
> @masyfil: Running `chmod +x ~/BirdNET-Pi/scripts/*.sh` once should do the trick without the need to re-run the installer.

thanks!

Rebero, 2023-06-11:
After running the installer with the new rights enabled I run successfully. Now I am getting the caddy error screen when going to birdnetpi.local after the reboot. Not sure where the issue is but guess this is still some bug. Any ideas? Thx
[image]

masyfil, 2023-06-11:
Do you use Firefox? I had the same Problem today... Try Chrome and check if it is http and not https. I think Firefox force https maybe. I am a new BirdNetPi user. Started today... So maybe someone else have a good idea, too.

Rebero, 2023-06-11:
thanks! Was using firefox but the issue doesn't persist with Chrome.

ehpersonal38, 2023-06-12:
Sorry for this, I meant to make the file executable but forgot.

closed 2023-06-12 by ehpersonal38

ehpersonal38, 2023-06-12:
Also the chmod above doesn't completely fix the issue, you still need to run the actual script with your language. Here's a one-liner which should resolve the issue for everyone:

```
language=$(grep 'DATABASE_LANG=' ~/BirdNET-Pi/birdnet.conf | awk -F '=' '{print $2}') && chmod +x ~/BirdNET-Pi/scripts/install_language_label_nm.sh && ~/BirdNET-Pi/scripts/install_language_label_nm.sh -l $language
```

krummrey, 2023-06-12:
I tried it on a RPi4 and got stuck at the same point. Did the chmod but still get the caddy screen. Chrome, Firefox, Safari, Brave, none worked