mcguirepr89/BirdNET-Pi #971: Selected Wrong User during Update

rezeero25, 2023-07-07:
https://github.com/mcguirepr89/BirdNET-Pi/blob/c85fa1e9280b79fadc2867416f5b25fea0aa30c2/scripts/update_birdnet_snippets.sh#L5C1-L5C48

After installing BirdNet-Pi on a machine with already existing user (e.g. "first_user", the update tries to find the BirdNET-Pi path in the home directory of the user "first_user" (/home/first_user/BirdNET-Pi/scripts/*.txt') and not in the path /home/birdnet/.

Is the reason for this a fixed UID "1000"?

joberreiter, 2023-07-08:
This is a known issue, see also #947.
According to @ehpersonal38, they won't fix it ...

joberreiter, 2023-07-08:
Try this to change it to the userid of the user running the command:
`grep -rl '/1000/' ~/BirdNET-Pi/ | xargs sed --quiet 's=/1000/=/$(id -u)/=g'`

You can replace `$(id -u)` with the desired user id of course.

rezeero25, 2023-07-09:
Thanks for your comment.
I'm worried that after an update these changes will be undone. so I did the following:

Use a root account, or the other ("first" user).

Check with 
~~~
id username
~~~

for your first user and birdnet user the UID and GID, and in which other groups is included.

Change the old user, which has the UID=1000 to e.g. 2000 and the UID from the birdnet user to 1000. For this you may have to stop all services from the birdnet, additionally I force quit all running processes and logged out the user
~~~
sudo pkill -u birdnet
~~~

To change UID and GID:
~~~
usermod -u 1000 birdnet
groupmod -g 1000 birdnet
~~~
to add user to different groups (e.g. the birdnet user didnt had permission for audio):

At the end the result of ~~~id birdnet~~~ should something like this:

uid=1000(birdnet) gid=1000(birdnet) Groups=1000(birdnet),4(adm),20(dialout),24(                                                  cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),104(input),                                                  106(render),108(netdev),999(spi),998(i2c),997(gpio),117(lpadmin)

However, I don't know if this will still correctly assign old read/write permissions for folders and files

Hopefully that helps other. And we can close this "issue".

Another issue with this topic: https://github.com/mcguirepr89/BirdNET-Pi/pull/909
(I only found the old topics because now i know for what to look/search for)

rezeero25, 2023-07-09:
> Run this once to change it to the userid of the user BirdNET is running under: `grep -rl '/1000/' ~/BirdNET-Pi/ | xargs sed --quiet 's=/1000/=/$(id -u)/=g'`
> 
> You can replace `$(id -u)` with the desired user id of course.

This will search and replace all "/1000/" in all files in BirdNET-Pi folder e.g. scripts/update_birdnet_snippets.sh

closed 2023-07-09 by rezeero25