mcguirepr89/BirdNET-Pi #973: "sox FAIL trim: Position 1 is behind the following position"

jgelsey, 2023-07-08:
BirdNet stoped analyzing yesterday with "`sox FAIL trim: Position 1 is behind the following position`" in the log.  I see "unable to reach network" in the output of `print_diagnostic_info.sh` (below) but a ping has no trouble reaching the mentioned network device.  `Tools->System Controls->Clear ALL Data` got detections started again.  FWIW using 7 rtsp streams for audio input (from Unifi cameras), no USB microphone.

```
birdnet:~$ cat print_diagnostic_info.txt 
========== caddy status ==========
● caddy.service - Caddy
     Loaded: loaded (/lib/systemd/system/caddy.service; enabled; vendor preset: enabled)
    Drop-In: /etc/systemd/system/caddy.service.d
             └─override.conf
     Active: active (running) since Sat 2023-07-08 08:20:32 PDT; 22min ago
       Docs: https://caddyserver.com/docs/
   Main PID: 1711 (caddy)
      Tasks: 9 (limit: 8755)
        CPU: 508ms
     CGroup: /system.slice/caddy.service
             └─1711 /usr/bin/caddy run --environ --config /etc/caddy/Caddyfile

Jul 08 08:20:32 birdnet caddy[1711]: {"level":"warn","ts":1688829632.3530626,"msg":"Caddyfile input is not formatted; run the 'caddy fmt' command to fix inconsistencies","adapter":"caddyfile","file":"/etc/caddy/Caddyfile","line":1}
Jul 08 08:20:32 birdnet caddy[1711]: {"level":"info","ts":1688829632.3585544,"logger":"admin","msg":"admin endpoint started","address":"localhost:2019","enforce_origin":false,"origins":["//localhost:2019","//[::1]:2019","//127.0.0.1:2019"]}
Jul 08 08:20:32 birdnet caddy[1711]: {"level":"info","ts":1688829632.3655381,"logger":"tls.cache.maintenance","msg":"started background certificate maintenance","cache":"0x40004091f0"}
Jul 08 08:20:32 birdnet caddy[1711]: {"level":"warn","ts":1688829632.368039,"logger":"http","msg":"server is listening only on the HTTP port, so no automatic HTTPS will be applied to this server","server_name":"srv0","http_port":80}
Jul 08 08:20:32 birdnet caddy[1711]: {"level":"info","ts":1688829632.377894,"logger":"tls","msg":"cleaning storage unit","description":"FileStorage:/var/lib/caddy/.local/share/caddy"}
Jul 08 08:20:32 birdnet caddy[1711]: {"level":"info","ts":1688829632.378084,"logger":"tls","msg":"finished cleaning storage units"}
Jul 08 08:20:32 birdnet caddy[1711]: {"level":"info","ts":1688829632.384936,"logger":"http.log","msg":"server running","name":"srv0","protocols":["h1","h2","h3"]}
Jul 08 08:20:32 birdnet caddy[1711]: {"level":"info","ts":1688829632.3893967,"msg":"autosaved config (load with --resume flag)","file":"/var/lib/caddy/.config/caddy/autosave.json"}
Jul 08 08:20:32 birdnet caddy[1711]: {"level":"info","ts":1688829632.390183,"msg":"serving initial configuration"}
Jul 08 08:20:32 birdnet systemd[1]: Started Caddy.

========== birdnet_analysis status ==========
● birdnet_analysis.service - BirdNET Analysis
     Loaded: loaded (/home/jon/BirdNET-Pi/templates/birdnet_analysis.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2023-07-08 08:20:17 PDT; 22min ago
   Main PID: 395 (bash)
      Tasks: 2 (limit: 8755)
        CPU: 1.016s
     CGroup: /system.slice/birdnet_analysis.service
             ├─ 395 bash /usr/local/bin/birdnet_analysis.sh
             └─3569 /home/jon/BirdNET-Pi/birdnet/bin/python3 /home/jon/BirdNET-Pi/scripts/analyze.py --i /home/jon/BirdSongs/StreamData/2023-07-07-birdnet-RTSP_3-10:18:25.wav --o /home/jon/BirdSongs/StreamData/2023-07-07-birdnet-RTSP_3-10:18:25.wav.csv --lat 48.6212 --lon -122.8459 --week 26 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7 --birdweather_id zEX9jPQnTXLxfS1g9FUxx39P

Jul 08 08:20:17 birdnet systemd[1]: Started BirdNET Analysis.
Jul 08 08:20:37 birdnet birdnet_analysis.sh[395]: Files loaded
Jul 08 08:20:38 birdnet birdnet_analysis.sh[395]: RECORDING_LENGTH set to 15
Jul 08 08:20:38 birdnet birdnet_analysis.sh[395]: /home/jon/BirdNET-Pi/birdnet/bin/python3 /home/jon/BirdNET-Pi/scripts/analyze.py --i /home/jon/BirdSongs/StreamData/2023-07-07-birdnet-RTSP_3-10:18:25.wav --o /home/jon/BirdSongs/StreamData/2023-07-07-birdnet-RTSP_3-10:18:25.wav.csv --lat 49.XX --lon -122.XX --week 26 --overlap 0.0 --sensitivity 1.25 --min_conf 0.7 --birdweather_id "IN_USE"

========== birdnet_log status ==========
● birdnet_log.service - BirdNET Analysis Log
     Loaded: loaded (/home/jon/BirdNET-Pi/templates/birdnet_log.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2023-07-08 08:20:17 PDT; 22min ago
   Main PID: 392 (gotty)
      Tasks: 14 (limit: 8755)
        CPU: 979ms
     CGroup: /system.slice/birdnet_log.service
             ├─   392 /usr/local/bin/gotty --address localhost -p 8080 -P log --title-format BirdNET-Pi Log birdnet_log.sh
             ├─300665 bash /usr/local/bin/birdnet_log.sh
             ├─300667 journalctl --no-hostname -q -o short -fu birdnet_analysis -ubirdnet_server -uextraction
             └─300669 sed s/Jul 08 //g;s/\/home\/jon\///g;/Line/d;/find/d;/systemd/d;s/ .*\[.*\]: /---/

Jul 08 08:32:40 birdnet gotty[392]: 2023/07/08 08:32:40 127.0.0.1:45668 200 GET /log/auth_token.js
Jul 08 08:32:40 birdnet gotty[392]: 2023/07/08 08:32:40 127.0.0.1:45668 200 GET /log/config.js
Jul 08 08:32:40 birdnet gotty[392]: 2023/07/08 08:32:40 127.0.0.1:45682 200 GET /log/css/xterm_customize.css
Jul 08 08:32:40 birdnet gotty[392]: 2023/07/08 08:32:40 127.0.0.1:45676 200 GET /log/js/gotty-bundle.js
Jul 08 08:32:40 birdnet gotty[392]: 2023/07/08 08:32:40 New client connected: 127.0.0.1:45692, connections: 1/0
Jul 08 08:33:08 birdnet gotty[392]: 2023/07/08 08:33:08 Connection closed by client: 127.0.0.1:45692, connections: 0/0
Jul 08 08:33:20 birdnet gotty[392]: 2023/07/08 08:33:20 127.0.0.1:45676 200 GET /log/
Jul 08 08:33:20 birdnet gotty[392]: 2023/07/08 08:33:20 127.0.0.1:45682 200 GET /log/auth_token.js
Jul 08 08:33:20 birdnet gotty[392]: 2023/07/08 08:33:20 127.0.0.1:45676 200 GET /log/config.js
Jul 08 08:33:20 birdnet gotty[392]: 2023/07/08 08:33:20 New client connected: 127.0.0.1:48282, connections: 1/0

========== birdnet_recording status ==========
● birdnet_recording.service - BirdNET Recording
[... 354 lines cut ...]
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
usbstream:CARD=Headphones
    bcm2835 Headphones
    USB Stream Output
usbstream:CARD=vc4hdmi0
    vc4-hdmi-0
    USB Stream Output
usbstream:CARD=vc4hdmi1
    vc4-hdmi-1
    USB Stream Output

========= Date and Time ==========
Sat  8 Jul 08:42:48 PDT 2023
===========================================
birdnet:~$ 
```

PeterMescher, 2023-10-09:
I've hit this one twice so far.  I'll collect more data next time if needed, but for whatever reason, Analyzer is reading in more data (in my case, 5 chunks instead of 10); if there's a successful detection in the latter half, this error occurs when it tries to write the resulting .mp3.

To temporarily fix this issue, you simply need to edit the offending .csv in StreamData-Analyzed to claim the detection occurred earlier in the file.  (You can also simply delete it, but this way you don't even lose the detection!)

I'll gather more data next time (the offending .csv and .wav, and the following .csv and .wav.)  Don't quote me on this, but I do think that the .wav was indeed twice as long as it should have been.

And this certainly will happen again, so I expect to have a re-create in the next day or two: I searched the logs, and "READ 10 CHUNKS" (signaling a too-long file) seems to occur once or twice an hour.  The next time there's a successful detection in the 2nd half of one of those, the bug will trigger again.

So I can be helpful, what would be useful to collect?  Besides the relevant journalctl entries (although they don't reveal anything interesting; it _looks_ like the sound file should only be 15 seconds long.)

Here's my diagnostic data:

[sox_fail_trim.txt](https://github.com/mcguirepr89/BirdNET-Pi/files/12842085/sox_fail_trim.txt)

PeterMescher, 2023-12-11:
Here's a script I put in place to work around the issue: (courtesy chatgpt... I dimly remember shell scripting from college a quarter-century ago... took me minutes instead of hours to write this!)  This looks for the oldest file of a particular type in a directory, and if it's still there 30 seconds later, delete it.  In our case, we want to delete a .csv file that's still there 30 seconds after creation, since it's supposed to process these in less than 15 seconds.  (After the deletion, the normal BirdNet scripts will clean up the whole directory once it gets caught up, including the over-sized .wav file.)

Obviously you'll want to modify the sleep time if your interval is different from 15 seconds.  (e.g. If running on a Pi Zero once a minutes, you'll want to sleep for two minutes.)

The script is stored in /root.

> #!/bin/bash
> 
> # Define the directory and file extension
> directory="/home/pi/BirdSongs/StreamData-Analyzed"
> extension=".csv"  # Change this to your desired file extension
> logfile="/root/csv_logfile.txt"  # Specify the path to your logfile
> 
> # Check if the directory exists
> if [ ! -d "$directory" ]; then
>   echo "Error: Directory does not exist.">> "$logfile"
>   exit 1
> fi
> 
> # Check if there are any files with the specified extension
> files=("$directory"/*"$extension")
> if [ ${#files[@]} -eq 0 ]; then
>   echo "No files with the specified extension found." >> "$logfile"
>   exit 0
> fi
> 
> # Find the oldest file
> oldest_file="${files[0]}"
> oldest_time=$(stat -c %Y "$oldest_file") 2>> /dev/null
> 
> for file in "${files[@]}"; do
>   file_time=$(stat -c %Y "$file") 2>> /dev/null
>   if [ "$file_time" -lt "$oldest_time" ]; then
>     oldest_file="$file"
>     oldest_time="$file_time"
>   fi
> done
> 
> # Wait for 30 seconds
> sleep 30
> 
> # Check if the oldest file is still present
> if [ -e "$oldest_file" ]; then
>   # Delete the oldest file
>   rm "$oldest_file"
>   echo "$(date '+%Y-%m-%d %H:%M:%S') Deleted the oldest file: $oldest_file" >> "$logfile"
> fi

And don't forget to chmod 755 csv_cleanup.sh to make it executable!

I put in a crontab entry of 
*/5 * * * * /root/csv_cleanup.sh 

and that seems to have stopped the problem cold.  I had it set to every 30 minutes, but if the problem happened twice in a half-hour, it still got stuck.

Obviously, you will not have recordings for these particular birds in your files, though birdweather still runs on the detection.