mcguirepr89/BirdNET-Pi #988: Constant Identification of wrong bird

talkingdragon, 2023-07-20:
Im not to this, but im getting a high probability of this:
14:01:17---15.0;18.0-('Spizella pallida_Clay-colored Sparrow', 0.81639445)
Its right and wrong at the same time, its a Sparrow, just not that one, as im in Spain and we don't have that we have the typical house Sparrow
what can i do tell it what it should be?
the lat,long is correct, which means it give me 0 results for it as its not form this area
I looked in settings to see if any additional location options, but could find none
on the birdnet app it uses the location to find a more probable option

hope you can help me :)

Sarge2, 2023-08-06:
Folks, I'm new to BirdNet and have a similar problem. I've run the server for a day now and gotten IDs on Victorins Warbler (primary range South Africa), Laced Woodpecker (primary range Southeast Asia), and Australian Masked-Owl (primary range Australia). 
My latitude, longitude are set to 42.305 , -71.642, so I'm in the Northeast of the US. and none of these birds should be coming up. 
I'm using a cheap USB microphone from SEED Studio, so I can say the audio quality is not very good but should I even be getting these matches (all rated to >70% and sometimes up to 99% confidence). Do I have something in the software set incorrectly? 
Thx, Sarge

fede2cr, 2023-08-30:
Of course that a good microphone is a good thing to have, but something that is working for me, is changing the model that is used to BirdNET_GLOBAL_6K_V2.4_Model_FP16 (2023), by going to Tools -> Settings.

The main reason is because this model (click on more info in "Species Occurence Frequency Threshold") looks up if the bird has been reported in eBird near your area, and only shows up the species if it is local to your area.

Remember to configure your LAT/LONG in this Settings page as well for this to work.

Sarge2, 2023-09-04:
Thanks Alvaro. I will try that.

closed 2023-10-25 by talkingdragon