mcguirepr89/BirdNET-Pi #1000: Apprise notification to Discord fails with certain characters in the Title and Body fields

avaryx, 2023-08-08:
**Describe the bug**
I am attempting to format messages issued to a Discord WebHook. When composing the messages, I found that the apostrophe (`'`) character causes the process to fail, with the following in the logs:
[image]

My discovery was made when I set these values:
**Notification Title:** Oh look, a $comname!
**Notification Body:** I prefer to call it a $sciname but I know that's not as familiar for some. I'm $confidencepct% sure of this one.

I eventually settled on the following:
**Notification Title:** Oh look, a $comname!
**Notification Body:**  I prefer to call it a $sciname but I know that is not as familiar. I am $confidencepct% sure of this one.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Tools, then Settings
2. Set up an Apprise notification
3. In the Notification Title or Notification Body field, include an apostrophe (`'`)
4. Click Send Test Notification and notice you do not receive the expected output:
[image]
Instead, you will see the sendAppriseNotifications error in the View Log interface (as seen in the screenshot posted at the beginning of this report).

**Expected behavior**
The message should be sent with apostrophes (`'`) in place. These characters should be escaped properly.

**Screenshots**
See above.

**Additional context**
Add any other context about the problem or your installation here.

**Your build**
```
OS: Debian GNU/Linux 11 (bullseye) aarch64
Host: Raspberry Pi 4 Model B Rev 1.4
Kernel: 6.1.21-v8+
Uptime: 18 mins
Packages: 835 (dpkg)
Shell: bash 5.1.4
Terminal: /dev/pts/0
CPU: BCM2835 (4) @ 1.800GHz
Memory: 546MiB / 7812MiB
```

**Code or log snippets**
See above.

avaryx, 2023-08-08:
Also, I did make sure to run an update of BirdNET-Pi during my validation of what I found here, and it happens while on the most recent release:

```
++ sudo -u birdpi git -C /home/birdpi/BirdNET-Pi rev-parse HEAD
++ set +x
+ sudo -u birdpi git -C /home/birdpi/BirdNET-Pi reset --hard
HEAD is now at c85fa1e Merge pull request #965 from srd424/serverpy-fix-2
+ set +x
+ sudo -u birdpi git -C /home/birdpi/BirdNET-Pi fetch origin main
From https://github.com/mcguirepr89/BirdNET-Pi
 * branch            main       -> FETCH_HEAD
+ set +x
+ sudo -u birdpi git -C /home/birdpi/BirdNET-Pi switch -C main --track origin/main
Reset branch 'main'
Branch 'main' set up to track remote branch 'main' from 'origin'.
Your branch is up to date with 'origin/main'.
```

avaryx, 2023-08-08:
I should have tried this sooner, but double-quote (`"`) also seems to have this same effect. It was able to dispatch the test message, but after that the logs show the same error and audio processing is stopped.

avaryx, 2023-08-08:
Additionally, the `$listenurl` replacement can be affected by characters surrounding it.

For example, this string: `Also known as $sciname. I am $confidencepct% sure of this one. ($listenurl)`
Generates a `$listenurl` hyperlink ending in `.mp3\` - note the extra backslash (`\`). If you leave off the close parenthesis (`)`), the URL ends in `.mp3`

srd424, 2023-08-11:
I've had a quick look at the code and it's not obvious to me why this is happening - the only bit that seems to be obviously not escaping stuff properly is the actual test notification handling itself. You could try editing the configuration file by hand and see if it still happens? The would rule the php side of things in or out, at least.

avaryx, 2023-08-13:
> I've had a quick look at the code and it's not clear to me why this is happening - the only bit that seems to be obviously not escaping stuff properly is the actual test notification handling itself. You could try editing the configuration file by hand and see if it still happens? The would rule the php side of things in or out, at least.

Yeah that's a good idea. I'm not familiar with where that configuration is, however. What file do I need to modify to test this?

srd424, 2023-08-16:
`/home/pi/BirdNET-Pi/birdnet.conf` usually, I think. There should be a symlink to it at `/etc/birdnet/birdnet.conf` as well.