mcguirepr89/BirdNET-Pi #1001: Waterproof USB microphone

jeffehobbs, 2023-08-09:
I'm setting up a BirdNetPi instance for my parents, who I believe will really enjoy this project. However, they have a climate controlled house and have windows closed fairly often to always. The best recording setup would be a waterproof USB mic that I could hang out a window, but this is surprisingly hard to find.

Does anyone have any experience with waterproof recording USB microphones? Price honestly no object, although obviously less expensive the better.

blogdemoi, 2023-08-11:
I am using an EM272 microphone and very happy with it. I know it does exist a rugged version of this microphone, not waterproof but it may be useful to you. Their site is currently under maintenance, so check in a few days: https://micbooster.com/modules-with-primo-microphone-capsules/113-rugged-mount-primo-em172-mono.html

If you do a search on the Internet there are a few waterproof or weatherproof microphones but I don't know anything about their quality.

oliverk71, 2023-08-11:
In most cases, it is sufficient to protect the microphone from rain. This can be easily done by many ways, depending on the actual situation.

simon-ball, 2023-08-14:
I have used a LM40 capsule outdoors for about 8 months now, including through the recent storm Hans. It's not outdoor rated. It gets a bit of shielding from nearby bushes, but I've not assembled any kind of proper shelter. 

I've not noticed any long-term degredation due to water ingress, although it may happen at some point. However, if it gets water on/in the aperture, it can't sense above about 500Hz until the water evaporates (usually 2-4 hours).