mcguirepr89/BirdNET-Pi #1011: Latest updates causing several problems on several installations!

Svardsten53, 2023-08-20:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
One or a few of the updates in August have caused us to have problems on at least three of the installations in Sweden.

The image at the bottom of the first page (Custom Image in Advanced Settings) is no longer displayed, only a placeholder is visible. If the image is fetched from external server it works.

On two of the systems, there is a very long response time to Settings on one system and Log on the other.

On one of the systems crontab suddenly disappeared! I know, it sounds very strange, could be something else wrong, but happened after the update.

**To Reproduce**
Steps to reproduce the behavior:
Please see the bottom of this page: [https://lobirds.ddns.net/](url) where the bottom image is missing.
This installation is also the one with long response time to the log ("Logg").

**Expected behavior**
Normal behavior as it was before the update when everything worked very well.

**Your build**
Raspberry Pi 4B 4GB, all updates installed on all of the installations.

Svardsten53, 2023-08-25:
The bottom image works if you specify an external link.

ehpersonal38, 2023-08-26:
Hmm.. I looked at the 3 commits in the past month and they all are fairly harmless and I don't see why they would cause the issues you describe. Unfortunately I don't have an active BirdNET-Pi installation at the moment, so I am going to have to rely on others' investigations and input to fix this.

Svardsten53, 2023-09-04:
The problem of not being able to display the image at the bottom of the front page is temporarily solved by placing the image on an external server.

However, on one of my installations it takes >30 seconds to get the log. Anyone have any ideas on how to troubleshoot or fix this? Both installations have the same level both on OS and BirdNET-Pi.