mcguirepr89/BirdNET-Pi #1019: Species stats not showing when birdnet-pi behind a reverse proxy

tbard1, 2023-09-05:
### PLEASE READ THE DOCUMENTATION BEFORE SUBMITTING AN ISSUE
**Describe the bug**
The "Species Stats" pages not showed when birndet-pi install is accessed through a reverse proxy :

 Birdnet-Pi's Species Stats pages when accessed directly (from LAN)

[image]

 Birdnet-Pi's Species Stats pages when trough a reverse proxy - nginx (from internet)

[image]

Every other pages behavior is ok is these conditions.

**Additional context**
Reverse proxy is an nginx with this topology :

[image]

**Your build**
Linux bird 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

**Code or log snippets**
nothing shown in logs

saidone75, 2023-09-11:
Hi Thomas,

I managed to fix that by disabling CORS, etc. by adding:

`--server.enableXsrfProtection=false --server.enableCORS=false`

to /etc/systemd/system/birdnet_stats.service. Hope that help.

tbard1, 2023-09-11:
Hi @saidone75, no success for me. Some improvement (I can see the white screen while streamlit is loading the stats, but as soon as stats are loaded, green screen)

Continuing the quest. 

Thanks for your help !

saidone75, 2023-09-13:
Hi Thomas,

I think I spent over a week to make everything working fine :)
Other than disabling CORS I bound the stats service to the address pointed by proxy rather than localhost:

--server.address birdnetpi1.local

(being birdnetpi1.local defined in /etc/hosts as 192.168.xxx.xxx, the RPi IP in the internal network)
and then the mod_proxy section on Apache httpd:

 ProxyPass "/stats" "http://birdnetpi1.local:8501/stats" upgrade=websocket

tbard1, 2023-09-14:
Hi @saidone75, it works !

I had to adapt the last bit to make it work with nginx (with nginx proxy manager). Here is the adapted @saidone75 trick for nginx : 
[image]

Thanks a lot. 

Issue closed.

closed 2023-09-14 by tbard1